openthread/openthread #6688: Incorrect Enh-Ack frame counter handling on key rotation

hubertmis, 2021-05-27:
It looks OpenThread implementation deviates from Thread 1.2 spec (MAC Security PIB Settings -> secKeyDescriptor for Key ID Mode 1). The spec requires to use separated frame counters per each Key ID Mode 1 key, but OpenThread implements single global frame counter for all keys.

This deviation may result in broken CSL links during key rotation procedure. Links would remain broken until next key rotation, or  SSED re-attachement.

**How links get broken?** 

Network key index is 5. Parent frame counter is 1000, SSED frame counter is 1500.
Key rotation timer is about to expire, Parent's clock is a little faster.

```
           Parent                            SSED
             |                                 |
             |                                 |
             |                                 |
key_rotation |                                 |
   timer     |                                 |
------------>|                                 |
             |---+                             |
             |   | key_index = 6               |
             |   | fc = 0                      |
             |<--+                             |
             |                                 |
             |                                 |
             |                                 |
             |    data_frame:                  |
             |      key_index = 6              |
             |      fc = 0                     |
             |-------------------------------->|
             |                                 |
             |                      Enh-Ack    |
             |                   key_index = 6 |
             |                   fc = 1500     |  Invalid FC value. Should be 0 for new key
             |<--------------------------------|
             |                                 |
             |--+ Update child info:           |--+ Perform key_rotation:
             |  |  key_index = 6               |  | key_index = 6
             |  |  fc = 1500                   |  | fc = 0
             |<-+                              |<-+
             |                                 |
             |                data_frame:      |
             |                  key_index = 6  |
             |                  fc = 0         |
             |<--------------------------------|
             |                                 |
             |--+                              |
             |  | Ignore data frame            |
             |  | fc too small for given key   |
             |<-+                              |
             |                                 |
             |                                 |
             |                                 |
```

**Expected behavior**
Maintaining stable CSL links on key rotation by putting correct FrameCounter value to Enh-Acks.
To achieve it, implementation should follow the specification regarding managing frame counters per keys.

Irving-cl, 2021-07-21:
Hi @hubertmis,

Not sure if I understand this correctly: So after SSED received the `data_frame`(key_index = 6, fc = 0), it will start the key_rotation soon. However the Enh-Ack was sent out before the key_rotation happens. Do you mean we should try to make the key_rotation happen and update the frame counter before the Enh-Ack was sent? Or some other approaches?

hubertmis, 2021-07-21:
> Do you mean we should try to make the key_rotation happen and update the frame counter before the Enh-Ack was sent?

I don't think it is possible to make key rotation during Ack IFS. There would be too much code to execute during short timing-critical period.

I think the best way to solve this issue would be to follow closely Thread 1.2 specification (MAC Security PIB Settings -> secKeyDescriptor for Key ID Mode 1) in OpenThread implementation. Specification requires to use separated frame counter per key used in Key ID Mode 1 (one frame counter for `prev_key`, another one for `curr_key`, yet another one for `next_key`). This way Enh-Ack would use frame counter value for `next_key` which would be 0 (expected value), incremeting it to 1. During key rotation `curr_key`'s frame counter would be moved as `prev_key`'s frame counter (1500 in my example), and `next_key's` frame counter would be moved as `curr_key`'s frame counter (1).

Irving-cl, 2021-07-22:
@hubertmis 

That makes sense. I think we should do in this way. Thanks!

jwhui, 2025-01-07:
Closing stale issue.