openthread/openthread #6714: [srp] support for service sub-types

abtink, 2021-06-09:
It can be helpful to add  (native) support for service sub-types in SRP/DNS modules in OpenThread. This issue is to track the efforts to enable this

FYI. @LuDuda, @Damian-Nordic

Damian-Nordic, 2021-06-09:
FYI @kkasperczyk-no

@abtink mDNS responder allows you to specify subtypes after the main service type, e.g. "_chip._udp,S1,D2" etc. We wondered if OpenThread couldn't supported a similar, compact format.

abtink, 2021-06-09:
That may be useful for CLI but I think in `otSrpClientService` it would be easier to provide sub-type labels as an (optional) array of strings.

```
typedef struct otSrpClientService
{
    const char *         mName;          ///< The service name labels (e.g., "_chip._udp", not the full domain name).
    const char *         mInstanceName;  ///< The service instance name label (not the full name).
    const char **        mSubTypeNames;  ///< Array of service sub-type labels (must end with `NULL` or can be `NULL`).  
    ...
```

kkasperczyk-no, 2021-06-10:
@abtink I made some early implementation in order to verify if the assumptions that we made with @Damian-Nordic are correct and it seems that this way we can properly register sub types for both MDNS and Avahi Publishers. That is WIP branch that still needs some work to do (naming refactor, docs, tests alignment) but I hope it can give you a highlight what the general idea is: https://github.com/kkasperczyk-no/ot-br-posix/commit/645c3e582e8e4c170844dbfed47e5acab8e621d9.

I think that presented approach could be beneficial, because it is very compact solution from the SRP client perspective, as we can just add following subtypes separated with comma to the service type. No additional data are necessary to store, no static buffers allocated for it and there is no increasing memory consumption that is already pretty big problem in Matter.

As far as I understand subtype support could be achieved just by modifying OTBR posix code that is responsible for parsing instance name, service type and domain name. Additionally I added vector of subtypes that could be filled with parsed data and passed to the MDNS/Avahi Publisher. I would be grateful for your statement/comments on that proposition.

abtink, 2021-06-10:
Thanks @KaSroka .

If I understand your approach correctly, the idea is to include the sub-types as comma separated list at end of service name which is then used in all PTR and SRV records in the message. I am not sure if this model is aligned and follows the RFCs on how the names should be formatted and how subtype to be supported.

With this, we will have a PTR  with service name like `"_chip._udp,S1,D2.<domainname>"` mapping to a service instance name formatted as `"<instance>._chip._udp,S1,D2.<domainname>"` and then SRV record with such a instance name. Is there any RFC defining this naming model?

My understanding and assumption from [RFC-6763](https://datatracker.ietf.org/doc/html/rfc6763#section-7.1) is that each service subtype should have its own PTR record mapping the service subtype name to the same service instance name.

```
   Using DNS zone file syntax, the service instance "A web page" is
   advertised using one PTR record, while the instance "A printer's web
   page" is advertised using two: the primary service type and the
   additional subtype.  Even though the "A printer's web page" service
   is advertised two different ways, both PTR records refer to the name
   of the same SRV+TXT record pair:

   ; One PTR record advertises "A web page"
   _http._tcp.local. PTR A\032web\032page._http._tcp.local.

   ; Two different PTR records advertise "A printer's web page"
   _http._tcp.local. PTR A\032printer's\032web\032page._http._tcp.local.
   _printer._sub._http._tcp.local.
                     PTR A\032printer's\032web\032page._http._tcp.local.
```

I expect from SRP protocol perspective, each subtype is handled as a "Service Discovery Instruction"  (basically a PTR record mapping the service subtype name to the service instance name).

@Abhayakara I appreciate your inputs on this.

kkasperczyk-no, 2021-06-11:
@abtink, 
> My understanding and assumption from RFC-6763 is that each service subtype should have its own PTR record mapping the service subtype name to the same service instance name.

Yes and I believe it can be achieved the proposed way.  
Let's assume that we want to add, as you wrote `<instance>._chip._udp.<domainname>` having 3 subtypes: `_S0`, `_A1` and `_T1234`. The service is requested by SRP client as `<instance>._chip._udp,_S0,_A1,_T1234.<domainname>`, parsed in OTBR posix and registered in one of two ways:

- For Avahi using `avahi-publish-service --subtype=_S0._sub._chip._udp`, `avahi-publish-service --subtype=_A1._sub._chip._udp`, `avahi-publish-service --subtype=_T1234._sub._chip._udp`.
- For mDNS using `dns-sd -R <instance> _chip._udp,_S0,_A1,_T1234`.

From the resolver perspective, behavior seems to be as expected. I used `avahi-browse` to verify that we can query the 4 different services (service name itself + 3 subtypes) and we always get the same answer, as they are set to point the same service instance name:

```
$ avahi-browse -rt _chip._udp
+  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
=  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
   hostname = [56B460D5B6FFD7FA.local]
   address = [fd11:22::1445:cf93:a0d5:c843]
   port = [11097]
   txt = ["CRA=300" "CRI=300"]

$ avahi-browse -rt _S0._sub._chip._udp
+  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
=  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
   hostname = [56B460D5B6FFD7FA.local]
   address = [fd11:22::1445:cf93:a0d5:c843]
   port = [11097]
   txt = ["CRA=300" "CRI=300"]

$ avahi-browse -rt _A1._sub._chip._udp
+  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
=  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
   hostname = [56B460D5B6FFD7FA.local]
   address = [fd11:22::1445:cf93:a0d5:c843]
   port = [11097]
   txt = ["CRA=300" "CRI=300"]

$ avahi-browse -rt _T1234._sub._chip._udp
+  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
=  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._udp           local
   hostname = [56B460D5B6FFD7FA.local]
   address = [fd11:22::1445:cf93:a0d5:c843]
   port = [11097]
   txt = ["CRA=300" "CRI=300"]
```

> I expect from SRP protocol perspective, each subtype is handled as a "Service Discovery Instruction" (basically a PTR record mapping the service subtype name to the service instance name).

I'm afraid that in the proposed way the mentioned requirement would not be met. If those are expectations it indeed looks like we would need changing SRP server and SRP client implementation in OpenThread.

abtink, 2021-06-11:
> The service is requested by SRP client as <instance>._chip._udp,_S0,_A1,_T1234.<domainname>, parsed in OTBR posix and registered in one of two ways.

We need the message from SRP client on a thread device to follow the RFCs and use valid DNS name formats. If we use the name format `"_chip._udp,_S0,_A1.<domainname>"` in a PRT/SRV record (in an SRP update message sent from SRP client) it won't be considered valid format. This model is basically adding a custom solution that only works between OT implementation of SRP client and server. Note that OT SRP client implementation is expected to work with other SRP server implementation and not just the OT implementation. 

> I'm afraid that in the proposed way the mentioned requirement would not be met. If those are expectations it indeed looks like we would need changing SRP server and SRP client implementation in the OpenThread.

Yes. I think we do. This is the reason for this issue (to track the efforts). I will have a PR with client side soon and then we need to consider the server side change.

Abhayakara, 2021-06-11:
> The service is requested by SRP client as ._chip._udp,_S0,_A1,_T1234.,

I don't think this proposal is a good approach. The way dns name
compression works, you aren't saving much space by doing this. SRP already
supports multiple services in one update; with name compression, these do
not add much more data than is used by the additional service type name.

simonlingoogle, 2021-06-12:
> * For Avahi using `avahi-publish-service --subtype=_S0._sub._chip._udp --subtype=_A1._sub._chip._udp --subtype=_T1234._sub._chip._udp`.
> * For mDNS using `dns-sd -R <instance> _chip._udp,_S0,_A1,_T1234`.

I think this is just the way of Avahi and mDNS to specify subtypes for services. It does not mean the subtypes are actually encoded in the service name. Everything should work along with RFC6763.

Abhayakara, 2021-06-12:
That’s right. That’s an api convenience. The names are published as
regular domain names with no commas.

abtink, 2021-06-12:
Submitted PR https://github.com/openthread/openthread/pull/6724 earlier. It adds support for handling service subtype on SRP client side along with support for it in CLI commands.

kkasperczyk-no, 2021-06-17:
@abtink I have a problem with understanding how much effort is needed - beside of your PR with SRP client subtypes support - to provide complete working solution. I assume that still some change in OTBR posix will be needed to extract subtypes list from service structure and add it to mDNS/Avahi registration methods, but I don't know if something else in the OpenThread itself will be needed? 

I'm asking, as I tried to checkout your PR and test it with the latest OTBR, but I didn't manage to make it work, because SRP server was refusing services that I tried to register through OT CLI using syntax suggested in readme file (`_test2._udp,_sub1,_sub2`). I would like to briefly estimate how much time is needed to allow us start using subtypes in Matter, so could you share some plans regarding that? Also I'm not sure if my problem is valid one, so is it normal that for now server refuses such requests or I did something wrong and I should investigate it further.

abtink, 2021-06-17:
@kkasperczyk-no , yes, the OT SRP server implementation needs to be updated to add support for handling sub-types (the current implementation rejects a message if it sees same "service instance name" being used in multiple PTR records).

I am working on the SRP server side code to add support for this (you may have noticed some related PRs in past few days like https://github.com/openthread/openthread/pull/6732 or https://github.com/openthread/openthread/pull/6733  😃 ...these are mainly intended to help with the server side sub-type support). 

I expect/hope to have the server side PR in OT up for review by mid next week.

Beyond this, I think (and as you said) that we need some changes in OTBR to hook subtypes to mDNS/Avahi, etc. I leave that for people who are more familiar with the OTBR and the related modules. Would you be willing to help with the OTBR part?

kkasperczyk-no, 2021-06-18:
@abtink, thanks for clarification, no doubts now.
> you may have noticed some related PRs in past few days
>
Yes I did, however I was wondering how many ongoing PRs are needed to call it complete and working :smiley: 
> I leave that for people who are more familiar with the OTBR and the related modules. Would you be willing to help with the OTBR part?
>
I'm not an expert there, but I guess I know what should be done, so sure, I can try to handle it. Please could you keep me in the loop then and mention when there will something pretty much ready for testing?

abtink, 2021-06-18:
> Please could you keep me in the loop then and mention when there will something pretty much ready for testing?

Will do. 
Will also link all related PRs to this issue as well (so we can track).

abtink, 2021-06-23:
@kkasperczyk-no, I just submitted PR https://github.com/openthread/openthread/pull/6760. This fully adds support for service sub-type in OpenThread's SRP server implementation.

This PR also includes the commit from #6724 (*[srp-client] add support for service subtypes.*) so it contains all changes for subtype support on both client and server sides. There is also a CI test case added for this.

abtink, 2021-06-30:
Upon testing we realized we need some smaller changes in DNS-SD server and clinet for subtype support (being able to browse for service subtype). Submitted the following PR:
- https://github.com/openthread/openthread/pull/6773

jwhui, 2021-07-02:
Resolved by:

- #6724 
- #6760 
- #6773

closed 2021-07-02 by jwhui