openthread/openthread #6743: [github-actions] test_srp_server_reboot_port.py fails intermittently

jwhui, 2021-06-17:
[logs_82299.zip](https://github.com/openthread/openthread/files/6667852/logs_82299.zip)
[packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/6667853/packet-verification-1.1-on-1.2-pcaps.zip)

jwhui, 2021-06-17:
[logs_82295.zip](https://github.com/openthread/openthread/files/6667861/logs_82295.zip)
[packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/6667862/packet-verification-pcaps.zip)

superwhd, 2021-06-17:
Some investigation:

The test complained that the SRP server listens to the same port after a reboot.

I reproduced the test failure locally. According to the logs, the SRP server actually listens to a different port after every reboot. The problem is that the test gets the server port from stale network data. Sometimes the port number is not changed in time so the test still gets the old port number and the test fails. However, 5 seconds should be sufficient for the netdata to be renewed.

I extended the wait time after reboot to 100 seconds then there are no failures. However, that's not a reasonable fix.

@simonlingoogle also mentioned there might be some issue in the implementation of `reset()`, which could be relevant to this issue.

simonlingoogle, 2021-06-18:
This issue is much more likely to happen when the CPU is busy. So, it's very likely a virtual time simulator issue. 

```
2: reset
alarm ('127.0.0.1', 9002) current_time 12000000 delay 2147483647
2: ifconfig up
alarm ('127.0.0.1', 9002) current_time 12000000 delay 10250000
alarm ('127.0.0.1', 9002) current_time 12000000 delay 10250000
```
In this test, node 2 sent multiple alarm events after command ifconfig up.  But I think a node should never send multiple alarm events for one timestamp without further input. 
It may be due to Node unable to receive the full input ifconfig up\r\n within one select loop, so that it ran multiple select loop and sent multiple alarm events.

abtink, 2021-06-23:
Submitted PR https://github.com/openthread/openthread/pull/6763 which I think should help address the intermittent failures of this test (we can just restart the server instead of doing full reset in the test  and it will cover the SRP server port change behavior that we want to verify).

Maybe we can keep this issue open (or file a new one) to further investigate the simulator reset issue (I think it would be helpful to dig deeper and address this).

abtink, 2021-06-24:
I think we may have another issue in `Srp::Server` code for `SRP_SERVER_PORT_SWITCH_ENABLE` where `mHasRegisteredAnyService` was not being cleared from `Server::Stop()`. So if the server is stoped and restarted a bunch of times in a row and we have services registered each time, we would only save the port in `Settings` the first time a service is added and the next time the server restarts even though the port changes, we won't save it in `Settings` since `mHasRegisteredAnyService` remained `true` from last time.

I added a new commit to change this in the same PR #6763.

PR #6763 (2021-06-24): [tests] restart SRP server (instead of device reset) to validate port change
Added a second commit to this PR. First commit changes the test to do restart:

**[test] restart SRP server (instead of device reset) to validate port change (#6763)**

This commit updates `test_srp_server_reboot_port` to just restart the
SRP sever instead of performing a full device reset. The SRP restart
will trigger the port number change by the server which is the
behavior this test-case is intended to verify. This change should
help address intermittent failures of this test.

Second commit:

**[srp-server] ensur [... cut]
closed 2021-06-24 by jwhui