openthread/openthread #6748: GRL TH 1.1 error when adding device (Err: AddConnectionInfo DM: Look at this and add counter --<type 'unicode'>)

JaneFromSilabs, 2021-06-17:
**Describe the bug** A clear and concise description of what the bug is.
```
Error when attempting to add device.
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id - `95c5cb793`
2. IEEE 802.15.4 hardware platform `EFR32MG21 + Raspi/OTBR host`
4. Network topology - `GRL 1.1`

**Expected behavior** A clear and concise description of what you expected to happen.

```
I was hoping to add this device (xml below) to the 1.1 GRL TH as I did with previous THCI
```

**Console/log output** If applicable, add console/log output to help explain your problem.
```
GRL TH v1.1 error:
```

```
None - Jun 17 17:07:33 - FUNC intialize failed: <type 'unicode'>
Err: AddConnectionInfo DM: Look at this and add counter --<type 'unicode'>
Log -->   [DM: [AddConnectionInfoForDevice] --> <type 'unicode'>]
```

**Additional context** Add any other context about the problem here.
```
Xml entry used:
```

```
    <DEVICE name="OTBR" thumbnail="SBR.jpg" description = "SiLabs OTBR" THCI="OpenThread_BR">
        <ITEM label="IP addr" 
              type="text" 
              forParam="TelnetIP" 
              validation="ipv4" 
              hint="eg: 10.4.164.122">10.4.164.122
        </ITEM>
        <ITEM label="PORT" 
              type="text" 
              forParam="TelnetPort" 
              validation="port" 
              hint="eg: 4901">22
        </ITEM>
        <ITEM type="hidden" label = ""
              forParam="Param5">ip
        </ITEM>
    </DEVICE>    
```

```
Altered 1 line in OpenThread_BR THCI:
```
```
    def _cliWriteLine(self, line):
        cmd = 'docker exec otbr sh -c "sudo ot-ctl %s"' % line
```

JaneFromSilabs, 2021-06-17:
It looks like I am getting this from the 1.2 TH as well:
```
COM16 - Jun 17 19:00:07 - FUNC intialize failed: <type 'unicode'>
Err: AddConnectionInfo DM: Look at this and add counter --<type 'unicode'>
Log -->   [DM: [AddConnectionInfoForDevice] --> <type 'unicode'>]
```

karthick-grl, 2021-06-18:
I think this is due to the assertion error, `SerialPort` parameter is expected in unicode. 

https://github.com/openthread/openthread/blob/e66b9a3958bfdc3fb495a189737265561e9a99bd/tools/harness-thci/OpenThread.py#L365

JaneFromSilabs, 2021-06-18:
If I comment out the assertion on that line, I am able to add my device.  Thank you, @karthick-grl !

simonlingoogle, 2021-06-23:
Hi @JaneFromSilabs ,

Can you submit a fix for this issue? I don't have a TesTH1.1 test bed on hand.

JaneFromSilabs, 2021-06-23:
The PR is here:
https://github.com/openthread/openthread/pull/6762

PR #6762 (2021-06-24): [thci] remove assert which checks if the port is unicode
In the case where the connection is over Telnet, `self.port` will be `None`.

Closes #6748
closed 2021-06-24 by jwhui