openthread/openthread #6750: Issues in Manager::GetBackboneRouterPrimary() - right Primary BBR isn't always selected

EskoDijk, 2021-06-18:
**Describe the bug**
Based on code inspection, it looks like the right Primary BBR is not selected in various cases, based on multiple potential BBR Datasets in the Network Data.

**To Reproduce**
Looking at the code below from network_data_service.cpp

`
void Manager::GetBackboneRouterPrimary(ot::BackboneRouter::BackboneRouterConfig &aConfig) const
{
    const ServerTlv *                 rvalServerTlv  = nullptr;
    const BackboneRouter::ServerData *rvalServerData = nullptr;
    Iterator                          iterator;

    aConfig.mServer16 = Mac::kShortAddrInvalid;

    iterator.mServiceTlv = Get<Leader>().FindService(kThreadEnterpriseNumber, &BackboneRouter::kServiceData,
                                                     sizeof(BackboneRouter::kServiceData));

    while (IterateToNextServer(iterator) == kErrorNone)
    {
        const BackboneRouter::ServerData *serverData;

        if (iterator.mServerSubTlv->GetServerDataLength() < sizeof(BackboneRouter::ServerData))
        {
            continue;
        }

        serverData = reinterpret_cast<const BackboneRouter::ServerData *>(iterator.mServerSubTlv->GetServerData());

        if (rvalServerTlv == nullptr ||
            (iterator.mServerSubTlv->GetServer16() ==
             Mle::Mle::Rloc16FromRouterId(Get<Mle::MleRouter>().GetLeaderId())) ||
            serverData->GetSequenceNumber() > rvalServerData->GetSequenceNumber() ||
            (serverData->GetSequenceNumber() == rvalServerData->GetSequenceNumber() &&
             iterator.mServerSubTlv->GetServer16() > rvalServerTlv->GetServer16()))
        {
            rvalServerTlv  = iterator.mServerSubTlv;
            rvalServerData = serverData;
        }
    }
`

Git commit id: e66b9a3958bfdc3fb495a189737265561e9a99bd

**Expected behavior** A clear and concise description of what you expected to happen.
1. The Get<Leader>().FindService() call will only return exact-matching services, i.e. of exactly length sizeof(BackboneRouter::kServiceData). However the length of the BBR Dataset may be longer as specified in Thread, and the recipient MUST ignore the additional bytes if any. So expected behavior is that longer BBR Datasets are also accepted.
2. In addition, the FindService() call does not check for the presence of the uint8 constant THREAD_SERVICE_DATA_BBR = 0x01 which needs to be present in the S_service_data part of the Service TLV. If other value is present there, it is another type of data from same vendor ID. So expected behavior is that the code will check for presence of 0x01 in the Service TLV and only use those datasets.
3. The assignment of using a 'better' BBR Dataset (assigning to rvalServerTlv/rvalServerData) is not strictly using ranked criteria. For example, if a Leader offers a BBR Dataset, it will immediately supersede any other found dataset even one with a higher sequence number. And also, if a next dataset in the iteration offers a higher sequence-number, this will immediately supersede any previous set including sets offered by a Leader. So the selection depends on the iteration order which we don't want. Expected behavior here is that when a Leader's BBR Dataset is found, the iteration stops and the dataset offered by the Leader is used directly.

Note: above behavior on point 3 is compliant to Rules 3,4,5 in the 1.2.1-draft5 spec, but not to rules 1, 2.  (The latter two would be more work to integrate properly.)

**Console/log output** 
Not based on running code

**Additional context** 
None

simonlingoogle, 2021-06-18:
> * The Get().FindService() call will only return exact-matching services, i.e. of exactly length sizeof(BackboneRouter::kServiceData). However the length of the Service TLV data of the BBR Dataset may be longer as specified in Thread, and the recipient MUST ignore the additional bytes if any. So expected behavior is that longer BBR Datasets are also accepted.

Agree. 

> * In addition, the FindService() call does not check for the presence of the uint8 constant THREAD_SERVICE_DATA_BBR = 0x01 which needs to be present in the S_service_data part of the Service TLV. If [...]

Isn't it checked here?

https://github.com/openthread/openthread/blob/e66b9a3958bfdc3fb495a189737265561e9a99bd/src/core/thread/network_data.cpp#L600

> * The assignment of using a 'better' BBR Dataset (assigning to rvalServerTlv/rvalServerData) is not strictly using ranked criteria. For example, if a Leader offers a BBR Dataset, it will immediately [...]

Good catch!

abtink, 2021-06-18:
Thanks @EskoDijk .

> The Get().FindService() call will only return exact-matching services

Yes `FindService()` expects an exact match of service data. I think this was assumed since the first BBR PR #4430 ([diff](https://github.com/openthread/openthread/pull/4430/files#diff-32a3a35c91c6dc788b9c4d2549a37ce9eb81c9ddfcbc732d278afe486efe5fd1R97))

We now have `FindNextMatchingService()` which will be better fit for this:
https://github.com/openthread/openthread/blob/e66b9a3958bfdc3fb495a189737265561e9a99bd/src/core/thread/network_data.hpp#L494-L518

----

> In addition, the FindService() call does not check for the presence of the uint8 constant THREAD_SERVICE_DATA_BBR = 0x01 which needs to be present in the S_service_data part of the Service TLV.

That should be part of `BackboneRouter::kServiceData` which we pass to to `Find()` method to match with the service data.

https://github.com/openthread/openthread/blob/e66b9a3958bfdc3fb495a189737265561e9a99bd/src/core/thread/network_data_service.hpp#L66-L75

jwhui, 2021-06-18:
@EskoDijk , thanks for raising this issue.

Who wants to submit a PR? :)

simonlingoogle, 2021-06-19:
Submitted https://github.com/openthread/openthread/pull/6751 to fix issue 3.

EskoDijk, 2021-06-22:
Thanks for the feedback, indeed the THREAD_SERVICE_DATA_BBR = 0x01 is already tested I see now! 
One general question: which version of the 1.2/1.2.1 spec is being targeted by current code for "1.2" ? Always the latest draft release, or a specific version?

jwhui, 2021-06-22:
> One general question: which version of the 1.2/1.2.1 spec is being targeted by current code for "1.2" ? Always the latest draft release, or a specific version?

@EskoDijk , we are targeting whatever will be required for 1.2 certification.

simonlingoogle, 2021-06-23:
@EskoDijk 

Thanks for reporting the bugs. 
I have filed #6751 and #6752 to fix the issues. Please take a look when you have time.

EskoDijk, 2021-06-28:
@simonlingoogle Thanks, I've added my comments to #6752 .

jwhui, 2021-08-11:
Resolved by #6751 and #6752 .

closed 2021-08-11 by jwhui