openthread/openthread #6755: Missing duration update in wait() method in otci.py

qilinkang, 2021-06-21:
**Describe the bug** A clear and concise description of what the bug is.

https://github.com/openthread/openthread/blob/main/tools/otci/otci/otci.py

For wait() in otci.py duration variable in while loop has not been updated:

while duration > 0:
                output = self.__otcmd.wait(1)
                if match_line(expect_line, output):
                    success = True
                    break

if not success:
                raise ExpectLineTimeoutError(expect_line)

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

simonlingoogle, 2021-06-23:
Good catch. Thanks @qilinkang . 
I will file a fix ASAP.

simonlingoogle, 2021-06-23:
@qilinkang I have sent https://github.com/openthread/openthread/pull/6759 to fix this bug. PTAL.

PR #6759 (2021-06-25): [otci] fix wait duration
Fixes #6755
closed 2021-06-25 by jwhui