openthread/openthread #6772: thread return PollPeriod error

taoyinzhou, 2021-06-29:
**Describe the bug** A clear and concise description of what the bug is.

In function "otLinkGetPollPeriod", it calls GetKeepAlivePollPeriod. I think this might be GetExternalPollPeriod.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2021-06-30:
> In function "otLinkGetPollPeriod", it calls GetKeepAlivePollPeriod. I think this might be GetExternalPollPeriod.

@taoyinzhou , the current implementation is as intended. `GetKeepAlivePollPeriod()` returns the maximum poll interval that the device will use. In other words, it returns the minimum of the user-specified poll interval and the default poll interval.

jwhui, 2021-08-09:
Closing stale issue.

closed 2021-08-09 by jwhui