openthread/openthread #6774: ot-daemon unable to establish spinel connection with Arduino Nano 33 BLE running zephyr+ot-rcp

nageshshamnur, 2021-06-30:
**Describe the bug** I was trying to setup my Arduino Nano 33 BLE sense as openthread RCP along with zephyr as suggested in the zephyr page: https://docs.zephyrproject.org/latest/samples/net/openthread/coprocessor/README.html. I was able to flash the image successfully and the board is up with port ```/dev/ttyACM1`` available for ot-daemon.
   When i run the ```ot-daemon``` on my linux host with command : ```sudo ./build/posix/src/posix/ot-daemon -v 'spinel+hdlc+uart:///dev/ttyACM2?uart-baudrate=115200'```, then init( ) was failure. On further debugging found out that ```ot-daemon``` works with spinel communicating with target board via socket and the select waiting for the socket in ot-daemon was failing, thus unable to establish the spinel connection with the board. 

**To Reproduce** 
1. zephyr commit id: https://github.com/zephyrproject-rtos/zephyr/commit/457a28bf78a4cdd2b17f3983c50a0183f2f71700 with patch: https://github.com/zephyrproject-rtos/zephyr/commit/6c9945aafa00c09149e2052a9c2bccad16dd1d8a applied.
2. openthread commit id: https://github.com/openthread/openthread/commit/3a7e836dd7c96a932cbd8ab7733b1042bfd86588
2. IEEE 802.15.4 hardware platform: Arduino Nano 33 BLE Sense
3. zephyr build:
```
west build -p auto -b arduino_nano_33_ble samples/net/openthread/coprocessor -- -DCONF_FILE="prj.conf overlay-rcp.conf overlay-usb-nrf-br.conf"
west flash --bossac=<Arduino_bossac_path>
```
4. ot-daemon command:
```sudo ./build/posix/src/posix/ot-daemon -v 'spinel+hdlc+uart:///dev/ttyACM2?uart-baudrate=115200'```

5. Network topology: ot-daemon(Linux Host) <----spinel----> ot-rcp(Arduino Nano 33 ble)

**Expected behavior** ot-daemon should be success.

**Console/log output** 
ot-daemon error:
```/build/posix/src/posix/ot-daemon[232720]: [CRIT]-PLAT----: Init() at ../../src/lib/spinel/radio_spinel_impl.hpp:278: Failure```

**Additional context** None

jwhui, 2021-06-30:
@nageshshamnur , you mentioned that the board path appears as `/dev/ttyACM1`, but then the radio url has `/dev/ttyACM2`. Is that different intentional?

nageshshamnur, 2021-07-22:
Hi @jwhui : i have tried all the combination (ACM0, ACM1 and ACM2), since as per the documentation it creates 2 ports are created one for application access and another for spinel

jwhui, 2021-07-22:
@nageshshamnur , given that this issue seems specific to Zephyr, please post an issue in the Zephyr project: https://github.com/zephyrproject-rtos/zephyr/issues

jwhui, 2021-08-11:
Closing stale issue.

closed 2021-08-11 by jwhui