openthread/openthread #6792: What does BeginTransmit do?

purvisb, 2021-07-05:
https://github.com/openthread/openthread/blob/24177cade4867212da3739251cd68524fa879692/src/core/radio/trel_link.cpp#L123

Does this function assemble 802.15.4 frames for beacons? Is this called when the CLI command `scan` is called? Can I use this function as an example of how to customize beacon request? 

I'm also trying to see what happens when I call this function in mac.cpp on a simulated OT-CLI-FTD node. Why does it fail with this error? I thought maybe something needs to be enabled. I enabled TREL when compiling the binary and that did nothing. I'm not sure how to resolve this error.
```
ot-cli-ftd: /home/vagrant/openthread/examples/../src/core/mac/mac_links.hpp:517: void ot::Mac::Links::Send(): Assertion `error == kErrorNone' failed.
                                                   Aborted (core dumped)
```

abtink, 2021-07-07:
The `Trel::Link::BeginTransmit()` is a `private` method on `Trel::Link` class. It starts the tx operation (from a tasklet). It is intended to be used by `Trel::LInk` only and won't recommend calling it from any other class (e.g. from `Mac`). 

TREL is part of OpenThread support for multi-radio and "Thread over Infrastructure (WiFi)". PR https://github.com/openthread/openthread/pull/4440 can provide more details.

> Does this function assemble 802.15.4 frames for beacons?

No. It does not

> Is this called when the CLI command scan is called? 

It may be called if TREL is enabled and used. 

> Can I use this function as an example of how to customize beacon request?

Beacon requests are prepared by `Mac` class. The method `Mac::PrepareBeaconRequest()` prepares the frame.

purvisb, 2021-07-07:
Thank you for the clarifications.

closed 2021-07-07 by purvisb