openthread/openthread #6793: [github-actions] Cert_9_2_16_ActivePendingPartition fails intermittently

librasungirl, 2021-07-05:
[logs_84486.zip](https://github.com/openthread/openthread/files/6766494/logs_84486.zip)
[packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/6766498/packet-verification-1.1-on-1.2-pcaps.zip)

jwhui, 2021-09-14:
[logs_95452.zip](https://github.com/openthread/openthread/files/7164668/logs_95452.zip)
[packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/7164669/packet-verification-1.1-on-1.2-pcaps.zip)

jwhui, 2021-11-12:
[logs_102697.zip](https://github.com/openthread/openthread/files/7530517/logs_102697.zip)
[packet-verification-pcaps.zip](https://github.com/openthread/openthread/files/7530518/packet-verification-pcaps.zip)

jwhui, 2022-01-20:
[logs_111185.zip](https://github.com/openthread/openthread/files/7902938/logs_111185.zip)
[packet-verification-1.1-on-1.2-pcaps.zip](https://github.com/openthread/openthread/files/7902942/packet-verification-1.1-on-1.2-pcaps.zip)

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui