openthread/openthread #6794: Thread sniffer to catch the traffic for ongoing openthread connection

Hard1kPrajapat1, 2021-07-06:
Hello Team,

I have created an Openthread network where border routers, other routers, and end devices(nRF52840) are there. I just want to sniff the IPv6 packets on this Openthread network. Can anyone guide or send a link?

I have also checked that the nRf52840 USB dongle can be used for sniffing but I think it will work on the BLE protocol and in my case, I am only going to use thread, not BLE.

Thanks in advance,
Hardik

jwhui, 2021-07-06:
See:
- https://openthread.io/guides/pyspinel/sniffer
- https://openthread.io/guides/pyspinel/sniffer-extcap

Hard1kPrajapat1, 2021-07-06:
Thanks @jwhui, I will check those links. Also, can you please confirm the below statement?

The nrf52840 USB dongle is used as a sniffer only in the case of BLE, is it correct? What If I flash it with thread only binary, so will I be able to sniff the thread packets?

simonlingoogle, 2021-07-06:
> The nrf52840 USB dongle is used as a sniffer only in the case of BLE, is it correct? What If I flash it with thread only binary, so will I be able to sniff the thread packets?

Yes. Both nRF52840 DK and USB Dongle can be used as a Thread sniffer. 

BTW, Make sure to use `USB_trans -DOT_BOOTLOADER=USB` when building firmware for USB Dongle. Please check https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md#nrf52840-dongle-support-pca10059 for more details.

jwhui, 2021-08-11:
Closing stale issue.

closed 2021-08-11 by jwhui