openthread/openthread #6796: [dua] Deadlock on DUA.req of second MTD child

edmont, 2021-07-06:
**Describe the bug** A clear and concise description of what the bug is.

> A router sending a DUA.req on behalf of an MTD child will use the same information for Target EID TLV and ML-EID TLV as the used for a previously registered one for another MTD child. It seems the CoAP ACK from the BBR doesn't match with the expected Target EID and the router will enter a DUA registration deadlock of continuous DUA.req transmissions.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id

> https://github.com/openthread/ot-nrf528xx/commit/3a33879eaf9a48a3e9b56da1592867928ad9069e

2. IEEE 802.15.4 hardware platform

> 4 x nRF52840-DK

3. Build steps
```
OT_CMAKE_BUILD_DIR=/tmp/dk_build ./script/build nrf52840 UART_trans -DOT_THREAD_VERSION=1.2 -DOT_DUA=ON -DOT_BACKBONE_ROUTER_DUA_NDPROXYING=ON -DOT_BORDER_ROUTER=ON -DOT_SERVICE=ON -DOT_MAC_FILTER=ON
arm-none-eabi-objcopy -O ihex /tmp/dk_build/bin/ot-cli-ftd /tmp/ot.hex
nrfjprog --program /tmp/ot.hex --chiperase --reset
```
4. Network topology
```
# Leader
extaddr 166e0a0000000001
macfilter addr allowlist
macfilter addr add 166e0a0000000002
panid 0xface
masterkey 00112233445566778899aabbccddeeff
ifconfig up
thread start
bbr enable
prefix add fd00:7d03:7d03:7d03::/64 prosD med
netdata register

# Router
extaddr 166e0a0000000002
panid 0xface
masterkey 00112233445566778899aabbccddeeff
ifconfig up
thread start

# MED1
extaddr 166e0a0000000003
mode rn
panid 0xface
masterkey 00112233445566778899aabbccddeeff
ifconfig up
thread start

# MED2
extaddr 166e0a0000000004
mode rn
panid 0xface
masterkey 00112233445566778899aabbccddeeff
ifconfig up
thread start
```

**Expected behavior** A clear and concise description of what you expected to happen.

> DUA.req is sent properly for the second child.

edmont, 2021-07-06:
@librasungirl do you mind taking a look here?

simonlingoogle, 2021-07-06:
Thanks for revealing the bug. 

When the device is deadlocking, is it possible to get the running stack trace using J-Link? It can be extremely useful for identifying the cause. @edmont

simonlingoogle, 2021-07-06:
Without digging to much into this bug, I feel this has something to do with `mChildIndexDuaRegistering` logics. 
We intended to refactor `mChildIndexDuaRegistering` logics a while ago, but could make time for it yet.

edmont, 2021-07-06:
@simonlingoogle I updated the description to be more precise, not a CPU deadlock but a DUA registration infinite loop.

simonlingoogle, 2021-07-06:
> @simonlingoogle I updated the description to be more precise, not a CPU deadlock but a DUA registration infinite loop.

Thanks for clarification. If so, stack trace is less useful.

simonlingoogle, 2021-07-06:
@edmont I will send a potential fix ASAP.

simonlingoogle, 2021-07-06:
> It seems the CoAP ACK from the BBR doesn't match with the expected Target EID and the router will enter a DUA registration infinite loop of DUA.req transmissions.

What's the expected behavior in such case? Should the Parent wait for a while before resending?

librasungirl, 2021-07-07:
@edmont [v1_2_test_domain_unicast_address_registration.py](https://github.com/openthread/openthread/blob/main/tests/scripts/thread-cert/v1_2_test_domain_unicast_address_registration.py#L104-L105) should have covered the scenario, and we do have observed the same issue in the pcap after local run.
Also noticed some other issues when looking into the pcap and finer message field check should be added to avoid false pass. we'll further look into the fixes and keep you updated. Thanks.

simonlingoogle, 2021-07-07:
@librasungirl @edmont Submitted a quick fix #6797 and verified that it works.

edmont, 2021-07-07:
@librasungirl I started working in this patch for `v1_2_test_domain_unicast_address_registration.py`:
```
diff --git a/tests/scripts/thread-cert/v1_2_test_domain_unicast_address_registration.py b/tests/scripts/thread-cert/v1_2_test_domain_unicast_address_registration.py
index 17183d46d..89189ad99 100755
--- a/tests/scripts/thread-cert/v1_2_test_domain_unicast_address_registration.py
+++ b/tests/scripts/thread-cert/v1_2_test_domain_unicast_address_registration.py
@@ -33,6 +33,7 @@ import command
 import config
 import ipaddress
 import mle
+import network_layer
 import thread_cert
 
 BBR_1 = 1  # Collapsed with Leader Role
@@ -164,13 +165,16 @@ class TestDomainUnicastAddressRegistration(thread_cert.TestCase):
         '''
         return ''.join(ipaddress.ip_address(address).exploded.split(':')[4:])
 
-    def __check_dua_registration_tmf(self, node, occurences=1):
+    def __check_dua_registration_tmf(self, node, occurences=1, ml_eid=None):
 
         messages = self.simulator.get_messages_sent_by(node)
         for i in range(occurences):
             msg = messages.next_coap_message('0.02', '/n/dr', False)
             assert msg, 'Expected {}, but {}th not found\n node: {}(extaddr: {})'.format(
                 occurences, i + 1, node, self.nodes[node].get_addr64())
+            if ml_eid:
+                ml_eid_tlv = msg.get_coap_message_tlv(network_layer.MlEid)
+                self.assertEqual(ml_eid, ml_eid_tlv)
 
     def test(self):
         # starting context id
@@ -342,7 +346,7 @@ class TestDomainUnicastAddressRegistration(thread_cert.TestCase):
             WAIT_TIME = PARENT_AGGREGATE_DELAY + WAIT_REDUNDANCE
             print("waiting {}".format(WAIT_TIME))
             self.simulator.go(WAIT_TIME)
-            self.__check_dua_registration_tmf(ROUTER_1_2, 1)
+            self.__check_dua_registration_tmf(ROUTER_1_2, 1, node.get_ip6_address(config.ADDRESS_TYPE.ML_EID))
 
         # 6) Increase seqno on BBR_1, within  BBR_REREGISTRATION_DELAY, there should be one DUA.req from
         #    per [FED_1_2_1, MED_1_2_1, SED_1_2_1, FED_1_2_2], and 3 DUA.req from ROUTER_1_2 among which
```

Then to realize that this specific test is not being used by the CI: https://github.com/openthread/openthread/blob/main/.github/workflows/simulation-1.2.yml#L272-L277

Furthermore, when running it locally I get this error:
`FileNotFoundError: [Errno 2] No such file or directory: './ot-cli-ftd'`

simonlingoogle, 2021-07-07:
> FileNotFoundError: [Errno 2] No such file or directory: './ot-cli-ftd

Try use `export INTER_OP=1`

PR #6797 (2021-07-07): [dua] fix infinite DUA.req loop
Fixes #6796 

Verified that this commit fixes the infinite DUA.req loop.

This is a quick fix. We still need to reactor `mChildIndexDuaRegistering` because there are still potential issues.
closed 2021-07-07 by jwhui