openthread/openthread #6801: Difference between 802.15.4 scan and MLE discovery

purvisb, 2021-07-08:
[IEEE 802.15.4 Active Scan](https://github.com/openthread/openthread/blob/main/src/cli/README.md#scan-channel) seems to have the same output as [MLE Discovery operation](https://github.com/openthread/openthread/blob/main/src/cli/README.md#discover-channel). But the Wireshark capture is different. What is the practical difference of these two CLI commands?

jwhui, 2021-07-08:
IEEE 802.15.4 Active Scan operates at the MAC layer, whereas MLE Discovery operates at the MLE layer. By operating at a higher layer, MLE Discovery is more capable in terms of filtering.

See the API reference docs to see the differences:

- [otThreadDiscover()](https://openthread.io/reference/group/api-thread-general#otthreaddiscover)
- [otLinkActiveScan()](https://openthread.io/reference/group/api-link-link#otlinkactivescan)

purvisb, 2021-07-08:
Thank you for the clarification.

closed 2021-07-08 by purvisb