openthread/openthread #6805: [github-actions] MATN_12_HopLimitProcessing.py fails intermittently

jwhui, 2021-07-08:
[logs_85259.zip](https://github.com/openthread/openthread/files/6786223/logs_85259.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6786224/thread-border-router-results.zip)

superwhd, 2021-07-12:
Some notes:

The direct cause was that the variable `BR_1_RLOC` wasn't collected in the test phase.
```
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 282, in tearDown
    self._verify_packets(test_info_path)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 310, in _verify_packets
    self.verify(pv)
  File "./tests/scripts/thread-cert/border_router/MATN_12_HopLimitProcessing.py", line 178, in verify
    .filter_AMPLFMA(mpl_seed_id=vars['BR_1_RLOC']) \
KeyError: 'BR_1_RLOC'
```

Taking a look at the logs, I found that there wasn't any output for the command `ipaddr rloc`.
```
Jul  8 17:06:22 00f6248ffedc otbr-agent[118]: [INFO]-PLAT----: > ipaddr
Jul  8 17:06:22 00f6248ffedc otbr-agent[118]: [INFO]-PLAT----: > ipaddr mleid
2021-07-08 17:06:22,930 - INFO - Host<3> $ '48: eth0@if49: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP '
2021-07-08 17:06:22,930 - INFO - Host<3> $ '    inet6 fdfa:affb:4701:0:42:acff:fe12:3/64 scope global dynamic mngtmpaddr '
2021-07-08 17:06:22,930 - INFO - Host<3> $ '       valid_lft 1717sec preferred_lft 1717sec'
2021-07-08 17:06:22,930 - INFO - Host<3> $ '    inet6 9105::3/64 scope global nodad '
2021-07-08 17:06:22,930 - INFO - Host<3> $ '       valid_lft forever preferred_lft forever'
2021-07-08 17:06:22,930 - INFO - Host<3> $ '    inet6 fe80::42:acff:fe12:3/64 scope link '
2021-07-08 17:06:22,930 - INFO - Host<3> $ '       valid_lft forever preferred_lft forever'
2021-07-08 17:06:22,934 - DEBUG - Host<3>: get_ether_addrs: ['fdfa:affb:4701:0:42:acff:fe12:3', '9105::3', 'fe80::42:acff:fe12:3']
1: rloc16
2: rloc16
rloc16
6c00
Done
> 1: ipaddr rloc
Jul  8 17:06:22 00f6248ffedc otbr-agent[118]: [INFO]-PLAT----: > rloc16
Jul  8 17:06:22 00f6248ffedc otbr-agent[118]: [INFO]-PLAT----: > ipaddr rloc

Packets captured: 134
Packets received/dropped on interface 'backbone5': 134/0 (pcap:0/dumpcap:0/flushed:0/ps_ifdrop:0) (100.0%)
2021-07-08 17:06:25,977 - INFO - Backbone sniffer terminated successfully: pid=19124
1: thread stop
1: ifconfig down
```

Maybe the BR (or test script) exited too early, or the CLI didn't give any addresses as the output. 

Later on I'll investigate more.

jwhui, 2021-07-12:
> Maybe the BR (or test script) exited too early, or the CLI didn't give any addresses as the output.
> 
> Later on I'll investigate more.

@superwhd , thanks for the update!

superwhd, 2021-07-22:
Are there any recent failures? I previously tried to reproduce the error locally (~300 times) but failed.

superwhd, 2021-07-27:
[logs_87356.zip](https://github.com/openthread/openthread/files/6882074/logs_87356.zip)

closed 2021-08-11 by jwhui