openthread/openthread #6820: linkmetrics not an option

purvisb, 2021-07-15:
I'm trying to use linkmetrics.

Commit ID: 3a33879 

Running command: `./script/build nrf52840 USB_trans DOT_THREAD_VERSION=1.2 LINK_METRICS=1`

I don't see link metrics as a help option. And it appears to be an invalid command.

Irving-cl, 2021-07-15:
Hi, the Link Metrics option has been split into two: The initiator and the subject. See: [etc/cmake/options.cmake:232](https://github.com/openthread/openthread/blob/main/etc/cmake/options.cmake#L232)

And note that, for `script/build`, we should add trailing cmake options (`-DXXX`) at the end. So you can change the command: 
```
./script/build nrf52840 USB_trans -DOT_THREAD_VERSION=1.2 -DOT_LINK_METRICS_INITIATOR=1 -DOT_LINK_METRICS_SUBJECT=1
```

purvisb, 2021-07-16:
Thank you. I did have to `git pull` for this to work.

closed 2021-07-16 by purvisb