openthread/openthread #6841: [mbedtls] Undefined references to mbedTLS write CSR methods

AuSchell, 2021-07-21:
Hello everyone

I'm currently updating a project which used an old version of OpenThread and i've run into a dead end. In my project, i make use of several methods from the x509write_csr module and since the update of the OpenThread libraries i get a "undefined reference" error from the linker for every method in that module. 

The commit id of the old OpenTread version is b08fab42bb4b09632a69ce980db41c53243185b7, the one i'm currently using is 3d10c90da499d1c01b2e752747e604091e4d476c. I'm using a nRF52840-DK board. 

I used the build command:
`make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 DISABLE_BUILTIN_MBEDTLS=0 DNS_CLIENT=1 DIAGNOSTIC=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MLR=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 COAPS=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 COAP_BLOCK=1 COAP_OBSERVE=1 COAPS=1 SRP_CLIENT=1 FULL_LOGS=1`

The mbedtls-config.h file has to be modified to enable those methods, which used to work with the old version. Here is what i have added to the file:
```
#ifdef MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED
#define MBEDTLS_BASE64_C
#define MBEDTLS_ECDH_C
#define MBEDTLS_ECDSA_C
#define MBEDTLS_OID_C
#define MBEDTLS_PEM_PARSE_C
#define MBEDTLS_X509_USE_C
#define MBEDTLS_X509_CRT_PARSE_C
#define MBEDTLS_X509_USE_C
#define MBEDTLS_X509_CHECK_KEY_USAGE
#define MBEDTLS_PKCS1_V15
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED
#endif
```

Here is the console output:
```
  BUILD    nrf52840
make -j2 -C build/nrf52840 --no-print-directory all
  CHECK         .local-version
make --no-print-directory all-recursive
Making all in include
make  all-am
Making all in third_party
Making all in NordicSemiconductor
make[4]: Für das Ziel „all“ ist nichts zu tun.
Making all in jlink
make[4]: Für das Ziel „all“ ist nichts zu tun.
Making all in mbedtls
make[4]: Für das Ziel „all“ ist nichts zu tun.
make[4]: Für das Ziel „all-am“ ist nichts zu tun.
Making all in src
Making all in core
make[4]: Für das Ziel „all“ ist nichts zu tun.
Making all in lib
Making all in hdlc
make[5]: Für das Ziel „all“ ist nichts zu tun.
Making all in platform
make[5]: Für das Ziel „all“ ist nichts zu tun.
Making all in spinel
make[5]: Für das Ziel „all“ ist nichts zu tun.
make[5]: Für das Ziel „all-am“ ist nichts zu tun.
Making all in cli
  CXX      libopenthread_cli_ftd_a-cli.o
  CXX      libopenthread_cli_mtd_a-cli.o
  AR       libopenthread-cli-mtd.a
  AR       libopenthread-cli-ftd.a
Making all in ncp
make[4]: Für das Ziel „all“ ist nichts zu tun.
make[4]: Für das Ziel „all-am“ ist nichts zu tun.
Making all in examples
Making all in platforms
Making all in utils
make[5]: Für das Ziel „all“ ist nichts zu tun.
Making all in k32w
make[5]: Für das Ziel „all“ ist nichts zu tun.
Making all in nrf528xx
make[5]: Für das Ziel „all“ ist nichts zu tun.
make[5]: Für das Ziel „all-am“ ist nichts zu tun.
Making all in apps
Making all in cli
  CCLD     ot-cli-mtd
  CCLD     ot-cli-ftd
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: ../../../src/cli/libopenthread-cli-mtd.a(libopenthread_cli_mtd_a-cli.o): in function `ot::Cli::Interpreter::ParseJoinerDiscerner(char*, otJoinerDiscerner&)':
<path-to-repo>/openthread/examples/../src/cli/cli.cpp:217: undefined reference to `mbedtls_x509write_csr_init'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:1729: ot-cli-mtd] Fehler 1
make[5]: *** Auf noch nicht beendete Prozesse wird gewartet …
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: ../../../src/cli/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): in function `ot::Cli::Interpreter::ParseJoinerDiscerner(char*, otJoinerDiscerner&)':
<path-to-repo>/openthread/examples/../src/cli/cli.cpp:217: undefined reference to `mbedtls_x509write_csr_init'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:1725: ot-cli-ftd] Fehler 1
make[4]: *** [Makefile:434: all-recursive] Fehler 1
make[3]: *** [Makefile:435: all-recursive] Fehler 1
make[2]: *** [Makefile:591: all-recursive] Fehler 1
make[1]: *** [Makefile:517: all] Fehler 2
make: *** [examples/Makefile-nrf52840:329: do-build-cortex-m4] Fehler 2
```

Did i do something wrong, have i forgotten something?

Thanks in advance.

jwhui, 2021-07-22:
@AuSchell , Nordic transitioned to using prebuilt mbedTLS libraries. You can try enabling the builtin mbedTLS build flag:
https://github.com/openthread/openthread/blob/3d10c90da499d1c01b2e752747e604091e4d476c/examples/Makefile-nrf52840#L44-L46

AuSchell, 2021-07-23:
@jwhui, thank you for your answer, but that is what i already did with the inclusion of the DISABLE_BUILTIN_MBEDTLS=0 flag in the make command.

AuSchell, 2021-07-26:
@jwhui I have found the problem: x509_create.c, x509_crl.h, x509_csr.c, x509write_crt.c and x509write_csr.c are not included in third_party/mbedtls/Makefile.am in the version i'm using. They must have been reincluded since then.

closed 2021-07-26 by AuSchell