openthread/openthread #6844: Thread Admin Password for OTBR

OLmmyT, 2021-07-22:
I have followed this codelab (https://openthread.io/codelabs/openthread-border-router#0) to setup an OpenThread Border router and an end device.

On an Android phone, I have installed Thread application and I would like to access to my OTBR. It requires a Thread Admin Password. Is there a default password or where can I find it? 
Can I find this info in Openthread Border Router Web GUI?

Thanks

jwhui, 2021-07-22:
The Thread Admin Password is also referred to as the Commissioning Passphrase.

When configuring the Operational Dataset, the Commissioning Passphrase may be set using the `dataset pskc -p` command. See [src/cli/README_DATASET.md#pskc](https://github.com/openthread/openthread/blob/main/src/cli/README_DATASET.md#pskc) for more details.

jwhui, 2021-08-11:
Closing stale issue.

closed 2021-08-11 by jwhui