openthread/openthread #6859: [github-actions] POSIX / expects-macos fails continuously

superwhd, 2021-07-27:
**Describe the bug**

I made a PR that only changed a test case slightly: https://github.com/openthread/openthread/pull/6858.

While running CI, I saw the following error in POSIX / expects-macos / Run Native IP Mode
```
Run brew install dnsmasq
==> Downloading https://ghcr.io/v2/homebrew/core/dnsmasq/manifests/2.85-1
==> Downloading https://ghcr.io/v2/homebrew/core/dnsmasq/blobs/sha256:945ab265756b63a2040d5bdcc4f8c2c24e379d60ed2e21249b77eade885630ff
==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:945ab265756b63a2040d5bdcc4f8c2c24e379d60ed2e21249b77eade885630ff?se=2021-07-27T03%3A15%3A00Z&sig=4MGk5dyNdy7ryh5NktUw0cEGMWG07LegbdinOjfdIcQ%3D&sp=r&spr=https&sr=b&sv=2019-12-12
==> Pouring dnsmasq--2.85.catalina.bottle.1.tar.gz
Error: Failed to install service files
==> Caveats
To restart dnsmasq after an upgrade:
  sudo brew services restart dnsmasq
Or, if you don't want/need a background service you can just run:
  /usr/local/opt/dnsmasq/sbin/dnsmasq --keep-in-foreground -C /usr/local/etc/dnsmasq.conf -7 /usr/local/etc/dnsmasq.d,*.conf
==> Summary
🍺  /usr/local/Cellar/dnsmasq/2.85: 9 files, 619.3KB
Error: Process completed with exit code 1.
```

logs: [logs_87485.zip](https://github.com/openthread/openthread/files/6882400/logs_87485.zip)

abtink, 2021-07-27:
Thanks @superwhd  for filing. 
I noticed the same thing. Seems like it started from this morning (Monday Jun 26 morning).

jwhui, 2021-07-27:
> Thanks @superwhd for filing.
> I noticed the same thing. Seems like it started from this morning (Monday Jun 26 morning).

Yes, the last successful run was about 11 hours ago.

BTW, I think you meant Jul :)

jwhui, 2021-07-27:
Looks like it is passing now: https://github.com/openthread/openthread/runs/3168137069?check_suite_focus=true

closed 2021-07-27 by jwhui