openthread/openthread #6860: Network Loss increases when number of nodes are more than 50 in the network

bhumikaajain, 2021-07-27:
In my open thread network, there are a total of 68 devices.
Among these 68 devices, there are 26 FTDs (Full thread devices) and 42 SEDs( Sleepy end devices).
Both FTDs and SEDs send data every minute to the Border Router (NCP).
FTDs send confirmable messages and SEDs send Non-confirmable messages.
If we take data of 24 hours, then we observe data loss in the range of 15-40% in the case of SEDs.
While in FTDs, data loss is around 0-5%.
Also, SEDs are placed such that their RSSI is under -80dbm
Is there any solution to increase the stability of network when the number of nodes in the network is more than 50?
How many hops can a non-confirmable packet take before it gets lost in the network?

jwhui, 2021-07-28:
@bhumikaajain , can you provide more information about your setup?

For example:
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. SDK (if any)

bhumikaajain, 2021-07-28:
Hi, 

Git Commit ID: https://github.com/openthread/openthread/commit/8a1992e2e42fb707babe9491a6a1456e553490e8
IEEE 802.15.4 hardware platform  : Nordic's nrf52840
SDK: NRF_SDK_for_thread_and_zigbee_4.0

jwhui, 2021-07-28:
@bhumikaajain , thanks for the additional information. Some things that have changed since your commit from 2019:

- https://github.com/openthread/openthread/blob/e8b085e09c12a9714d857b81a200dd89e16cee4f/src/core/config/mac.h#L73
- https://github.com/openthread/openthread/blob/e8b085e09c12a9714d857b81a200dd89e16cee4f/src/core/config/mle.h#L193
- https://github.com/openthread/openthread/blob/e8b085e09c12a9714d857b81a200dd89e16cee4f/src/core/config/tmf.h#L106

You may also try increasing this to 64:
- https://github.com/openthread/openthread/blob/e8b085e09c12a9714d857b81a200dd89e16cee4f/src/core/config/tmf.h#L45

If you experience any SEDs that drop off for long periods of time, you can also try enabling:
- https://github.com/openthread/openthread/blob/e8b085e09c12a9714d857b81a200dd89e16cee4f/src/core/config/child_supervision.h#L45

bhumikaajain, 2021-07-29:
@jwhui Thanks for the above suggestions. I will implement these changes in my stack.

In addition to the above changes, I have one more query, what exact roles do the below parameters play in networking:
1. kMinDowngradeNeighbors 
2. OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE
3. OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS

What is the idle value for the above parameters for long networks?

jwhui, 2021-07-29:
1. `kMinDowngradeNeighbors` is the number of neighbors a router must have in common with neighboring routers before it can consider downgrading back to an end device. This is part of Thread's router selection process.
2. `OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE` is an experimental feature that allows Thread networks to extend beyond several hops by increasing the cost metric from 4 bits to 8 bits. Note that this feature is not conformant with the Thread Specification.
3. `OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS` is the number of receive-only links an rx-on-when-idle child can maintain with neighboring routers. This is to help with receiving multicast messages, since the child can receive a given multicast transmission from multiple different routers.

bhumikaajain, 2021-07-30:
> @bhumikaajain , thanks for the additional information. Some things that have changed since your commit from 2019:
>
> * [...]

As per the changes suggested here, would these be enough to minimize the data losses in a large thread network?

The application where I am facing this issue is a client site so I can't do multiple DFUs over there and these SEDs are battery-powered, so in multiple DFUs there is battery constraint also.

I have also gone through an issue  [https://github.com/openthread/openthread/issues/4508](url) which suggested changing "kMinDowngradeNeighbors" from 7 to 4 and set "OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE" and "OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS" parameters. 
So finally what would be the best approach to reduce non-confirmable packet loss in larger networks where the number of Nodes reaches 70-80?

jwhui, 2021-07-30:
> As per the changes suggested here, would these be enough to minimize the data losses in a large thread network?

Without knowing the root cause, I can only guess at what might help your scenario.

> I have also gone through an issue [https://github.com/openthread/openthread/issues/4508](url) which suggested changing "kMinDowngradeNeighbors" from 7 to 4 and set "OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE" and "OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS" parameters.

The `LONG_ROUTES` feature is experimental and not conformant with the Thread specification. The `LONG_ROUTES` feature allows for having routes with larger number of hops in a Thread network, which has impact to other Thread protocol operations. Unless you are using the `LONG_ROUTES` feature, there should be no reason to change `kMinDowngradeNeighbors`.

bhumikaajain, 2021-07-31:
How many hops a thread non-conformal message can take before it gets lost?

And also "OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS" what impact does this parameter have in thread mesh?

jwhui, 2021-07-31:
> How many hops a thread non-conformal message can take before it gets lost?

Thread has a 4-bit route cost, which means that the maximum route cost for a given path is 15. Each hop can have link cost between 1 and 4. The `LONG_ROUTES` feature increases the route cost to 8 bits, allowing for routes that have more hops.

> And also "OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS" what impact does this parameter have in thread mesh?

I answered tried to provide a brief explanation in https://github.com/openthread/openthread/issues/6860#issuecomment-888811626 above.

jwhui, 2021-08-25:
Closing stale issue.

closed 2021-08-25 by jwhui