openthread/openthread #6864: Force OT End Device's router

farhangj, 2021-07-27:
Hi OpenThread devs,

Wondering if forcing an End Device's router (via ID or rloc) is an option in OT.

Basically I have a network with multiple routers (FTD) and multiple SEDs.
So far what I'm observing is that the SEDs join the first router available. When another router becomes available, they don't automatically join it, even though RSSI might be better with the new router. 

I do realize there's a [Periodic Parent Search](https://openthread.io/guides/build/features/periodic-parent-search) potentially available, but that does not satisfy our requirement.

If we have other information (not from/via Thread) that would lead to a preferred choice of router for a given SED, how can I enforce the SED to leave its current router and join the other one?

jwhui, 2021-07-27:
The Thread Specification does not provide a mechanism to restrict attaching to a specific parent/router.

OpenThread does provide a MAC filter, which can be used to filter packets to/from a given Extended Address. See [src/cli/README.md#macfilter](https://github.com/openthread/openthread/blob/main/src/cli/README.md#macfilter) for how to use it via the CLI.

Hope that helps.

farhangj, 2021-07-27:
Thanks, very helpful.

farhangj, 2021-07-27:
> Thanks, very helpful.

closed 2021-07-27 by farhangj