openthread/openthread #6873: otIp6Send sends only 4 Bytes of payload over the thread network and need to send more

DanielHailu3, 2021-07-29:
I am using Thread stack and using otIP6Send command to send and receive data over the Thread network . It is only sending 4 bytes over the Thread network and want to know how to increase the number of bytes or payload to 8 bytes or more.  I define the payload to be 20 bytes and it still only sends 4 bytes over the network. I just need advice on sending more data over the network.

I'm using nrf52840.

     bufferindata[0] = 0x48;
     bufferindata[1] = 0x45;
     bufferindata[2] = 0x4C;
     bufferindata[3] = 0x4C;
     bufferindata[4] = 0x4F;
     bufferindata[5] = 0x0D;
     bufferindata[6] = 0x00;
     bufferindata[7] = 0x00;
Code:
/**@brief Function for initializing the Thread Stack.
 */
static void thread_instance_init(void)
{
    thread_configuration_t thread_configuration =
    {
        .radio_mode        = THREAD_RADIO_MODE_RX_ON_WHEN_IDLE,
        .autocommissioning = false,
        .autostart_disable = false,
        .autocommissioning = true,
    };

    thread_init(&thread_configuration);
    thread_state_changed_callback_set(thread_state_changed_callback);

    uint32_t err_code = bsp_thread_init(thread_ot_instance_get());
    APP_ERROR_CHECK(err_code);

    otIp6SetReceiveFilterEnabled(thread_ot_instance_get(), true);
    otIp6SetReceiveCallback(thread_ot_instance_get(), ot_receive_handler, NULL);
   

}

uint32_t* contructip6add (uint16_t pay_len)
{
   static uint32_t header [10];
    header[0]=lwip_htonl((((uint32_t)ver)<<28)|(((uint32_t)TRAFFIC_CLASS)<<20)|FLOW_LABEL);
    header[1]=lwip_htonl((((uint32_t)pay_len)<<16)|(((uint32_t)NEXT_HEADER)<<8)|HOP_LIMIT);
    header[2]=(0xADDEFD);
    header[3]=(0xEFBE2);
    header[4]=(0xFF000000);
    header[5]=(0x00FC0FE);
    header[6]=lwip_htonl(0xFF030000);
    header[7]=lwip_htonl(0x00000000);
    header[8]=lwip_htonl(0x00000000);
    header[9]=lwip_htonl(0x00000001);

    return header;
}
static void ip_packet_transmit_now(uint8_t *bufferindata)
{
      uint8_t err;
      uint8_t* header5;
    
      header5= (uint8_t*)contructip6add(20);
      otMessage *amessage;
      amessage=otIp6NewMessage(thread_ot_instance_get(), false); 
     
      if (amessage != NULL)
      { 

        err=otMessageAppend(amessage, header5, 40);

        err=otMessageAppend(amessage, bufferindata, 20);
        

        err=otIp6Send(thread_ot_instance_get(),amessage);

        if(err == OT_ERROR_NONE)
        {
           printf("Sent successfully!");
        }
      } 
}

static void ot_receive_handler(otMessage *aMessage, void *aContext)
{
    // Read the message itself and output it to serial
    uint16_t length = otMessageGetLength(aMessage);
    uint8_t *data_read = malloc(length);

    uint32_t bytes_read = otMessageRead(aMessage,0,data_read,length);
    
        
    if (bytes_read > 0)
    {
      print_data(&data_read[40]);
          
    }
 
    free(data_read);
    otMessageFree(aMessage);
    
    
}
It is sending data but only 4 Bytes with header from one node to the other nodes on the network.

jwhui, 2021-07-29:
I don't see anything immediately obvious yet.

What value is `otMessageGetLength()` in `ot_receive_handler()` returning?

Is it possible for you to provide OpenThread logs for greater visibility?

Also, can you provide more information on your setup:
1. Git commit id
2. IEEE 802.15.4 hardware platform

DanielHailu3, 2021-07-29:
otMessageGetLength() in ot_receive_handler() returning is returning 0x3C or 60 bytes for 40 bytes in the header and 20 bytes in the data.  I am using the nRF52840 DK evaluation hardware board.   I modified the code in nRF5 SDK for Thread and Zigbee v4.1.0.  I used the https://github.com/openthread/openthread/tree/main/src/cli .  Do you know of parameter to set or why only 4 bytes are being sent over the thread network?

jwhui, 2021-07-29:
`otMessageGetLength()` returning 60 indicates that the receive message is 60 bytes in length.

Can you provide more detail in how you are determining that only 4 bytes are being sent over the Thread network?

DanielHailu3, 2021-07-29:
The header part of the packet is the 40 bytes seems the same as the header I sent. Only 4 Bytes are valid data transmitted and received, which I put in the array to send. The rest of the 16 bytes are random values and zeros and not the 20 bytes of data in my array.  If send out [0x48, 0x45, 0x4C, 0x4C, 0x4F, 0x0D, 0x0A,0x00] and 16 zeros, I receive [0x48, 0x45, 0x4C, 0x4C, 0x00, 0x0A, 0x00,0x00] and other 16 random values. I only send and receive 4 valid bytes in the packet.

jwhui, 2021-07-29:
Above, I see:
```
static void ip_packet_transmit_now(uint8_t bufferindata)
```
and
```
err=otMessageAppend(amessage, bufferindata, 20);
```
Note that `bufferindata` has type `uint8_t`. Is that what you intended?

DanielHailu3, 2021-07-29:
It is actually 

`static void ip_packet_transmit_now(uint8_t *bufferindata)` 

in the code.

DanielHailu3, 2021-07-29:
When I only want to send 8 bytes I set the message length to 10.  Here is the code for that:
`static void ip_packet_transmit_now(uint8_t *bufferindata)
{
      uint8_t err;
      uint8_t* header5;
    
      header5= (uint8_t*)contructip6add(10);
      otMessage *amessage;
      amessage=otIp6NewMessage(thread_ot_instance_get(), false); 

      
      if (amessage != NULL)
      { 

        err=otMessageAppend(amessage, header5, 40);

        err=otMessageAppend(amessage, bufferindata, 10);
        

        err=otIp6Send(thread_ot_instance_get(),amessage);

        if(err == OT_ERROR_NONE)
        {
           printf("Sent successfully!");
        }
      } 
}
`

jwhui, 2021-07-29:
> The Github editor removed the pointer part.

For better formatting, you can use [code block formatting](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks).

jwhui, 2021-07-29:
One thing to test: if you send to link-local all nodes `ff02::1`, rather than realm-local all nodes `ff03::1`, are the bytes transmitted as expected?

DanielHailu3, 2021-07-29:
I send 
`[0x48,0x45, 0x4C, 0x4C, 0x4F, 0x0D, 0x0A, 0x00] ` 
and I get on the other node:
`[0x48,0x45, 0x4C, 0x4C,0x00,0x0A,0x0A,0x00]`
When using `ff02::1`. Is there a parameter I am missing ? Any suggestions?

jwhui, 2021-07-29:
I tried doing the same using the latest main branch (866c9602fb8f8611204823077001ab571b7b43a0), but all the payload bytes are being communicated properly from the sender to receiver.

Are you able to provide a packet sniffer capture? At least that can help determine whether it is an issue on the sender or receiver.

DanielHailu3, 2021-07-29:
Quick question: on the receive side does the data array have to be 32-bit or does byte array work?
`static void ot_receive_handler(otMessage *aMessage, void *aContext)
{
    uint16_t length = otMessageGetLength(aMessage);
    uint32_t data_read[length];
  
    uint32_t bytes_read=otMessageRead (aMessage,0,data_read,length);
   `

jwhui, 2021-07-29:
> Quick question: on the receive side does the data array have to be 32-bit or does byte array work?

A byte array should work.

DanielHailu3, 2021-07-29:
What tool do you use for the packet sniffer?

jwhui, 2021-07-29:
> What tool do you use for the packet sniffer?

See:

- [Packet Sniffing with Pyspinel](https://openthread.io/guides/pyspinel/sniffer)
- [Packet Sniffing using Extcap](https://openthread.io/guides/pyspinel/sniffer-extcap)

DanielHailu3, 2021-07-29:
Can you show me the code functions you used to send and receive that had payload bytes are being communicated properly ?

jwhui, 2021-07-29:
Here is the (really hacky) test code I used (diff from 866c9602fb8f8611204823077001ab571b7b43a0). I repurposed the `ping` command for the test. Execute the `ping` command on both devices to see the output.

```c++

diff --git a/src/cli/cli.cpp b/src/cli/cli.cpp
index 60391bb6c..c3dcf3a35 100644
--- a/src/cli/cli.cpp
+++ b/src/cli/cli.cpp
@@ -3103,9 +3103,69 @@ void Interpreter::HandlePingStatistics(const otPingSenderStatistics *aStatistics
     OutputResult(OT_ERROR_NONE);
 }
 
+static void ot_receive_handler(otMessage *aMessage, void *)
+{
+    // Read the message itself and output it to serial
+    uint16_t length = otMessageGetLength(aMessage);
+    uint8_t  data_read[100];
+
+    uint32_t bytes_read = otMessageRead(aMessage, 0, data_read, length);
+    printf("%d %d\r\n", bytes_read, length);
+
+    for (uint16_t i = 0; i < length; i++)
+    {
+        printf("%02x\r\n", data_read[i]);
+    }
+}
+
 otError Interpreter::ProcessPing(Arg aArgs[])
 {
-    otError            error = OT_ERROR_NONE;
+    otError error;
+
+    otIp6SetReceiveFilterEnabled(mInstance, true);
+    otIp6SetReceiveCallback(mInstance, ot_receive_handler, NULL);
+
+    OT_UNUSED_VARIABLE(aArgs);
+
+    static uint32_t header[10];
+    uint16_t        pay_len = 20;
+
+    uint8_t ver         = 6;
+    uint8_t HOP_LIMIT   = 64;
+    uint8_t NEXT_HEADER = 17;
+
+#define TRAFFIC_CLASS 0
+#define FLOW_LABEL 0
+
+    header[0] = htonl((((uint32_t)ver) << 28) | (((uint32_t)TRAFFIC_CLASS) << 20) | FLOW_LABEL);
+    header[1] = htonl((((uint32_t)pay_len) << 16) | (((uint32_t)NEXT_HEADER) << 8) | HOP_LIMIT);
+    header[2] = (0xADDEFD);
+    header[3] = (0xEFBE2);
+    header[4] = (0xFF000000);
+    header[5] = (0x00FC0FE);
+    header[6] = htonl(0xFF030000);
+    header[7] = htonl(0x00000000);
+    header[8] = htonl(0x00000000);
+    header[9] = htonl(0x00000001);
+
+    static uint8_t payload[20];
+
+    for (size_t i = 0; i < sizeof(payload); i++)
+    {
+        payload[i] = i;
+    }
+
+    otMessage *message;
+
+    message = otIp6NewMessage(mInstance, NULL);
+    VerifyOrExit(message != NULL, error = OT_ERROR_NO_BUFS);
+
+    SuccessOrExit(error = otMessageAppend(message, header, sizeof(header)));
+    SuccessOrExit(error = otMessageAppend(message, payload, sizeof(payload)));
+    SuccessOrExit(error = otIp6Send(mInstance, message));
+
+#if 0
+  otError            error = OT_ERROR_NONE;
     otPingSenderConfig config;
 
     if (aArgs[0] == "stop")
@@ -3180,6 +3240,7 @@ otError Interpreter::ProcessPing(Arg aArgs[])
     config.mCallbackContext    = this;
 
     error = otPingSenderPing(mInstance, &config);
+#endif
 
 exit:
     return error;
```

DanielHailu3, 2021-08-03:
Do you have the function htonl?
`header[1] = htonl((((uint32_t)pay_len) << 16) | (((uint32_t)NEXT_HEADER) << 8) | HOP_LIMIT);`

DanielHailu3, 2021-08-04:
@jwhui I did not find the `htonl` function in the OpenThread code.  Can you tell me the definition of it you used?

closed 2021-08-06 by DanielHailu3