openthread/openthread #6882: Remove Joiner entry by Commissioner failed

LancerSky, 2021-08-02:
Hi,
I use commission command to attach a Thread devices to the Thread network, but when I remove it, It prompts NotFound. 
Did I make a mistake? Can I remove the specified Node? Thanks.
`> commissioner start
Commissioner: petitioning
Done
> Commissioner: active

> commissioner joiner add f4ce360364cc89c2 J01NME
Done
> ~ Discovery Request from 221ea152b51c6c0a: version=3,joiner=1
Commissioner: Joiner start db3756b49bf1d004
Commissioner: Joiner connect db3756b49bf1d004
Commissioner: Joiner finalize db3756b49bf1d004
Commissioner: Joiner end db3756b49bf1d004
Commissioner: Joiner remove db3756b49bf1d004

> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 38 | 0x9800 |       63 |         0 |     3 |      3 |  96 | 7ea3d319de81e3e2 |    1 |
| 54 | 0xd800 |       63 |         0 |     0 |      0 |   0 | 0000000000000000 |    0 |

Done
> commissioner joiner remove f4ce360364cc89c2
Error 23: NotFound
> commissioner joiner remove db3756b49bf1d004
Error 23: NotFound`

jwhui, 2021-08-25:
@LancerSky , this is expected behavior. Once a joiner has successfully been commissioned, the joiner is automatically removed. This is indicated by the following log line:
```
Commissioner: Joiner remove db3756b49bf1d004
```

Note that the `db3756b49bf1d004` is the Joiner ID, which is a hash of the IEEE EUI-64.

LancerSky, 2021-09-06:
Thanks for your reply. So is the Thread protocol not as like Zigbee can remove sub-device by manual? We may have a batch of joiner commissioning at the same time. There may be other devices that are not ours. Is this safe?

jwhui, 2021-09-10:
> Thanks for your reply. So is the Thread protocol not as like Zigbee can remove sub-device by manual? We may have a batch of joiner commissioning at the same time. There may be other devices that are not ours. Is this safe?

The `joiner remove` command simply removes the joiner entry from the Commissioner. It does not remove the device from the Thread network.

Also note that Thread Commissioning relies on a device-specific pairing code. A Commissioner cannot join arbitrary devices - only those that have matching pairing codes in its table.

LancerSky, 2021-09-10:
Ok. Thanks.

closed 2021-09-10 by LancerSky