openthread/openthread #6884: MLE advertisement with incorrect partition id observed when forming a network

schinchani, 2021-08-02:
When running the Thread Test Harness Leader test cases, for any vendor running the OT stack (like Nordic, Silicon Labs), an MLE advertisement with incorrect partition ID is observed when forming the network, right at startup. 

Note: While it is just one advertisement packet observed at startup, it is reproducible and is seen every time you form the network. On rare occasions, we have also observed the leader node using this incorrect partition ID when responding to a parent request of an attaching node in turn causing the attach process to fail.

To reproduce the issue, simply execute the following set of commands:

factoryreset
dataset channel 11
dataset panid 0xface
dataset extpanid 000db80000000000
dataset networkname GRL
dataset networkkey 00112233445566778899aabbccddeeff
dataset meshlocalprefix FD00:0DB8:0000:0000::
dataset pskc 00000000000000000000000000000000
dataset activetimestamp 1
dataset  commit active
ifconfig up
thread start

I am not running the `dataset init new` or `dataset clear` commands as the TH does not run these commands.

[5_1_1_Leader_2021_08_02_12_19.log](https://github.com/openthread/openthread/files/6918991/5_1_1_Leader_2021_08_02_12_19.log)
[ThreadConsoleLog_2021_08_02_12_16.log](https://github.com/openthread/openthread/files/6918992/ThreadConsoleLog_2021_08_02_12_16.log)
[image]

schinchani, 2021-08-04:
@jwhui  any thoughts here?

abtink, 2021-08-09:
@schinchani I tried to see if I can produce this (with simulation platform) and additional logs, but I don't see the issue happening. 

Here is branch where I added additional logs (e.g., we log the partition ID when appending "Leader Data TLV" to a message which is included in the MLE Advertisement) ->  https://github.com/abtink/openthread/tree/test/issue6884

The new logs start with `#######`.

Here in the CLI and log output (when issuing same CLI commands as you listed):
```
abtink@ ~/GitHub/openthread (test/issue6884) $ ./examples/apps/cli/ot-cli-ftd 1
[INFO]-CORE----: Notifier: StateChanged (0x100fc300) [KeySeqCntr NetData Channel PanId NetName ExtPanId NetworkKey PSKc ActDset]
[INFO]-MLE-----: [announce-sender] StartingChannel:11

> 
> factoryreset

[INFO]-CORE----: Notifier: StateChanged (0x00038200) [NetData PanId NetName ExtPanId]

> dataset channel 11
Done
> dataset panid 0xface
Done
> dataset extpanid 000db80000000000
Done
> dataset networkname GRL
Done
> dataset networkkey 00112233445566778899aabbccddeeff
Done
> dataset meshlocalprefix FD00:0DB8:0000:0000::
Done
> dataset pskc 00000000000000000000000000000000
Done
> dataset activetimestamp 1
Done
> dataset  commit active
[INFO]-CORE----: [settings] Saved ActiveDataset
[INFO]-MESH-CP-: Active dataset set
Done
> 
[INFO]-CORE----: Notifier: StateChanged (0x100fc100) [KeySeqCntr Channel PanId NetName ExtPanId NetworkKey PSKc ActDset]
[INFO]-MLE-----: [announce-sender] StartingChannel:11

> ifconfig up
Done
> [INFO]-CORE----: Notifier: StateChanged (0x01001009) [Ip6+ LLAddr Ip6Mult+ NetifState]

 
> thread start
[NOTE]-MLE-----: Role Disabled -> Detached
[INFO]-MLE-----: AttachState Idle -> Start
Done
>
[INFO]-CORE----: Notifier: StateChanged (0x10001015) [Ip6+ Role MLAddr Ip6Mult+ ActDset]
[INFO]-MLE-----: [announce-sender] Stopped
[NOTE]-MLE-----: Attempt to attach - attempt 1, any-partition reattaching with Active Dataset
[INFO]-MLE-----: AttachState Start -> ParentReqRouters
[INFO]-MLE-----: Send Parent Request to routers (ff02:0:0:0:0:0:0:2)
[INFO]-MAC-----: Sent IPv6 UDP msg, len:84, chksum:d9ba, to:0xffff, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:54dd:24db:8ca8:aabd]:19788
[INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
[INFO]-MLE-----: AttachState ParentReqRouters -> ParentReqReeds
[INFO]-MLE-----: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
[INFO]-MAC-----: Sent IPv6 UDP msg, len:84, chksum:8be4, to:0xffff, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:54dd:24db:8ca8:aabd]:19788
[INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:2]:19788
[INFO]-MLE-----: AttachState ParentReqReeds -> Idle
[NOTE]-MLE-----: Allocate router id 56
[NOTE]-MLE-----: RLOC16 fffe -> e000
[NOTE]-MLE-----: Role Detached -> Leader
[NOTE]-MLE-----: Leader partition id 0x7b3c1e97
[INFO]-CORE----: Notifier: StateChanged (0x100012a5) [Ip6+ Role Rloc+ PartitionId NetData Ip6Mult+ ActDset]
[... 29 lines cut ...]
[NOTE]-MLE-----: ####### Mle::AppendLeaderData() -> leaderDataTlv.GetPartitionId()=0x7b3c1e97
[NOTE]-MLE-----: ####### MleRouter::SendAdvertisement()
[INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
[INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:d813, to:0xffff, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:54dd:24db:8ca8:aabd]:19788
[INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
[NOTE]-MLE-----: ####### Mle::AppendLeaderData() -> mLeaderDatat.GetPartitionId() =0x7b3c1e97
[NOTE]-MLE-----: ####### Mle::AppendLeaderData() -> leaderDataTlv.GetPartitionId()=0x7b3c1e97
[NOTE]-MLE-----: ####### MleRouter::SendAdvertisement()
[INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
[INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:792e, to:0xffff, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:54dd:24db:8ca8:aabd]:19788
[INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
```

-------

This all seems to be fine. All the Advertisement are sent after
```
[NOTE]-MLE-----: Role Detached -> Leader
[NOTE]-MLE-----: Leader partition id 0x7b3c1e97
```
and they seem to all include the same partition ID (`0x7b3c1e97`)

Can you please provide more details? It may be helpful to use the branch above with additional logs (and enabling log level INFO) to see if can be reproduced with the logs on the device? Thanks.

schinchani, 2021-08-09:
Hi @abtink  - Thank you for this update. If I recollect the issue correctly, the incorrect advertisement is observed in the packet capture (both in Wireshark - used by the GRL Test harness as well as in Simplicity Studio). Not sure if I saw it on the console logs. Let me try running it again and get back to you.

schinchani, 2021-08-25:
@abtink  - I was wondering if you got a chance to look at my last comment above? Thanks.

abtink, 2021-08-25:
@schinchani , thanks for the note. Somehow I missed your earlier comment. I read it now.

It may be a harness issue though I cannot think of how.  I am not myself very familiar with harness (I am a lot more familiar with  OT core and its modules) so not sure if I can be of much help.

One thing that I can suggest is if you can run the test (with harness?) on a build with additional logs enabled (may be with the branch I posted in earlier comment which adds even more logs around "Sending of Adv" and "preparing the Partition ID Tlv"). it may help shed some light on it.

schinchani, 2021-08-25:
Hi Abtin, Thank you for a quick reply. I am sailing in the same boat and not sure how the test harness introduces it (but looks like it probably does, given that it is seen on the TH Wireshark captures, but not on a local run). Capturing detailed logs from a specific test harness node has been a tricky situation in the past, especially as the Test harness controls the logs from multiple nodes at once. We can try playing with the Test harness and see if we can find more info.

jwhui, 2021-11-06:
@schinchani , any updates on this issue?

simonlingoogle, 2021-11-06:
[image]

This looks like a typical dirty traffic, meaning the `MLE Advertisement` with another Partition ID was actually sent by another device besides the test devices. One clue is that the `MAC Sequence Number` is not in order with other messages of the same device (see attachment). Since it's common for multiple devices to have the same MAC address because Test Harness will configure similar MAC addresses.

jwhui, 2021-12-22:
Closing stale issue.

closed 2021-12-22 by jwhui