openthread/openthread #6890: Thread network is abnormal because the MTD sends packets constantly

taoyinzhou, 2021-08-04:
**Describe the bug** A clear and concise description of what the bug is.

Hi All, 
    When I was testing the OTBR connection with multiple MTD, I found that sometimes some MTD would continuously and quickly send packets in the same format as the heartbeat packets, which seemed very abnormal. 
    I think this affects the normal communication between OTBR and other devices. Can you see from the captured packets why these packets are constantly sent? If I want to change the code logic to stop MTD from sending heartbeat packets rapidly, what should I do?
[image]

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
otbr:cf3a67383e9e25ca89208234d8efd8bee6b92f25

2. IEEE 802.15.4 hardware platform
   NRF52833
3. Build steps

4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.
[dc01-ERR.zip](https://github.com/openthread/openthread/files/6929214/dc01-ERR.zip)

abtink, 2021-08-04:
@taoyinzhou these are 802.15.4 Data Request frame or aka "data poll" frames. 

A sleepy child will periodically send data poll to its parent to check whether the parent has any queued message for it. The parent in the ack indicates whether it has any pending frame for the child. 

The poll period can be set by the application layer (please see [`otLinkSetPollPeriod`](https://openthread.io/reference/group/api-link-link#otlinksetpollperiod)).  You can also get/set it through CLI `pollperiod` command. 

I would suggest first checking what is the poll period being used on your devices.

taoyinzhou, 2021-08-05:
@abtink 
    Thank you for you quickly response.
    I think you're right. My PollPeriod is 100ms，so there are 10 of these packets per second.
    However, I have a question about your mentioned "A sleepy child will periodically send data poll to its parent to check whether the parent has any queued message for it". Does it mean that  when there are Internet packets sent to SED, OTBR will save those packets and not send them to SED until SED comes to inquire?
    If this is the case, how to increase the cache of the OTBR if hundreds of sub devices are to be connected in a single OTBR? I am using the RCP architecture.

jwhui, 2021-08-05:
> I think you're right. My PollPeriod is 100ms，so there are 10 of these packets per second.
> However, I have a question about your mentioned "A sleepy child will periodically send data poll to its [...]

Thread 1.1 requires the SED to poll its parent by sending an IEEE 802.15.4 Data Request message to receive messages. SED's Parent to buffer messages destined for the SED.

> If this is the case, how to increase the cache of the OTBR if hundreds of sub devices are to be connected in a single OTBR? I am using the RCP architecture.

See the following config that specifies the number of message buffers: https://github.com/openthread/openthread/blob/16c207825ff9a004ebdbd64e90af45267d078810/src/core/config/openthread-core-default-config.h#L150

jwhui, 2021-08-24:
Closing stale issue.

closed 2021-08-24 by jwhui