openthread/openthread #6912: Border Router Don't receive any data though network is up

bhumikaajain, 2021-08-10:
Hi,
In my open thread network, there are 3 types of nodes:

SED (Sleepy end devices): They are battery powered and the polling period is 30 seconds.
FTD (Full thread devices): They act as routers for SEDs. They have no power consumption restriction.
NCP: Which is attached to RPi based Border router. It receives all data packets from SEDs and FTDs.
Following are some additional details:
Git Commit ID: 8a1992e
IEEE 802.15.4 hardware platform : Nordic's nrf52840
SDK: NRF_SDK_for_thread_and_zigbee_4.0

Since SEDs are battery-powered, they send non-confirmable data packets every minute and sleep.

In SED, we have added a feature when the border router is not visible to SED (due to partitioning or border router is offline/powered off), it doesn't send data to border router instead it starts buffering data with itself.

Also, we have 2 counters in SED, one uptime counter and one packet_counter.
Uptime counter increases once every time SED records a data point, and packet_counter increases every time SED sends a packet to the border router. So if the Border router is not visible to SED, and it is recording data, its uptime will increase and it will keep on buffering data and once the border router is back, it will send that data.

My open thread setup consists of 76 nodes, in which there are 28 FTDs, and 48 SEDs. In this setup, sometimes, all Devices stop sending data to the border router all of a sudden while the NCP state is working. When I run "sudo wpanctl get" command during this time, the UID of devices are '0000000000000000" but the thread network is up. It shows multiple devices in the connected state, but their UIDs are not present. 
And when devices again start sending data, and everything is fine on the network, these SEDs send their buffer data along with the live data. 
Now my question is that what causes border-router to become invisible for SEDs, and all devices at once and again they get back to normal, sometime after 4 hours, sometimes after 2 hours. 
Also, Network reboots don't help in this case.
The issue is inconsistent and can't be reproduced.

jwhui, 2021-08-10:
> Following are some additional details:
> Git Commit ID: [8a1992e](https://github.com/openthread/openthread/commit/8a1992e2e42fb707babe9491a6a1456e553490e8)

Note that this commit is quite old (December 2019).

> In SED, we have added a feature when the border router is not visible to SED (due to partitioning or border router is offline/powered off), it doesn't send data to border router instead it starts buffering data with itself.

Can you provide more information on how the SED determines that the border router is not visible?

> My open thread setup consists of 76 nodes, in which there are 28 FTDs, and 48 SEDs. In this setup, sometimes, all Devices stop sending data to the border router all of a sudden while the NCP state [...]

"UID" is not a term that is used in Thread or OpenThread. Can you provide more information on what the "UID" is?

> Now my question is that what causes border-router to become invisible for SEDs, and all devices at once and again they get back to normal, sometime after 4 hours, sometimes after 2 hours.
> Also, Network reboots don't help in this case.
> The issue is inconsistent and can't be reproduced.

Unfortunately, it is difficult to determine the issue without more visibility. Can you provide more information, such as neighbor / route / child tables, OpenThread logs, etc?

bhumikaajain, 2021-08-11:
for Border router, I am using following function:

> static bool isOtBrConnected(void) {
    otBorderRouterConfig borderRouterConfig;
    otNetworkDataIterator iter = OT_NETWORK_DATA_ITERATOR_INIT;
    if (otNetDataGetNextOnMeshPrefix(thread_ot_instance_get(), &iter, &borderRouterConfig) == OT_ERROR_NONE)
        return true;
    else
        return false;
}

This is the current router table of that site:

> Thread:RouterTable = [
	"0000000000000000, RLOC16:0000, RouterId:0, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:4, LinkEst:no"
	"0000000000000000, RLOC16:0400, RouterId:1, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:3, LinkEst:no"
	"0000000000000000, RLOC16:0800, RouterId:2, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:1400, RouterId:5, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:3, LinkEst:no"
	"0000000000000000, RLOC16:1800, RouterId:6, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:1c00, RouterId:7, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:2000, RouterId:8, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:2800, RouterId:10, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:3c00, RouterId:15, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:5000, RouterId:20, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:6400, RouterId:25, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:7000, RouterId:28, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:7400, RouterId:29, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:7c00, RouterId:31, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:1, LinkEst:no"
	"0000000000000000, RLOC16:8800, RouterId:34, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:9400, RouterId:37, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:9800, RouterId:38, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:ac00, RouterId:43, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"0000000000000000, RLOC16:b000, RouterId:44, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"98AABD96E411C5B5, RLOC16:c000, RouterId:48, NextHop:63, PathCost:0, LQIn:2, LQOut:1, Age:4, LinkEst:yes"
	"960BC63045C73532, RLOC16:c400, RouterId:49, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:2, LinkEst:yes"
	"0000000000000000, RLOC16:cc00, RouterId:51, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:128, LinkEst:no"
	"237F210EB791650E, RLOC16:d800, RouterId:54, NextHop:8, PathCost:3, LQIn:3, LQOut:2, Age:0, LinkEst:yes"
	"0000000000000000, RLOC16:ec00, RouterId:59, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:3, LinkEst:no"
]

By UID I mean EUID, which is visible in router table.

jwhui, 2021-08-11:
Thanks for the additional information.

The first field in the router table is the IEEE 802.15.4 Extended Address (not the same as EUI-64). A Thread router only maintains the Extended Address of direct neighbors that it has established neighbor links with. In this case, most of the enries do not have a direct link established.

Over the past ~2 years, we've made a number of enhancements to make the publishing of network data more reliable. I would strongly encourage you to use a newer commit.

bhumikaajain, 2021-08-11:
@jwhui  Thanks for the update. We are working on porting our application to the latest Nrf SDK and open thread commit. But this will take some time at our end. 

In the meantime, we have done further analysis for this issue. NCP on this site is switching between Router and Child state. 
We suspect that when this issue occurs, NCP goes into partitioning mode when it is in a child state. 

If SEDs and FTDs partition themselves from the network, who tries harder to come out of partitioning mode?
If FTD is more likely to rejoin first, then is there any method through which I can prevent my NCP to go into SED state.

jwhui, 2021-08-11:
> @jwhui Thanks for the update. We are working on porting our application to the latest Nrf SDK and open thread commit. But this will take some time at our end.

Great to hear.

> In the meantime, we have done further analysis for this issue. NCP on this site is switching between Router and Child state.
> We suspect that when this issue occurs, NCP goes into partitioning mode when it is in a child state.

Yes, this was an old bug that was fixed in #4425. If you search for "network-data" in the commits, you'll see many fixes/enhancements in the past ~2 years.

> If SEDs and FTDs partition themselves from the network, who tries harder to come out of partitioning mode?

SEDs cannot form their own partition - they can only join FTDs.

> If FTD is more likely to rejoin first, then is there any method through which I can prevent my NCP to go into SED state.

An FTD does not automatically go into SED state, since the rx-on-when-idle mode must be changed explicitly by the user.

bhumikaajain, 2021-08-11:
In the last message, by SED I mean Child state, and FTD I mean router state. 
I have not explicitly made NCP as FTD or SED.
So What I really mean to ask is when an  NCP in a child state, partition from the network, how fast can it recover?
And if NCP alone partitions, to join back, which state is more efficient, child or router?
Last, is there any method that can prevent NCP to enter the child state?

jwhui, 2021-08-11:
> So What I really mean to ask is when an NCP in a child state, partition from the network, how fast can it recover?

Thread devices learn about neighboring partitions by receiving MLE Advertisements. Routers send MLE Advertisements with a period of ~30 seconds or less. REEDs send MLE Advertisements with a period ~10 minutes.

> And if NCP alone partitions, to join back, which state is more efficient, child or router?

The partition merging process is the same for both child and router. A device migrating to a new partition must always attach first as a child.

> Last, is there any method that can prevent NCP to enter the child state?

Thread requires all devices to initially attach as a child. The one exception is the leader, but that only occurs when forming a new partition.

bhumikaajain, 2021-08-12:
Thanks for the reply. Now my major task is to port the application to the latest nrf_SDK and the latest open thread.
I was working on it yesterday, Latest Nrf_SDK is Nrf_SDK_For_thread_and_Zigbee_4.1.0. It was released on week 18 of 2020.
As per the nordic info center, the process to build the latest open thread libraries is defined in this [https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/thread_ot_libraries_building.html](url).

But when I took the latest open thread stack in my system and followed the command, I found that "examples/Makefile-nrf52840" has been removed.
Can you guide me to the latest process to build open thread libraries for nrf52840 ?

jwhui, 2021-08-12:
> But when I took the latest open thread stack in my system and followed the command, I found that "examples/Makefile-nrf52840" has been removed.
> Can you guide me to the latest process to build open thread libraries for nrf52840 ?

See https://github.com/openthread/ot-nrf528xx

jwhui, 2021-08-25:
Closing stale issue.

closed 2021-08-25 by jwhui