openthread/openthread #6929: ot-daemon in Ubuntu Host, fails to connect to RCP device(nrf528xx)

RoyAnupam, 2021-08-16:
Hi,
I am not able to somehow connect to my RCP device from ot-daemon running in the Host.  I have followed following steps.

**Environment:** 
   a) HostOS: Ubuntu 18.04
   b) RCP Device: nRF52840 USB dongle

**Build**
   a) Built ot-rcp.hex for my nrf52840 USB dongle by following steps given in [this](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md). 
  b) OT daemon: Built by following [this](https://openthread.io/platforms/co-processor/ot-daemon) 
  c) Run OT daemon in Ubuntu: 
              $./output/posix/bin/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM1?uart-baudrate=115200'
              $ sudo ./output/posix/bin/ot-ctl
                 **connect session failed: No such file or directory**

 Note: my RCP device is enumerated in /dev/ttyACM1. Also, I am able to run OTBR docker image on the same RCP device from Ubuntu Host  & form a THREAD network with no problem. So, it apparently seems, no issue with the RCP device.

Am I missing something basic? Any guidance/insights will be appreciated. Thank You

BR,

Irving-cl, 2021-08-16:
Hi @RoyAnupam,

it looks like the `ot-daemon` didn't start successfully. What's the output of `$./output/posix/bin/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM1?uart-baudrate=115200'`? Did it exit directly?

RoyAnupam, 2021-08-16:
> Hi @RoyAnupam,
> 
> it looks like the `ot-daemon` didn't start successfully. What's the output of `$./output/posix/bin/ot-daemon 'spinel+hdlc+uart:///dev/ttyACM1?uart-baudrate=115200'`? Did it exit directly?

Hi @Irving-cl , Actually, output of the command is *Nothing. So, yes, it seems to be fail to launch completely.  I am wondering how to get this running..

Irving-cl, 2021-08-16:
Hi @RoyAnupam,

could you try the guide [here](https://github.com/openthread/openthread/blob/main/src/posix/README.md) instead?

And I think you can find some logs from syslog on ubuntu. Could you also try `cat /var/log/syslog | grep ot-daemon` and check the output?

RoyAnupam, 2021-08-16:
Hi @Irving-cl 

> And I think you can find some logs from syslog on ubuntu. Could you also try cat /var/log/syslog | grep ot-daemon and check the output?

By doing a quick check, it is spitting below error.

 ./ot-daemon[17887]: Running OPENTHREAD/20200818-01160-gb42261f88; POSIX; Aug 16 2021 15:43:41
./ot-daemon[17887]: Thread version: 3
 ./ot-daemon[17887]: [CRIT]-PLAT----: RCP API Version 3 is not in the supported range [4-4]
 ./ot-daemon[17887]: [CRIT]-PLAT----: CheckRcpApiVersion() at /<openthread project>/openthread/src/posix/../../src/lib/spinel/radio_spinel_impl.hpp:414: RadioSpinelIncompatible

jwhui, 2021-08-16:
There was a recent change to the spinel protocol that requires updating the RCP (https://github.com/openthread/openthread/commit/a94e88cd5081148a8bcc99c5123337b94612f995).

Can you try building a new RCP with the latest main branch of https://github.com/openthread/ot-nrf528xx ?

RoyAnupam, 2021-08-16:
> There was a recent change to the spinel protocol that requires updating the RCP ([a94e88c](https://github.com/openthread/openthread/commit/a94e88cd5081148a8bcc99c5123337b94612f995)).
> 
> Can you try building a new RCP with the latest main branch of https://github.com/openthread/ot-nrf528xx ?

Sure, will give it a try ASAP. Thank You~

RoyAnupam, 2021-08-17:
Hi @jwhui 

> There was a recent change to the spinel protocol that requires updating the RCP ([a94e88c](https://github.com/openthread/openthread/commit/a94e88cd5081148a8bcc99c5123337b94612f995)).
> 
> Can you try building a new RCP with the latest main branch of https://github.com/openthread/ot-nrf528xx ?

 After updating new RCP, the ot-daemon seems to be launched successfully. Following are sys logs

./output/posix/bin/ot-daemon[27074]: Running OPENTHREAD/20200818-01168-g3a90ed81f; POSIX; Aug 17 2021 10:39:13
./output/posix/bin/ot-daemon[27074]: Thread version: 3
./output/posix/bin/ot-daemon[27074]: Thread interface: wpan0
./output/posix/bin/ot-daemon[27074]: RCP version: OPENTHREAD/thread-reference-20200818-1160-gb42261f88; NRF52840; Aug 16 2021 21:45:47

**Thank You for your guidance!** 

However, I ran into a different problem this time.
The OTBR docker image which use to run previously has now stopped run on top of this new RCP.
Do you see some incompatibility between RCP & host ?

avahi-daemon[189]: Registering new address record for 127.0.0.1 on lo.IPv4.
otbr-agent[217]: [INFO]-UTILS---: Running 0.3.0-8e0ee63
otbr-agent[217]: [INFO]-UTILS---: Thread version: 1.2.0
otbr-agent[217]: [INFO]-UTILS---: Thread interface: wpan0
otbr-agent[217]: [INFO]-UTILS---: Backbone interface: eth0
otbr-agent[217]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/radio
otbr-agent[217]: [INFO]-PLAT----: RCP reset: RESET_POWER_ON
otbr-agent[217]: [NOTE]-PLAT----: RCP API Version: 4
otbr-agent[217]: [CRIT]-PLAT----: RCP API Version 4 is not in the supported range [1-3]
otbr-agent[217]: [CRIT]-PLAT----: CheckRcpApiVersion() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:414: RadioSpinelIncompatible
otbr-web[245]: [INFO]-WEB-----: Running 0.3.0-8e0ee63
otbr-web[245]: [INFO]-WEB-----: Border router web started on wpan0
otbr-web[245]: [ERR ]-WEB-----: OpenThread daemon is not running.

jwhui, 2021-08-17:
> The OTBR docker image which use to run previously has now stopped run on top of this new RCP.
> Do you see some incompatibility between RCP & host ?

Yes, this is due to the same incompatibility issue, this time the host is too old for the given RCP. See the relevant log output below:

```
otbr-agent[217]: [NOTE]-PLAT----: RCP API Version: 4
otbr-agent[217]: [CRIT]-PLAT----: RCP API Version 4 is not in the supported range [1-3]
```

RoyAnupam, 2021-08-19:
Hi @jwhui 

> Yes, this is due to the same incompatibility issue, this time the host is too old for the given RCP. See the relevant log output below:
> 
> ```
> otbr-agent[217]: [NOTE]-PLAT----: RCP API Version: 4
> otbr-agent[217]: [CRIT]-PLAT----: RCP API Version 4 is not in the supported range [1-3]
> ```

After updating the otbr via [this](https://openthread.io/guides/border-router/docker), the issue seems to go away. Thank you very much for sharing insights!!

I had a query (off topic). As I understand, OBR can launch, either by directly running docker or by building ot-br-posix repo & executing otbr-agent manually & it has no pre-requirement of ot-daemon running in background, as per my current understanding. Then, why do it sometimes complains of  "otbr-web[245]: [ERR ]-WEB-----: OpenThread daemon is not running." ? Am I missing something?

And another one:: how does ot-ctl & ot-cli utility applications differ? I observed that, both are working & able to create a Network from command line  while otbr-agent is running in background(with my nRF528XX RCP device) ? Is there any recommendation about which one should be used when?

jwhui, 2021-08-21:
> I had a query (off topic). As I understand, OBR can launch, either by directly running docker or by building ot-br-posix repo & executing otbr-agent manually & it has no pre-requirement of ot-daemon [...]

When you get this error message, did you verify whether `ot-daemon` and/or `otbr-agent` running?

> And another one:: how does ot-ctl & ot-cli utility applications differ? I observed that, both are working & able to create a Network from command line while otbr-agent is running in background(with my nRF528XX RCP device) ? Is there any recommendation about which one should be used when?

`ot-ctl` is a utility for issuing CLI commands to `ot-daemon`. In this case, `ot-daemon` includes the OpenThread stack, whereas `ot-ctl` is simply a frontend for CLI interactions and connects to `ot-daemon` using a UNIX domain socket.

`ot-cli` is a single process that incorporates both the CLI and OpenThread. It does not does not provide a UNIX domain socket.

RoyAnupam, 2021-08-24:
> When you get this error message, did you verify whether `ot-daemon` and/or `otbr-agent` running?

Actually, my observation is following
- When I run OTBR via docker, I always get the **otbr-web[245]: [ERR ]-WEB-----: OpenThread daemon is not running** error
But, even if that error is output, OTBR eventually runs fine & I am able to form network via OpenThead WebGUI.
So, I did not really check whether ot-daemon process is actually running inside docker environment or not.
- When I launch otbr-agent manually, I never get that error message.

> `ot-ctl` is a utility for issuing CLI commands to `ot-daemon`. In this case, `ot-daemon` includes the OpenThread stack, whereas `ot-ctl` is simply a frontend for CLI interactions and connects to [...]
Thank you for sharing the insights~

jwhui, 2021-08-25:
> Does otbr-agent provide DBUS interface for ot-ctl to communicate with RCP?

`otbr-agent` does support a D-Bus interface. However, that is not what `ot-ctl` uses. Instead, `ot-ctl` communicates with `otbr-agent` using a UNIX domain socket and generally forwards CLI commands and output verbatim.

closed 2021-08-25 by RoyAnupam