openthread/openthread #6930: [github-actions] tests/scripts/expect/cli-commissioner.exp fails intermittently

jwhui, 2021-08-16:
[logs_91907.zip](https://github.com/openthread/openthread/files/6993891/logs_91907.zip)

jwhui, 2021-09-14:
[logs_95457.zip](https://github.com/openthread/openthread/files/7164852/logs_95457.zip)

jwhui, 2021-11-10:
[logs_102281.zip](https://github.com/openthread/openthread/files/7514798/logs_102281.zip)
[syslog-expect-linux.zip](https://github.com/openthread/openthread/files/7514800/syslog-expect-linux.zip)

simonlingoogle, 2021-11-11:
Should assign to @wgtdkp ?

jwhui, 2021-11-11:
@wgtdkp , can you help take a look?

jwhui, 2022-01-15:
[logs_110576.zip](https://github.com/openthread/openthread/files/7873928/logs_110576.zip)
[syslog-expect-linux.zip](https://github.com/openthread/openthread/files/7873929/syslog-expect-linux.zip)

wgtdkp, 2022-01-15:
The log shows that the `commissioner mgmtget sessionid steeringdata joinerudpport locator binary 0b081209` command fails with `Invalid Args` which is expected because the CLI format is changed to use `-x` for binary arguments in #5549:
```
2022-01-15T00:38:35.7052385Z expect: set expect_out(buffer) "\r\r\n> Donecommissioner mgmtget sessionid steeringdata joinerudpport locator binary 0b081209\r\n\r\r\n> Error 7: InvalidArgs\r\r\n> "
2022-01-15T00:38:35.7052766Z Timed out
2022-01-15T00:38:35.7052953Z     while executing
2022-01-15T00:38:35.7053133Z "error $message"
2022-01-15T00:38:35.7053339Z     (procedure "fail" line 3)
2022-01-15T00:38:35.7053537Z     invoked from within
2022-01-15T00:38:35.7053738Z "fail "Timed out" "
2022-01-15T00:38:35.7053941Z     invoked from within
2022-01-15T00:38:35.7054230Z "expect -re "\[\r\n \]($line)(?=\[\r\n>\])""
2022-01-15T00:38:35.7054477Z     (procedure "expect_line" line 3)
2022-01-15T00:38:35.7054689Z     invoked from within
2022-01-15T00:38:35.7054894Z "expect_line "Done""
2022-01-15T00:38:35.7055231Z     (file "tests/scripts/expect/cli-commissioner.exp" line 69)
2022-01-15T00:38:35.7068282Z ##[error]Process completed with exit code 1.
```

https://github.com/openthread/openthread/pull/7325 is opened to fix the cli command.

------

It's unexpected this test case doesn't fail consistently. A success test result shows that the `Done` output of the previous command is captured by `expect_line` below the `commissioner mgmtget` command:

```
> commissioner mgmtset sessionid 14283 locator 0x0100
commissioner mgmtget sessionid steeringdata joinerudpport locator binary 0b081209
Done

> Error 7: InvalidArgs
```

And so the `InvalidArgs` error is ignored.

I think there is something wrong with either the CLI output driver or the `expect` driver. @bukepo thoughts?

jwhui, 2022-03-14:
https://github.com/openthread/openthread/runs/5544984585?check_suite_focus=true

jwhui, 2025-03-11:
Stale issue.

closed 2025-03-11 by jwhui