openthread/openthread #6940: RadioSpinelNoResponse after restarting otbr-agent with nrf52840-dongle

mspang, 2021-08-17:
**Describe the bug**

Restarting otbr-agent results in a crash loop with the following message

```
Aug 17 20:54:36 ubuntu otbr-agent[2514]: [CRIT]-PLAT----: HandleRcpTimeout() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2204: RadioSpinelNoResponse
```

**To Reproduce**

raspberry pi 4b (Ubuntu 21.04)
ot-br-posix @ b66e46416a34dc25ec17693affab006629db1385
ot-nrf528xx @ 063233bf4a0833bda55295bc89890a96abb5aeb8

- Set up otbr via
INFRA_IF_NAME=eth0 ./script/setup

- Set up nrf52840-dongle via
script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB -DOT_THREAD_VERSION=1.2
arm-none-eabi-objcopy -O ihex build/bin/ot-rcp build/bin/ot-rcp.hex
nrfutil pkg generate --hw-version 52 --sd-req=0x00 --application build/bin/ot-rcp.hex --application-version 1 build/bin/ot-rcp.zip
nrfutil dfu usb-serial -pkg build/bin/ot-rcp.zip -p /dev/ttyACM0

- Reboot Pi
reboot

- Restart otbr-agent
systemctl restart otbr-agent

This results in a timeout.

**Expected behavior**

otbr-agent restarts

**Console/log output** 

```
Aug 17 20:54:33 ubuntu systemd[1]: Started Border Router Agent.
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Running 0.3.0-b66e464
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Thread version: 1.2.0
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Thread interface: wpan0
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Backbone interface: eth0
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Radio URL: trel://eth0
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-PLAT----: RCP reset: RESET_POWER_ON
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [NOTE]-PLAT----: RCP API Version: 4
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-PLAT----: Backbone interface is configured to eth0 (2)
Aug 17 20:54:34 ubuntu systemd-udevd[2513]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Aug 17 20:54:34 ubuntu networkd-dispatcher[1695]: WARNING:Unknown index 5 seen, reloading interface list
Aug 17 20:54:34 ubuntu avahi-daemon[1683]: Registering new address record for fe80::5873:39b0:f14e:c1b5 on eth0.*.
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-AGENT---: Set state callback: OK
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-AGENT---: Stop Thread Border Agent
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-ADPROXY-: Stopped
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-DPROXY--: stopped
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-AGENT---: Initialize OpenThread Border Router Agent: OK
Aug 17 20:54:34 ubuntu named[1767]: listening on IPv6 interface eth0, fe80::5873:39b0:f14e:c1b5%2#53
Aug 17 20:54:34 ubuntu otbr-agent[2514]: [INFO]-UTILS---: Border router agent started.
Aug 17 20:54:36 ubuntu otbr-agent[2514]: [CRIT]-PLAT----: HandleRcpTimeout() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2204: RadioSpinelNoResponse
Aug 17 20:54:41 ubuntu systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Aug 17 20:54:41 ubuntu systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
```

**Additional context** 

Reverting https://github.com/openthread/openthread/pull/6849 seems to avoid this issue.

edmont, 2021-08-18:
Hi @mspang, thanks for raising this. I've just [built](https://openthread.io/guides/border-router/docker#build_the_dockerfile) the Docker image with the commit you indicate for my laptop (Ubuntu 21.04) and can't observe the issue:

```
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Running 0.3.0-b66e46416
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Thread version: 1.2.0
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Thread interface: wpan0
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Backbone interface: eth0
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM5
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-PLAT----: RCP reset: RESET_POWER_ON
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [NOTE]-PLAT----: RCP API Version: 4
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-PLAT----: Backbone interface is configured to eth0 (22)
Aug 18 14:37:11 2d4b577c6445 otbr-web[245]: [INFO]-WEB-----: Running 0.3.0-b66e46416
Aug 18 14:37:11 2d4b577c6445 otbr-web[245]: [INFO]-WEB-----: Border router web started on wpan0
Aug 18 14:37:11 2d4b577c6445 otbr-web[245]: [ERR ]-WEB-----: OpenThread daemon is not running.
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-AGENT---: Set state callback: OK
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-AGENT---: Stop Thread Border Agent
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-ADPROXY-: Stopped
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-DPROXY--: stopped
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-AGENT---: Initialize OpenThread Border Router Agent: OK
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [INFO]-UTILS---: Border router agent started.
Aug 18 14:37:11 2d4b577c6445 otbr-agent[217]: [DEBG]-BBA-----: BackboneAgent: HandleBackboneRouterState: state=1, mBackboneRouterState=0
Aug 18 14:37:11 2d4b577c6445 avahi-daemon[189]: Server startup complete. Host name is 2d4b577c6445.local. Local service cookie is 1286869766.
Aug 18 14:37:11 2d4b577c6445 rsyslogd: rsyslogd's groupid changed to 101
Aug 18 14:37:11 2d4b577c6445 rsyslogd: rsyslogd's userid changed to 101
Aug 18 14:37:11 2d4b577c6445 rsyslogd:  [origin software="rsyslogd" swVersion="8.32.0" x-pid="122" x-info="http://www.rsyslog.com"] start
Aug 18 14:37:56 2d4b577c6445 otbr-agent[217]: [INFO]-AGENT---: Stop Thread Border Agent
Aug 18 14:37:56 2d4b577c6445 otbr-agent[217]: [INFO]-ADPROXY-: Stopped
Aug 18 14:37:56 2d4b577c6445 otbr-agent[217]: [INFO]-DPROXY--: stopped
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Running 0.3.0-b66e46416
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Thread version: 1.2.0
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Thread interface: wpan0
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Backbone interface: eth0
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM5
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-PLAT----: RCP reset: RESET_POWER_ON
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [NOTE]-PLAT----: RCP API Version: 4
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-PLAT----: Backbone interface is configured to eth0 (22)
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-AGENT---: Set state callback: OK
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-AGENT---: Stop Thread Border Agent
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-ADPROXY-: Stopped
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-DPROXY--: stopped
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-AGENT---: Initialize OpenThread Border Router Agent: OK
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [INFO]-UTILS---: Border router agent started.
Aug 18 14:37:56 2d4b577c6445 otbr-agent[280]: [DEBG]-BBA-----: BackboneAgent: HandleBackboneRouterState: state=1, mBackboneRouterState=0
```
I restarted the otbr-agent as follows:
```
edmo@edmo-laptop:~$ docker ps
CONTAINER ID        IMAGE               COMMAND                  CREATED             STATUS              PORTS                  NAMES
2d4b577c6445        openthread/otbr     "/app/etc/docker/doc…"   11 seconds ago      Up 11 seconds       0.0.0.0:8080->80/tcp   gifted_swirles
edmo@edmo-laptop:~$ docker exec -it 2d4b577c6445 bash
root@2d4b577c6445:/app# service otbr-agent restart
 * Stopping thread border agent otbr-agent                                           [ OK ] 
 * Starting thread border agent otbr-agent                                           [ OK ] 
```
I was also able to properly form a Thread network from the web UI.

I'll need to verify with an actual Raspberry Pi when I'll have access to it.

canisLupus1313, 2021-08-20:
Hi,
@mspang

I'm using the following config

HW
```
pbida@pbida-desktop:~$ uname -a
Linux pbida-desktop 5.11.0-1016-raspi #17-Ubuntu SMP PREEMPT Thu Jul 29 15:33:06 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
pbida@pbida-desktop:~$ cat /etc/issue
Ubuntu 21.04 \n \l

pbida@pbida-desktop:~$
```
Raspberry Pi 4b 
AC adapter with output 5.1v 3A

SW versions as requested:
- ot-br-posix @ b66e46416a34dc25ec17693affab006629db1385
- ot-nrf528xx @ 063233bf4a0833bda55295bc89890a96abb5aeb8

The output from otbr no matter how manytims I restart the otbr
```
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Running 0.3.0-b66e46416
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Thread version: 1.2.0
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Thread interface: wpan0
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Backbone interface: eth0
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Radio URL: trel://eth0
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-PLAT----: RCP reset: RESET_POWER_ON
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [NOTE]-PLAT----: RCP API Version: 4
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-PLAT----: Backbone interface is configured to eth0 (2)
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Set state callback: OK
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Stop Thread Border Agent
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Stopped
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-DPROXY--: stopped
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Initialize OpenThread Border Router Agent: OK
sie 20 10:54:40 pbida-desktop otbr-agent[630]: [INFO]-UTILS---: Border router agent started.
sie 20 10:55:18 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Thread is down
sie 20 10:55:18 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Stop Thread Border Agent
sie 20 10:55:18 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Stopped
sie 20 10:55:18 pbida-desktop otbr-agent[630]: [INFO]-DPROXY--: stopped
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Thread is up
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Stop Thread Border Agent
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Stopped
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-DPROXY--: stopped
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Started
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Publish meshcop service OpenThread_BorderRouter._meshcop._udp.local.
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-MDNS----: Add service: OpenThread_BorderRouter._meshcop._udp (ref: 0xaaaafddf8dd0)
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Publish all hosts and services
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-DPROXY--: started
sie 20 10:55:20 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Start Thread Border Agent: OK
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-MDNS----: Received reply for service OpenThread_BorderRouter._meshcop._udp.
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-MDNS----: Successfully registered service OpenThread_BorderRouter._meshcop._udp.
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Handle publish service 'OpenThread_BorderRouter._meshcop._udp.' result: 0
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-AGENT---: Publish meshcop service OpenThread_BorderRouter._meshcop._udp.local.
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-MDNS----: Update service OpenThread_BorderRouter._meshcop._udp
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [INFO]-ADPROXY-: Handle publish service 'OpenThread_BorderRouter._meshcop._udp' result: 0
sie 20 10:55:21 pbida-desktop otbr-agent[630]: [NOTE]-BBA-----: BackboneAgent: Backbone Router becomes Primary!
```
I have installed this otbr with following sequence:
```
./script/bootstrap
INFRA_IF_NAME=eth0 ./script/setup
sudo reboot
```

I've noticed that You haven't added the `./script/bootstrap` in install step Can You confirm that this was done?
Also can You verify your power adapter?

mspang, 2021-08-23:
Confirmed that I ran the bootstrap script.

Power adapter: Canakit 5.1V 3.5A

ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.11.0-1016-raspi #17-Ubuntu SMP PREEMPT Thu Jul 29 15:33:06 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

I have the dongle plugged into one of the blue (USB3) ports.

mspang, 2021-08-23:
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mspang, 2021-08-23:
It seems flaky. Sometimes it comes alive after a few retries (but usually the failures seem persistent). I have also seen occasional failures at boot.

mspang, 2021-08-23:
I have two different dongles, the labels are:

PCA10059
2.0.0
2021.9
E8:B4:32:C4:43:2A

PCA10059
1.2.0
2020.19
C8:27:C5:91:85:8C

Haven't tried a different Pi or reimaging the Pi, though. It worked OK before the noted patch.

canisLupus1313, 2021-08-24:
One more question have You tried to clean the build artifacts from ot-nrf528xx repo? I have noticed that when I don't do it between git checkout/fetch/submodue I'm getting some strange problems.