openthread/openthread #6951: On-mesh commissioner having multiple Border Routers and partitions

tlifschitz, 2021-08-23:
Hi All,

In our application we have SEDs and Border Routers (a custom Posix platform with NCP).  We have a script running on each BR which forms an OpenThread network using the CLI, always with the same parameters. Since we want that a new SED can be commissioned and be able to join the network at any time, this script also periodically checks if the on-mesh built-in commissioner is active, otherwise activating it and adding the corresponding joiner. 

In some cases two or more BR may see each other, and the two partitions merge into a single one. However since signal between them may be marginal the BRs can loose connection from time to time, and become two different networks again.

We found that if a device in the partition starts its commissioner, the commissioner already active is stopped, the new commissioner petition succeeds and becomes active. And since our script always starts the commissioner if it is not active, the two BRs start fighting for having the commissioner active, and so Network Data changes very frequently.

I could not find anything regarding commissioners and merging/splitting partitions in the specification, specifically in section 8.4.2.1 Commissioner Petitioning. I also could not find any test having more than one commissioner in the same network.

The questions are:

- Why is the leader is accepting a new commissioner if there is already an active one in the network?

- How can we guarantee having always one active commissioner per network partition so any device can join at any time?

- Is there a way to check if there is already an active commissioner in the network before trying to start one?

jwhui, 2021-08-25:
@tlifschitz , thanks for raising this issue.

The current leader implementation will accept a new petition from a commissioner if it has the same commissioner id.

https://github.com/openthread/openthread/blob/f1e7ceb12ee624188a50fbdadd7e1038bc1cf48e/src/core/meshcop/meshcop_leader.cpp#L88-L90

However, the current OpenThread commissioner hardcodes the commissioner id to a fixed value.
https://github.com/openthread/openthread/blob/f1e7ceb12ee624188a50fbdadd7e1038bc1cf48e/src/core/meshcop/commissioner.cpp#L836

Some ideas to consider:

1. Include some kind of unique ID (e.g. Extended Address) to the commissioner id.
2. Add a command-line argument that enables setting the commissioner id.

Thoughts?

tlifschitz, 2021-08-31:
Thank you for the clarification @jwhui!

I agree that adding a custom id would solve the problem of having the commissioner alternating from device to device, as the leader will reject other petitions. A command line argument would allow to have a consistent commissioner ID between resets (e.g. chip id), instead of using extaddress which is generated randomly. 

With this modification, every BR on the network would still try to start its commissioner periodically, getting its petition rejected and generating unnecessary traffic. Does every router in the network have stored the information about the active commissioner? Is this information available in the CLI interface?

jwhui, 2022-05-05:
Resolved by #7520

closed 2022-05-05 by jwhui