openthread/openthread #6953: Change of the default datagram count of 106-octet IPv6 datagram per attached SED doesn't take effect (i.e. fixed as 1)

HeroHerbert, 2021-08-24:
There is a requirement from application that for SED device,  the number of buffered messages can be greater than 1, for example 8 messages, we found that the current openthread (till 08-24) stack can only support 1 buffered message for each SED device, please help to implement.

1.  I changed the default setting of OPENTHREAD_CONFIG_DEFAULT_SED_DATAGRAM_COUNT  to 8
#ifndef OPENTHREAD_CONFIG_DEFAULT_SED_DATAGRAM_COUNT
#define OPENTHREAD_CONFIG_DEFAULT_SED_DATAGRAM_COUNT 8
#endif

2. Yet in the below function for SED devices,  buffered message can only be 1 (shall be OPENTHREAD_CONFIG_DEFAULT_SED_DATAGRAM_COUNT?)
void IndirectSender::AddMessageForSleepyChild(Message &aMessage, Child &aChild)
{
    uint16_t childIndex;

    OT_ASSERT(!aChild.IsRxOnWhenIdle());

    childIndex = Get<ChildTable>().GetChildIndex(aChild);
    VerifyOrExit(!aMessage.GetChildMask(childIndex));

    aMessage.SetChildMask(childIndex);
    mSourceMatchController.IncrementMessageCount(aChild);

    if ((aMessage.GetType() != Message::kTypeSupervision) && (aChild.GetIndirectMessageCount() > 1))
    {
        Message *supervisionMessage = FindIndirectMessage(aChild, /* aSupervisionTypeOnly */ true);

        if (supervisionMessage != nullptr)
        {
            IgnoreError(RemoveMessageFromSleepyChild(*supervisionMessage, aChild));
            Get<MeshForwarder>().RemoveMessageIfNoPendingTx(*supervisionMessage);
        }
    }

    RequestMessageUpdate(aChild);

exit:
    return;
}

librasungirl, 2021-08-25:
OpenThread provides `best-effort` message queue, regardless of indirect message for sed. The `best-effort` means evicting message buffers of low priority or same priority when `OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS`  runs out and no free message buffer available. And I think there is no plan to guarantee dedicated message buffer capacity per child. 

For 1) sorry for the confusion on `OPENTHREAD_CONFIG_DEFAULT_SED_DATAGRAM_COUNT`.  It's placed in a config file but it is not expected to be changeable.  Will enhance to move it into some internal constants.

For 2) the code snippet you pasted, there might be some misunderstanding.
OpenThread provides some enhanced features, [Child Supervision](https://openthread.io/guides/build/features/child-supervision) is one of them. The basic idea is the parent will send a supervision message (of type `Message::kTypeSupervision`) to check the liveness of the sed child if there is no message interaction between them in some period.

The code here is that if there is some non-supervision message (e.g. application message) destined to the sed, previous queued `supervision` message, if there is, could be removed. As the delivery of this non-supervision message could be an indicator of the liveness of sed as well. Note that only supervision message will be removed, not the general application message.

If there are frequent application messages to the SED in your use case, some possible factors you may consider for better performance
1) total available message buffers (`OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS`) on Router device
2) the message priority` otMessagePriority`
3) appropriate poll period of the SED

HeroHerbert, 2021-08-25:
Thanks for your clarification. 
The next question is - how to pass the priority information from the Application to the thread stack?

jwhui, 2021-09-10:
Closing stale issue.

closed 2021-09-10 by jwhui