openthread/openthread #6971: I have installed pyrebase , When i try to executed  , I got this from terminal

rajcomputer-dwd, 2021-08-30:
C:\Users\Naveen\AppData\Local\Programs\Python\Python39\python.exe C:/Users/Naveen/PycharmProjects/pythonProject1/main.py
Traceback (most recent call last):
  File "C:\Users\Naveen\PycharmProjects\pythonProject1\main.py", line 1, in <module>
    from pyrebase import pyrebase
  File "C:\Users\Naveen\AppData\Local\Programs\Python\Python39\lib\site-packages\pyrebase\__init__.py", line 1, in <module>
    from .pyrebase import initialize_app
  File "C:\Users\Naveen\AppData\Local\Programs\Python\Python39\lib\site-packages\pyrebase\pyrebase.py", line 23, in <module>
    from Crypto.PublicKey import RSA
ModuleNotFoundError: No module named 'Crypto'

jwhui, 2021-08-30:
This seems similar to #1137. Have you worked through those comments?

rajcomputer-dwd, 2021-08-30:
I have tried pip install pycryptodome==3.4.3
 but it is not able to install , in terminal it shows like this 
Running setup.py install for pycryptodome ... error

rdxpt, 2021-09-01:
@rajcomputer-dwd is there any other stack trace that you can share? Not sure if I can help, but I have been able to install `pycryptodome`, only getting the dist info details for it and not the sources itself.

jwhui, 2021-09-21:
Closing stale issue.

closed 2021-09-21 by jwhui