openthread/openthread #6980: More than one device with leader role in thread network.

mohinishnmrinq, 2021-09-02:
I am facing an issue wherein more than one device is acquiring the leader role in a single thread network.

In my setup I have a total of 4 devices. Three of them are nfr52833_dk and one is nfr52840_dk. I am using the NCS v1.6.1 SDK version to build the projects. On three of the nfr52833_dk, I flashed coap_server open-thread sample and one nfr52840_dk I flashed coap_client open-thread sample.

When I turned on all the nfr52833_dk (running coap_server) I can see one of the devices acquire leader roles and the other two devices acquire router and child roles respectively.

But once I turn on the nfr52840_dk (running coap_client), I can see it acquire leader role.

After few seconds the devices with router and child role change to the leader role. Thus resulting in multiple leaders' role devices in the thread network.

I haven't done any changes to the sample program. Am I missing anything here?
Another thing is I verified the panid, master key, channel set on all devices. It's the same on all devices.

jwhui, 2021-09-02:
Given that you are using NCS, you should try posting to [Nordic DevZone](https://devzone.nordicsemi.com/).

At the same time, it is difficult to diagnose your issues without more visibility. Can you provide OpenThread logs and/or a packet sniffer trace? Can you provide more detail in how you are configuring the Active Operational Dataset?

jwhui, 2021-09-21:
Closing stale issue.

closed 2021-09-21 by jwhui