openthread/openthread #6981: Question about intended CSL tx/rx behavior and recommended configuration

suveshpratapa, 2021-09-02:
We're having some trouble getting CSL transmission / reception working on the SiLabs platform.  Our SSED completes the child update / child update response exchange, however, we're not noticing them communicating on the expected CSL windows.  Additionally, the SSED never invokes `otPlatRadioReceiveAt` for scheduled-rx.  I have attached a Wireshark trace showing how far we get: [latest-csl-trace-silabs.pcapng.zip](https://github.com/openthread/openthread/files/7101465/latest-csl-trace-silabs.pcapng.zip)

I had a set of questions about recommended CSL configuration, and the functions we should be providing in our PAL/radio layer.

These are Thread v1.2 devices, so `OPENTHREAD_CONFIG_MAC_CSL_TRANSMITTER_ENABLE` is automatically enabled.
Additionally, there are the following config options.  Can you please let us know if our understanding of these options is correct?

- `OPENTHREAD_CONFIG_MAC_CSL_RECEIVER_ENABLE`: **Enabled** for all devices. For SEDs, enabling this makes them SSEDs, capable of CSL synchronization and communication.
- `OPENTHREAD_CONFIG_TIME_SYNC_ENABLE`: **Enabled**. We assume this has to be turned on, for time synchronization calculation to work for CSL.
- `OPENTHREAD_CONFIG_PLATFORM_USEC_TIMER_ENABLE`: **Enabled**. Microsecond timer support is required for CSL to work.
- `OPENTHREAD_CONFIG_MAC_HEADER_IE_SUPPORT`: **Enabled** automatically based on the above options.
- `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_SECURITY_ENABLE`: **Enabled**. We have to secure outgoing messages in Thread 1.2 at the PAL layer, for cases such as adding IE elements before transmission.
- `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE`: **_Disabled_**. If this option is turned on, the platform provides s/w transmission timing logic.  Looking at the code, our assumption is that the stack expects the radio layer to schedule a tx with its own timing parameters (https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.cpp#L378).  However, since our design supports a single tx/callback mechanism, we have disabled this. We now expect the stack to issue a request to tx on the scheduled window. (Effectively a Transmit function, rather than the radio providing TransmitAt)
- `OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE`: **Enabled**. When enabled, the platform is expected to implement scheduled receive API, otPlatRadioReceiveAt, which we are. We assume that aStart is absolute time used by the scheduler, whereas aDuration is relative to the start time.
- `OPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE`: **_Disabled_**. We see that this option was recently added.  It seems to allow CSL synchronization using data polls.  I'm unclear on when this is exactly required.

Note that we are not changing any of the other CSL timing defaults in: https://github.com/openthread/openthread/blob/main/src/core/config/mac.h

In our platform abstraction layer, we take note of CSL period and sample times (`otPlatRadioEnableCsl` and `otPlatRadioUpdateCslSampleTime`); and calculate CSL phase before including the IE headers in outgoing packets. Additionally, as mentioned above, we assume the stack will issue tx on the expected transmit window, whereas we take care of the receive window, using `otPlatRadioReceiveAt`.

---

Additionally, I have one more question about the following piece of code, which performs synchronization calculation only for "secured enhanced ACKs":
https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.cpp#L276

When we generate an enhanced ACK, we secure it if the incoming data message was also secured.  For example, for messages such as the Child Update Response (which is not MAC-secured), we reply with an unsecured enhanced ACK. Therefore, this wasn't working for us.  Removing the check for `aFrame->mInfo.mRxInfo.mAckedWithSecEnhAck` made it work. Could this be a bug?

suveshpratapa, 2021-09-02:
Linking to https://github.com/openthread/ot-efr32/issues/137

suveshpratapa, 2021-09-02:
@jwhui @romacdon

George-Stefan, 2021-09-02:
^^ @doru91, @marius-preda, @mihai-ignat

Irving-cl, 2021-09-04:
Hi @suveshpratapa 

I think your understanding of the options are basically right.  Note that:
* `OPENTHREAD_CONFIG_TIME_SYNC_ENABLE` is actually a unrelated feature (it's not in the spec). So it's actually not required.
* `OPENTHREAD_CONFIG_PLATFORM_USEC_TIMER_ENABLE` actually depends on the platform. If a platform's timer supports microsecond precision then it should set it in its config, for example, `examples/platforms/simulation/openthread-core-simulation-config.h`. For CSL it's required. I think SiLabs supports microsecond. So you need to set this in your config file.
* `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE`:

> However, since our design supports a single tx/callback mechanism, we have disabled this. We now expect the stack to issue a request to tx on the scheduled window. (Effectively a Transmit function, rather than the radio providing TransmitAt)

If the radio driver doesn't use something like 'transmit_at' and directly transmit when `otPlatRadioTransmit` is called, I think we should enable `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE` here.

Some explanation:
When CslTxScheduler requests a CSL TX from the Mac layer, the Mac layer will first set up a timer which fires at a rough time (slightly ahead of the precise expected TX time). And when the timer fires, it passes the TX request to the SubMac layer. At this moment, there should be still a very short time to do the TX (a few microseconds). So now there are two choices: 1. SubMac starts a precise timer and will pass the TX to the radio driver when the second timer fires, and the radio directly does the TX. 2. The radio provides 'transmit_at' function. The SubMac then doesn't start any timer and directly passes the TX request to the radio layer. When the option is enabled, it goes to case 1. Otherwise it goes to case2. I think in our case we should go to case 2. If we don't do this, then the TX should go a little earlier as I explained. A few microseconds is also fatal to CSL.
* `OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE`: You're correct about this. I think we can turn it off when we debug it.
* `OPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE`: You're also correct. For CSL to work, this isn't a MUST.

---

Regarding this question:

> Removing the check for aFrame->mInfo.mRxInfo.mAckedWithSecEnhAck made it work. Could this be a bug?

I think an unsecured Enhanced ACK should also let `mCslLastSync` be updated. So yes. This thing is introduced from #6311 as an enhancement though. Do you mean, after you remove this check, the CSL can work?

---

Some of my observation:
[image]
The time for the packet No.27 Child Update Requeset (when the first symbol is sent) is: `33.572245 - 104 * 0.000032 = 33.568917`
The phase in the CSL IE is **1195**, namely `1195 * 0.000160 = 0.1912` seconds.
So the expected time of the Child Update Response is `33.568917 + 0.1912 = 33.760117`. For the expected time displayed on wireshark (add the time for the bytes): `33.760117 + 71 * 0.000032 = 33.762389`
And the actual time (in wireshark) is **34.764021**. The TX time seems much later than expected.

BTW, the enhanced ACK (No.28) seems malformed.

---

You mentioned:

> Additionally, the SSED never invokes otPlatRadioReceiveAt for scheduled-rx.

`otPlatRadioReceiveAt` is called via `Radio::ReceiveAt` in `SubMac::HandlerTimer`: [link](https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.cpp#L1038) When CSL (SSED) started, the timer should run regularly. Could you add some logs and find why `ReceiveAt` isn't called in the timer handler.

---

I think both TX and RX side have some problems. Probably we could first try to solve the RX side's issue. BTW, feel free for any further questions.

suveshpratapa, 2021-09-08:
@Irving-cl Thanks for the reply! I have made some changes, and have some questions for you:

1. Disabled `OPENTHREAD_CONFIG_TIME_SYNC_ENABLE`.
2. Enabled `OPENTHREAD_CONFIG_PLATFORM_USEC_TIMER_ENABLE`.  If the platform doesn't support microsecond precision and disables this setting, will CSL work as intended? If so, should CSL automatically be disabled?
3. Enabled `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE`, because you say this should be turned on if only `otPlatRadioTransmit` is implemented (with no `TransmitAt` functionality).
But I have a question.  At this point in code: https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.cpp#L378, you can see that `TransmitAt` is called if `ShouldHandleTransmitTargetTime` is **true** (ie. `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE` is **true**).  Am I misunderstanding this piece of code applying to CSL?
4. Enabled `OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE`
5. Disabled `OPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE`

---

I submitted a PR to fix the issue with the check for `aFrame->mInfo.mRxInfo.mAckedWithSecEnhAck` https://github.com/openthread/openthread/pull/6994.  With this check present, Child Update Response packet doesn't seem to be sent.

---

I'm fixing the issue with malformed enhanced ACKs, and I will get back to you with some logs to track why `otPlatRadioReceiveAt` is not being called.

Irving-cl, 2021-09-09:
Hi @suveshpratapa 

> Enabled OPENTHREAD_CONFIG_PLATFORM_USEC_TIMER_ENABLE. If the platform doesn't support microsecond precision and disables this setting, will CSL work as intended? If so, should CSL automatically be disabled?

I think if `OPENTHREAD_CONFIG_PLATFORM_USEC_TIMER_ENABLE` is disabled, the CSL code won't be compiled successfully. Note in [sub_mac.hpp](https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.hpp#L671), there's a member `TimerMicro` for which a set of [AlarmMicroApi](https://github.com/openthread/openthread/blob/main/src/core/common/timer.cpp#L227) must be provided.

> But I have a question. At this point in code: https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.cpp#L378, you can see that TransmitAt is called if ShouldHandleTransmitTargetTime is true (ie. OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE is true).

Not sure if I understand this correctly. I don't see 'TransmitAt' in the code. From SubMac's perspective, it actually isn't aware of 'Transmit' or 'TransmitAt'. All it knows is to call `Radio::Transmit` (in `SubMac::BeginTransmit`). The code you mentioned:
```
        if (ShouldHandleTransmitTargetTime())
        {   // Case 1
            if (Time(static_cast<uint32_t>(otPlatRadioGetNow(&GetInstance()))) <
                Time(mTransmitFrame.mInfo.mTxInfo.mTxDelayBaseTime) + mTransmitFrame.mInfo.mTxInfo.mTxDelay -
                    kCcaSampleInterval)
            {   // Case 1.a
                mTimer.StartAt(Time(mTransmitFrame.mInfo.mTxInfo.mTxDelayBaseTime) - kCcaSampleInterval,
                               mTransmitFrame.mInfo.mTxInfo.mTxDelay);
            }
            else // Transmit without delay
            {   // Case 1.b
                BeginTransmit();
            }
        }
        else
        {   // Case 2
            BeginTransmit();
        }
```
I marked 3 cases in the abovce code: **Case 1.a**, **Case 1.b**, **Case 2**:
* Case 1: `ShouldHandleTransmitTargetTime` is true. In this case, SubMac will do the timer work and send the frame at the expected time
    * Case 1.a: As I mentioned in my earlier comment, when the code arrives here, there should be a short time until the expected tx time, so SubMac will start a timer for it and it will fires at the expected tx time.
    * Case 1.b: When the code arrives here, it finds that the current time is already close to the expected tx time. Then it directly call `BeginTransmit` to send the frame. This is most probably called after the timer set in Case 1.a fires. Then `SubMac::HandleTimer` is called and `StartCsmaBackoff` is called again, and it enters Case 1.b this time.
 * Case 2: SubMac won't be responsible for the accurate timing. It directly calls `BeginTransmit` to pass the request to the radio platform. Note that in these cases, though the sending functions are the same `BeginTransmit`, the radio driver can detect the difference and behave differently. The radio driver will check `mTxDelayBaseTime` and `mTxDelay` field of the frame. If the fields are set, the radio driver then knows that it should do the delay sending so that the frame is sent at the expected time. Otherwise, the radio driver directly sends it. In our case, `mTxDelayBaseTime` and `mTxDelay` are not set.

Irving-cl, 2021-10-22:
Hi @suveshpratapa,

For the first question:
I think when you use 'ReceiveTiming' you don't need to fall into the case under `kCslSample`. The change for 'ReceiveTiming' is introduced in #6343. I think its intention is that, in SubMac layer, the CSL state keeps 'kCslSleep' forever. But still keeps a timer which calls `Radio::ReceiveAt` regularly. And `Radio::ReceiveAt` will do the receiving at the expected time and change back to sleep after a short period of time.
Let's compare the difference b/t them:
```
Original:
Initial State: Sleep
Timer fires, State->Receive, call Radio::Receive
Timer fires, State->Sleep, call Radio::Sleep
Timer fires, State->Receive, call Radio::Receive
Timer fires, State->Sleep, call Radio::Sleep
...

ReceiveTiming:
Initial State: Sleep
Timer fires, State always Sleep, call Radio::ReceiveAt
Timer fires, State always Sleep, call Radio::ReceiveAt
...
```
Does your ReceiveAt API in radio have the same behavior? It has been some time since this change and I forgot some details about this API. @edmont Could you share some details for ReceiveAt API? Will it receive at the expected time for a specified period, and then radio will turn back to sleep?

After scanning the code just now, I think it's indeed not easy-to-read. I'll try to refactor the `HandleCslTimer` function. Or you can do it if you're interested :)

---
For the second question:
We also have these warnings from wireshark with our implementation. When I updated these code, I just refactored them without changing the logic. So I'm not quite sure about these rules. Probably someone else can answer this question.

suveshpratapa, 2021-11-12:
Hi @Irving-cl

SiLabs implementation of CSL for SoCs has:

radio transmit timing=off
radio receive timing=on

as our radio layer currently doesn't support scheduled-tx.  Per previous discussion in this thread, this is a supported configuration.

However, I'm seeing issues with our OTBR acting as a CSL transmitter.  It basically seems to have issues scheduling a transmit, and frequently hits asserts like:
```
third_party/openthread/repo/src/core/thread/csl_tx_scheduler.cpp:253:
'OT_ASSERT(!aFrame.GetSecurityEnabled() || aFrame.IsHeaderUpdated());' failed
```

This indicates to me that we're not setting the header updated flag properly for outgoing packets.

For our RCP configuration, I had:
```
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE 1
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE 1
```
Because otherwise, the check in `CheckRadioCapabilities` in `src/lib/spinel/radio_spinel_impl.hpp` fails.

I think this is wrong however.  In our RCP case, if we don't support tx-timing in our radio, I think we should be setting:
```
OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE=1
```

This means that the check for OT_RADIO_CAPS_TRANSMIT_TIMING should be removed in `CheckRadioCapabilities` in `src/lib/spinel/radio_spinel_impl.hpp`

Do I have this correct?  Can I get some advice on how we can get CSL tx scheduling set up properly if the radio doesn't support transmit timing?

Irving-cl, 2021-11-12:
Hi @suveshpratapa 

Regarding the `IsHeaderUpdated()` issue:
I think you have turned on the `OT_RADIO_CAPS_TRANSMIT_SEC` in your radio capability, right? If the radio platform (RCP) has such capability, the host won't process the security stuffs when preparing a frame. (HeaderUpdated is not set in the frame). And the radio platform should process security and set it. On RCP, `mHeaderUpdated` is set through `otMacFrameSetFrameCounter`. Here is how it's called on simulation platform and nrf platform:
* [simulation/radio.c](https://github.com/openthread/openthread/blob/main/examples/platforms/simulation/radio.c#L650)
* [ot-nrf528xx/src/src/radio.c](https://github.com/openthread/ot-nrf528xx/blob/main/src/src/radio.c#L296)

So you also need to call this when you process security for the frame in your radio platform.

---

Regarding the configuration issue:
Note that the two options means 'software timing'. If `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE` is turned on, then we use the timer in openthread core, not the one in the radio platform. So you said your platform doesn't support tx timing, then you should set `OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE` to 1.

So I think you should have such configuration:
```
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE 1
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE 0
```

suveshpratapa, 2021-11-12:
Thank you, I've adjusted the config as per your advice:
```
Radio doesn't support scheduled transmit, supports scheduled receive:

otPlatRadioGetCaps: returns OT_RADIO_CAPS_RECEIVE_TIMING only.

#define OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE 1
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE 0

... and this configuration should remain true for both SoC and RCP.
```

So, with the RCP case, I'm still seeing HeaderUpdated asserts for csl transmission.

In our platform code, we do call process transmit security for every outgoing frame, and make sure to call set frame counter / set header updated. In fact, we use the helper functions in `examples/platforms/utils/mac_frame.cpp` (`otMacFrameSetFrameCounter`, `otMacFrameProcessTransmitAesCcm`, etc.) to achieve this.

This works for SoC case, so are there any other considerations should I take into account for host / RCP?

suveshpratapa, 2021-11-12:
It's strange @Irving-cl I'm able to see the RCP set header updated flag for the outgoing frame.

In `LinkRawTransmitDone`, I'm seeing that when sending over the Spinel interface: `SuccessOrExit(mEncoder.WriteBool(headerUpdated));` is using a value of 1.

I'm really confused as to why the host sees headerUpdated as `false`.

Irving-cl, 2021-11-12:
@suveshpratapa 

That's really weird. Have you checked the value of `headerUpdated` when the host first read it from the RCP in `RadioSpinel::HandleTransmitDone` ([radio_spinel_impl.hpp](https://github.com/openthread/openthread/blob/main/src/lib/spinel/radio_spinel_impl.hpp#L1897))? I suggest to add a log here (using `otLogWarnPlat` to print the value here). And at the same time, check the `headerUpdated` on RCP in `LinkRawTransmitDone`. Let's see if there are any issues in Spi transfer.

jwhui, 2022-01-28:
@suveshpratapa , are there any further updates on this issue? Has this issue been resolved?

suveshpratapa, 2022-01-28:
@jwhui We *just* recently found some strange behavior with our OTBR as CSL transmitter.  I'm gathering some more data to ask a question here.  Can you please keep this open for some more time? Thanks!

suveshpratapa, 2022-02-11:
@Irving-cl I'm hoping you can take a look at this OTBR/CSL timing behavior we're noticing. I've exhausted my attempts to debug OTBR timing behavior, so I wanted to check whether I'm understanding some of the stack code properly.

The test is with OTBR/RCP as CSL transmitter, with an MED as a CSL receiver.

Our RCP configuration is as follows.  Our radio doesn't support scheduled-transmit, but supports scheduled-receive.
```
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_TX_TIMING_ENABLE 1
#define OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_TIMING_ENABLE 0
otPlatRadioGetCaps: configured with OT_RADIO_CAPS_RECEIVE_TIMING
```

After CSL synchronization, I tried three attempts to reach the MED using pings.  The first one failed, the second one succeeded, and the third one failed again.  When it fails, the tx schedule is very erratic:

```
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.063 [NOTE]-CLI-----: Input: ping fe80:0:0:0:d8a2:7170:b5e1:a453
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.064 [INFO]-ICMP----: Sent echo request: (seq = 3)
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.073 [INFO]-MAC-----: RescheduleCslTx delay: 269216 us. cslTxDelay: 4108490976 us.
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.073 [INFO]-MAC-----: RescheduleCslTx delay: 269216 us.
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.074 [INFO]-MAC-----: Prepping indir tx IPv6 ICMP6 msg, len:56, chksum:0000, to:0xc801, sec:yes, prio:normal
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.342 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4108490976 us. lastRxTimestamp: 191568426 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.368 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.368 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackreq:yes
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.369 [INFO]-MAC-----: CSL tx to child c801 failed, attempt 1/4
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.370 [INFO]-MAC-----: RescheduleCslTx delay: 471947 us. cslTxDelay: 4108990976 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.371 [INFO]-MAC-----: RescheduleCslTx delay: 471947 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.393 [INFO]-PLAT----: RCP => [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/0, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackre
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.844 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4109490976 us. lastRxTimestamp: 191568426 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.873 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.873 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackreq:yes
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.875 [INFO]-MAC-----: CSL tx to child c801 failed, attempt 2/4
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.875 [INFO]-MAC-----: RescheduleCslTx delay: 467027 us. cslTxDelay: 4109490976 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.877 [INFO]-MAC-----: RescheduleCslTx delay: 467027 us.
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.893 [INFO]-PLAT----: RCP => [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/0, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackre
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.345 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4109990976 us. lastRxTimestamp: 191568426 us.
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.368 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.368 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackreq:yes
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.369 [INFO]-MAC-----: CSL tx to child c801 failed, attempt 3/4
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.371 [INFO]-MAC-----: RescheduleCslTx delay: 471676 us. cslTxDelay: 4109990976 us.
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.371 [INFO]-MAC-----: RescheduleCslTx delay: 471676 us.
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.388 [INFO]-PLAT----: RCP => [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/0, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackre
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.842 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4109990976 us. lastRxTimestamp: 191568426 us.
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.863 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.864 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:225, type:Data, src:862e2a97d2826f4c, dst:daa27170b5e1a453, sec:yes, ackreq:yes
Feb 11 10:55:48 supratap-raspi2 otbr-agent[550]: 00:03:06.864 [INFO]-MAC-----: CSL tx to child c801 failed, attempt 4/4
[... cut ...]  version of the stack), there are no problems.

What could be the problem with OTBR timing, other than any possible delays over UART? (We have `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US` set at the default value of 2000)

Logs from all ping attempts:
[ping_attempt_logs.zip](https://github.com/openthread/openthread/files/8050043/ping_attempt_logs.zip)

Packet trace:
[capture-2022_Feb_11-10_20.pcapng.zip](https://github.com/openthread/openthread/files/8050041/capture-2022_Feb_11-10_20.pcapng.zip)

Irving-cl, 2022-02-12:
Hi @suveshpratapa,

The configuration for RCP is correct. I noticed that `Csl txDelay` is extremely big and seems weird.
```
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.064 [INFO]-ICMP----: Sent echo request: (seq = 3)
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.073 [INFO]-MAC-----: RescheduleCslTx delay: 269216 us. cslTxDelay: 4108490976 us.
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.073 [INFO]-MAC-----: RescheduleCslTx delay: 269216 us.
Feb 11 10:55:46 supratap-raspi2 otbr-agent[550]: 00:03:05.074 [INFO]-MAC-----: Prepping indir tx IPv6 ICMP6 msg, len:56, chksum:0000, to:0xc801, sec:yes, prio:normal
Feb 11 10:55:47 supratap-raspi2 otbr-agent[550]: 00:03:05.342 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4108490976 us. lastRxTimestamp: 191568426 us.
```
Roughly speaking, `Csl txDelay = nextTxWindow -  lastRxTimestamp`. In our case, `nextTxWindow` should be close to `lastRxTimestamp`(like a few seconds, minutes). But **4108490976 us** is more than one hour.

And the timer on host triggers correctly: At **00:03:05.073** we can see RescheduleCslTx delay: **269216** us. Then we can see at **00:03:05.342** `HandleFrameRequest` is called. And the difference is `342 - 73 = 269`ms. So I think the host timer works well and this may be an issue with the software timer on RCP. The timer on RCP would fire at the time `lastRxTimestamp + nextTxWindow`.

Could you add a log in `CslTxScheduler::GetNextCslTransmissionDelay` to print `radioNow`, `aChild.GetLastRxTimestamp()`, `aChild.GetCslPhase()`, `firstTxWindow` and `nextTxWindow`? And let's see if there's any errors here.

And is it possible to print some logs from RCP? I'd like to check how much time does the software timer on RCP wait. (The time b/w `mTimer.StartAt` called in `SubMac::StartCsmaBackoff` and `SubMac::HandleTimer`) Could you print the timestamp at the two places on RCP?

suveshpratapa, 2022-02-14:
Hi @Irving-cl thank you for your prompt reply!

I added some more logging as you requested:

- Log in CslTxScheduler::GetNextCslTransmissionDelay
    - rn: `radioNow`
    - clrs: `aChild.GetLastRxTimestamp()`
    - cp: `aChild.GetCslPhase()`
    - ftw: `firstTxWindow `
    - ntw: `nextTxWindow`

- As for RCP, I'm still looking into a proper way to grab the software timer logs on the RCP.  For some reason, I'm seeing weird behavior when I directly add logs under `SubMac::StartCsmaBackoff`.  I'll get back to you once I figure this out.

Here's an attempt at sending a message (ping) from OTBR->CSL child, with the new logging added:

```
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.289 [INFO]-PLAT----: > ping fe80:0:0:0:8872:26d3:87ef:c7c7
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.289 [NOTE]-CLI-----: Input: ping fe80:0:0:0:8872:26d3:87ef:c7c7
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.289 [INFO]-ICMP----: Sent echo request: (seq = 3)
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.305 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 23084977 clrs: 112642252 cp: 779 ftw: 112766892 ntw: 23266892
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.305 [INFO]-MAC-----: RescheduleCslTx delay: 169498 us. cslTxDelay: 4205591936 us.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.306 [INFO]-MAC-----: Prepping indir tx IPv6 ICMP6 msg, len:56, chksum:0000, to:0x0801, sec:yes, prio:normal
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.475 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 23254863 clrs: 112642252 cp: 779 ftw: 112766892 ntw: 23766892
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.475 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4206091936 us. lastRxTimestamp: 112642252 us.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.505 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:251, type:Data, src:6ecc898b291532ca, dst:8a7226d387efc7c7, sec:yes, ackreq:yes
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: =======================[TX ERR seqno=251 len=016]=======================
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: | 29 EC FB CE FA C7 C7 EF | 87 D3 26 72 8A CA 32 15 | )l{NzGGo.S&r.J2.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: ------------------------------------------------------------------------
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: =========================[TX seqno=251 len=052]=========================
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: | 29 EC FB CE FA C7 C7 EF | 87 D3 26 72 8A CA 32 15 | )l{NzGGo.S&r.J2.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: | 29 8B 89 CC 6E 0D 0B 00 | 00 00 01 7A 33 3A 80 00 | )..Ln......z3:..
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.506 [DEBG]-MAC-----: | 07 64 00 03 00 03 00 02 | 2A 13 00 00 00 00 D9 D8 | .d......*.....YX
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.507 [DEBG]-MAC-----: | 73 E4 AA 54 .. .. .. .. | .. .. .. .. .. .. .. .. | sd*T............
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.507 [DEBG]-MAC-----: ------------------------------------------------------------------------
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.507 [INFO]-MAC-----: CSL tx to child 0801 failed, attempt 1/4
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.507 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 23286677 clrs: 112642252 cp: 779 ftw: 112766892 ntw: 23766892
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.507 [INFO]-MAC-----: RescheduleCslTx delay: 467798 us. cslTxDelay: 4206091936 us.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.520 [INFO]-PLAT----: RCP => [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/0, len:52, seqnum:251, type:Data, src:6ecc898b291532ca, dst:8a7226d387efc7c7, sec:yes, ackre
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.975 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 23754139 clrs: 112642252 cp: 779 ftw: 112766892 ntw: 23766892
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:23.975 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 4206091936 us. lastRxTimestamp: 112642252 us.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.000 [WARN]-PLAT----: Handle transmit done failed: NoAck
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.000 [INFO]-MAC-----: Frame tx attempt 1/1 failed, error:NoAck, len:52, seqnum:251, type:Data, src:6ecc898b291532ca, dst:8a7226d387efc7c7, sec:yes, ackreq:yes
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.001 [DEBG]-MAC-----: =======================[TX ERR seqno=251 len=016]=======================
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.001 [DEBG]-MAC-----: | 29 EC FB CE FA C7 C7 EF | 87 D3 26 72 8A CA 32 15 | )l{NzGGo.S&r.J2.
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.001 [DEBG]-MAC-----: ------------------------------------------------------------------------
Feb 14 15:57:13 supratap-raspi2 otbr-agent[911]: 00:01:24.001 [DEBG]-MAC-----: =========================[TX seqno=251 len=052]=========================
[... cut ...]
Feb 14 15:57:14 supratap-raspi2 otbr-agent[911]: 00:01:25.007 [INFO]-MAC-----: CSL tx to child 0801 failed, attempt 4/4
Feb 14 15:57:14 supratap-raspi2 otbr-agent[911]: 00:01:25.020 [INFO]-PLAT----: RCP => [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/0, len:52, seqnum:251, type:Data, src:6ecc898b291532ca, dst:8a7226d387efc7c7, sec:yes, ackre
Feb 14 15:57:16 supratap-raspi2 otbr-agent[911]: 00:01:26.289 [NOTE]-CLI-----: Output: 1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Feb 14 15:57:16 supratap-raspi2 otbr-agent[911]: 00:01:26.289 [NOTE]-CLI-----: Output: Done
```

Irving-cl, 2022-02-15:
Hi @suveshpratapa,

Thanks for the log! It seems that the 'time offset' between Host and RCP isn't correct. I just checked the implementation, `radioNow` on host is obtained as:
```
  template <typename InterfaceType, typename ProcessContext }                                                                                                                                                                                                                         
  uint64_t RadioSpinel<InterfaceType, ProcessContextType>::GetNow(void)                                                                                                                                                                                                               
  {                                                                                                                                                                                                                                                                                   
      return mIsTimeSynced ? (otPlatTimeGet() + static_cast<uint64_t>(mRadioTimeOffset)) : UINT64_MAX;                                                                                                                                                                                
  }
 ```
`radioNow` should be a few seconds or minutes larger than `aChild.GetLastRxTimestamp()`. However in our case, `radioNow` (23084977) is smaller than `aChild.GetLastRxTimestamp()` (112642252) which is incorrect. 

`aChild.GetLastRxTimestamp()` comes from the timestamp (otRadioFrame::mInfo.mRxInfo.mTimestamp) which comes from the radio callback `otPlatRadioReceiveDone`. The `mRadioTimeOffset` is calculated at first by using a 'remote timestamp'. The 'remote timestamp' is obtained by running `otPlatTimeGet` on RCP. Could you help check, **on RCP**, in silabs radio driver, if the `otRadioFrame::mInfo.mRxInfo.mTimestamp` and `otPlatTimeGet` uses the same clock source to get the time?

If so,  could you add some more logs (sorry for bothering again):
* In `void RadioSpinel<InterfaceType, ProcessContextType>::CalcRcpTimeOffset(void)`, add a log to print `remoteTimestamp`, `localRxTimestamp`, `localTxTimestamp` and `mRadioTimeOffset`.
* In `uint64_t RadioSpinel<InterfaceType, ProcessContextType>::GetNow(void) `, print `mIsTimeSynced`, `otPlatTimeGet()` and `mRadioTimeOffset`.
* Still keep the logs we added in my last comment.

Let's find out why `otPlatRadioGetNow` doesn't get a time close to `aChild.GetLastRxTimestamp()`.

suveshpratapa, 2022-02-15:
@Irving-cl :

> otPlatTimeGet() must be implemented on both host and RCP.

I see we are actually not implementing this API on the RCP.  I'm checking what it uses as default instead, and will work on testing with it enabled.  I'll also make sure it uses the correct time case for the lastRxTimestamp as well.  I'll report back, thank you for elaborating the calculation.

@George-Stefan :

> On the host, otPlatRadioGetBusSpeed() must be implemented.

Thank you! I think this should be automatically happening, because this depends on the following call for POSIX:
./src/posix/platform/hdlc_interface.hpp:    uint32_t GetBusSpeed(void) const { return mBaudRate; }
and `mBaudRate` is derived from the `uart-baudrate` parameter of the radio URL.  I checked that this is working fine.

> On RCP, OT_RADIO_CAPS_TRANSMIT_TIMING is mandatory

In our case, our radio currently doesn't support sending with a TransmitAt API, so we depend on software timing and let openthread take care of timing the transmit and informing the RCP to send when it is ready.  This is why we disable `OT_RADIO_CAPS_TRANSMIT_TIMING`.

Are you implying that even if this is the case, we should enable `OT_RADIO_CAPS_TRANSMIT_TIMING` for RCP as a special case?

George-Stefan, 2022-02-15:
If you go in RCP without `OT_RADIO_CAPS_TRANSMIT_TIMING`, you assume no overhead/delay in OT stack (alarm events) and no overhead to put the radio in transmit mode except CCA (8 symbols).
Let's say that OT stack has no delays, but the radio usually has a warmup time to enter Tx.
`OT_RADIO_CAPS_TRANSMIT_TIMING` allows you to precisely control the timing of a CSL transmitter.

suveshpratapa, 2022-02-15:
@George-Stefan 

> If you go in RCP without `OT_RADIO_CAPS_TRANSMIT_TIMING`, you assume no overhead/delay in OT stack (for alarm events) and no overhead to put the radio in transmit mode except CCA (8 symbols). Let's say that OT stack has no delays, but the radio usually has a warmup time to enter Tx. `OT_RADIO_CAPS_TRANSMIT_TIMING` allows you to precisely control the timing of a CSL transmitter (must be implemented in radio.c or below it - MAC library).
> 
> See `SubMac::StartCsmaBackoff()`

Thank you! Yes, we have currently defaulted to using software tx timing. With the change to implement the `otPlatTimeGet` API on the OTBR, we're now seeing communication work as expected with our CSL test vectors for certification.

We are looking into fixing our radio layer to implement TransmitAt, but wanted a temporary solution until that's in place; so thanks for pointing us in the right direction.

suveshpratapa, 2022-02-15:
@Irving-cl Thank you so much to you and @George-Stefan for pointing us in the right direction! I had ignored the use of otPlatTimeGet, as I didn't see it directly called in any of the mac/sub-mac source code, but I can now see that it is integral in retrieving the RX timestamp and therefore the right offset against the RCP.

Implementing this API seemed to be the last piece of the puzzle, we can now see uninterrupted communication between OTBR and MED child as expected.  I will continue testing, but for now, I wanted to show a sample of a couple of runs so you can check that the timing looks right:

```
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.965 [INFO]-PLAT----: > ping fe80:0:0:0:b8c9:d32c:bc20:6863
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.965 [NOTE]-CLI-----: Input: ping fe80:0:0:0:b8c9:d32c:bc20:6863
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.965 [INFO]-ICMP----: Sent echo request: (seq = 1)
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.981 [INFO]-PLAT----: GetNow Synced:1 Now: 206936413 Radio_Time_Offset: -8024004 
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.981 [INFO]-PLAT----: GetNow now=198912409
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.981 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 198912409 clrs: 182224689 cp: 1377 ftw: 182445009 ntw: 198945009
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.981 [INFO]-MAC-----: RescheduleCslTx delay: 20183 us. cslTxDelay: 16720320 us.
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:09.982 [INFO]-MAC-----: Prepping indir tx IPv6 ICMP6 msg, len:56, chksum:0000, to:0xf001, sec:yes, prio:normal
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.001 [INFO]-PLAT----: GetNow Synced:1 Now: 206956409 Radio_Time_Offset: -8024004 
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.001 [INFO]-PLAT----: GetNow now=198932405
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.001 [INFO]-MAC-----: GetNextCslTransmissionDelay rn: 198932405 clrs: 182224689 cp: 1377 ftw: 182445009 ntw: 198945009
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.001 [INFO]-MAC-----: HandleFrameRequest Csl txDelay: 16720320 us. lastRxTimestamp: 182224689 us.
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.033 [DEBG]-MAC-----: =========================[TX seqno=133 len=052]=========================
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [DEBG]-MAC-----: | 29 EC 85 CE FA 63 68 20 | BC 2C D3 C9 BA D2 88 68 | )l.Nzch <,SI:R.h
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [DEBG]-MAC-----: | F5 BD D2 B0 E6 0D 03 00 | 00 00 01 7A 33 3A 80 00 | u=R0f......z3:..
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [DEBG]-MAC-----: | 74 41 00 01 00 01 00 03 | 28 48 00 00 00 00 C1 CD | tA......(H....AM
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [DEBG]-MAC-----: | 59 F9 B9 6A .. .. .. .. | .. .. .. .. .. .. .. .. | Yy9j............
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [DEBG]-MAC-----: ------------------------------------------------------------------------
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [INFO]-MAC-----: Sent IPv6 ICMP6 msg, len:56, chksum:0000, to:bac9d32cbc206863, sec:yes, prio:normal
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [INFO]-MAC-----:     src:[fe80:0:0:0:e4b0:d2bd:f568:88d2]
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.034 [INFO]-MAC-----:     dst:[fe80:0:0:0:b8c9:d32c:bc20:6863]
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.136 [INFO]-PLAT----: RCP => [INFO]-PLAT----: Received 60 bytes
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.171 [DEBG]-MAC-----: =========================[RX seqno=120 len=060]=========================
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.171 [DEBG]-MAC-----: | 29 EE 78 CE FA D2 88 68 | F5 BD D2 B0 E6 63 68 20 | )nxNzR.hu=R0fch 
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.171 [DEBG]-MAC-----: | BC 2C D3 C9 BA 0D 05 00 | 00 00 01 04 0D 8A 09 35 | <,SI:..........5
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.172 [DEBG]-MAC-----: | 0C 80 3F 7A 33 3A 81 00 | 73 41 00 01 00 01 00 03 | ..?z3:..sA......
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.172 [DEBG]-MAC-----: | 28 48 00 00 00 00 5B 90 | BF 82 7F DA .. .. .. .. | (H....[.?..Z....
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.172 [DEBG]-MAC-----: ------------------------------------------------------------------------
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.172 [INFO]-MAC-----: Received IPv6 ICMP6 msg, len:56, chksum:0000, from:bac9d32cbc206863, sec:yes, prio:normal, rss:-44.0
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.173 [INFO]-MAC-----:     src:[fe80:0:0:0:b8c9:d32c:bc20:6863]
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.173 [INFO]-MAC-----:     dst:[fe80:0:0:0:e4b0:d2bd:f568:88d2]
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.173 [NOTE]-CLI-----: Output: 16 bytes from fe80:0:0:0:b8c9:d32c:bc20:6863: icmp_seq=1 hlim=64 time=208ms
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.174 [NOTE]-CLI-----: Output: 1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 208/208.0/208 ms.
Feb 15 16:36:57 supratap-raspi2 otbr-agent[550]: 00:03:10.175 [NOTE]-CLI-----: Output: Done
```

```
[... cut ...]
Feb 15 16:37:08 supratap-raspi2 otbr-agent[550]: 00:03:21.165 [INFO]-MAC-----:     dst:[fe80:0:0:0:e4b0:d2bd:f568:88d2]
Feb 15 16:37:08 supratap-raspi2 otbr-agent[550]: 00:03:21.166 [NOTE]-CLI-----: Output: 16 bytes from fe80:0:0:0:b8c9:d32c:bc20:6863: icmp_seq=3 hlim=64 time=947ms
Feb 15 16:37:08 supratap-raspi2 otbr-agent[550]: 00:03:21.166 [NOTE]-CLI-----: Output: 1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 947/947.0/947 ms.
Feb 15 16:37:08 supratap-raspi2 otbr-agent[550]: 00:03:21.166 [NOTE]-CLI-----: Output: Done
```

suveshpratapa, 2022-03-02:
@jwhui Yes, please go ahead and resolve this issue! Thanks for all the advice and tips.

closed 2022-03-02 by suveshpratapa