openthread/openthread #6987: batch devices joining thread network

wfchair, 2021-09-06:
Can we use openthread to do Thread 1.2 Commercial Extensions for large scale authentication, network joining?

For now, we use thread device QR code scanning to join thread network, 
but we hope some way to do batch joining instead of scanning one by one.

LancerSky, 2021-09-06:
You can refer to the following commands.
joiner add
Usage: commissioner joiner add <eui64>|<discerner> <pskd> [timeout]

Add a Joiner entry.

eui64: The IEEE EUI-64 of the Joiner or '*' **to match any Joiner**.
discerner: The Joiner discerner in format number/length.
pskd: Pre-Shared Key for the Joiner.
timeout: joiner timeout in seconds.

jwhui, 2021-09-23:
> Can we use openthread to do Thread 1.2 Commercial Extensions for large scale authentication, network joining?

Thread 1.2 Commercial Extensions supports certificate-based, enterprise-scale network access control. To learn more, see the Thread Group's [Thread 1.2 in Commercial White Paper](https://www.threadgroup.org/Portals/0/documents/support/ThreadInCommercialWhitePaper_2542_1.pdf). However, at this time, the OpenThread project does not have an implementation of the commercial commissioning joiner role.

> For now, we use thread device QR code scanning to join thread network,
> but we hope some way to do batch joining instead of scanning one by one.

Whether it's Thread 1.1 Commissioning or Thread 1.2 Commercial Commissioning, both methods still require explicitly indicating which devices may join the network. Note that use of a QR code is just one way to obtain device-specific PSKd and you can use other ways to communicate the PSKd.

jwhui, 2021-11-06:
Closing stale issue.

closed 2021-11-06 by jwhui