openthread/openthread #7000: Radio coprocessor is unaware of a dead host

dismirlian, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
Consider a network consisting of 1 FTD (POSIX daemon + RCP) and 1 MTD. The FTD assumes the Leader role and the MTD is its child.

If, for whatever reason, ot-daemon dies, the radio coprocessor keeps responding to the MTD's data polls, so the MTD is never detached. This occurs because the RCP is unaware that the daemon is dead.

I don't know if there is a problem with our platform implementation, or this is effectively a problem.

In our case, this is a very undesirable scenario, because the battery-powered MTDs could make better choices to conserve power if they were aware that their parent is unavailable.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
In the ideal scenario, there would be a formal keep-alive between the RCP and the host; if the host dies, the OpenThread library in the RCP could take some action (eg. turn off the radio, reset the whole RCP, or any other customized action).

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
Use any incoming traffic from the host as a "keepalive", and timeout if no message is received for X seconds. Two questions arise:
- Is there a guarantee that the host will send a Spinel message to the RCP within a maximum period of time, and in every scenario (eg. when the FTD is leader, router, child, etc)?
- If the response to the previous question is affirmative, what would be the timeout?

**Additional context** Add any other context or screenshots about the feature request here.
None.

jwhui, 2021-09-10:
@dismirlian , the fundamental issue here is that MAC Data Request messages are a MAC-layer concept. However, we want to validate the parent-child relationship, which operates at a higher layer.

Thread 1.1.1 Section 4.7.5 actually states:
> Rx-off-when-idle Children SHOULD also periodically (at time intervals indicated by the application) resynchronize by sending a Child Update Request to their Parent to avoid cases when the MAC Data Request is being acknowledged erroneously by a receiver that is not a valid Parent.

You can also consider leveraging OpenThread's [Child Supervision feature](https://openthread.io/guides/build/features/child-supervision).

Both of the mechanisms above should help address your issue.

dismirlian, 2021-09-10:
Hi @jwhui excellent, thank you.

closed 2021-09-10 by dismirlian