openthread/openthread #7013: The child device can only Get IPv6 address with Prefix Length 64 bit using DHCPv6, cannot when prefix length is bigger than 64

HeroHerbert, 2021-09-16:
**Describe the bug** A clear and concise description of what the bug is.
The thread Child Can't get IPv6 address from the router with both DHCP server/client enabled when the prefix length is bigger than 64. (When the prefix is 64, the IPv6 address can be assigned).

process:
   ot-ctl netdata publish prefix fd11:96::/96 pdros med
   ot-ctl netdata show                 (display what is configured at both child and leader devices)
   ot-ctl ipaddr                            

    The syslog shows that the client had sent out DHCPv6 solicit packet, and the server has already replied that packet, yet the ipv6 address is not there.

Please help to resolve .

This is very important for NAT64 application, we want to allocate  ipaddress in some thread network to be of the form
fd11:22::c0a8:0500/128, so it can easily be used in IPV4/V6 interworking place with a simple Stateless NAT64 server.

jwhui, 2021-09-23:
@HeroHerbert , which DHCPv6 Server implementation are you using? Are you using OpenThread's implementation (i.e. [src/core/net/dhcp6_server.cpp](https://github.com/openthread/openthread/blob/main/src/core/net/dhcp6_server.cpp) or something else)? Note that OpenThread's implementation is not a truly stateful server. As a result, does not support prefixes longer than 64.

Also, can you provide a wireshark packet capture to provide more visibility?

HeroHerbert, 2021-09-24:
Dear jwhui, we are using OpenThread's DHCPv6 implementation.

I know currently it doesn't support prefixes longer than 64 from the source code, can it be supported later and when?

You can also find the DHCPv6 solicit & reply packet in the attachment.

Thanks.

jwhui, 2021-09-27:
@HeroHerbert , I am not aware of any current plans to implement a proper stateful DHCPv6 server in OpenThread. For your use case, I would suggest utilizing an existing, more complete DHCPv6 Server implementation.