openthread/openthread #7032: otbr-agent is getting restarted continuously and getting "connect session failed: No such file or directory" upon using ot-ctl

bmathukiya, 2021-09-26:
otbr-agent is restarting continuously in my ubuntu machine. I am following https://openthread.io/guides/border-router/build guide to use otbr in ubuntu. RCP I am using in my custom board. When I am checking status of otbr-agent, sometimes it seems running and sometimes it is not. Due to which ot-ctl gives **"connect session failed: No such file or directory"** 
[image]

`nxp_user@pe-dt647:~$ sudo systemctl status otbr-agent.service
● otbr-agent.service - Border Router Agent
   Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-09-26 14:07:28 IST; 1s ago
 Main PID: 2523 (otbr-agent)
    Tasks: 1 (limit: 4536)
   CGroup: /system.slice/otbr-agent.service
           └─2523 /usr/sbin/otbr-agent -I wpan0 -B enp2s0 spinel+hdlc+uart:///dev/ttyUSB0 trel://enp2s0

Sep 26 14:07:28 pe-dt647 systemd[1]: Started Border Router Agent.
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Running 0.3.0-ea85fce09
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Thread version: 1.2.0
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Thread interface: wpan0
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Backbone interface: enp2s0
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0
Sep 26 14:07:28 pe-dt647 otbr-agent[2523]: [INFO]-UTILS---: Radio URL: trel://enp2s0

nxp_user@pe-dt647:~$ sudo systemctl status otbr-agent.service
● otbr-agent.service - Border Router Agent
   Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2021-09-26 14:07:30 IST; 802ms ago
  Process: 2523 ExecStart=/usr/sbin/otbr-agent $OTBR_AGENT_OPTS (code=exited, status=6)
 Main PID: 2523 (code=exited, status=6)
nxp_user@pe-dt647:~$
`

jwhui, 2021-09-27:
@bmathukiya , thanks for raising this issue.

What hardware platform are you using for the RCP?

Can you provide more complete log output from `otbr-agent`?

Jayesh19101999, 2023-02-13:
even I am also facing the same issue