openthread/openthread #7053: [csl] otPlatRadioGetNow() value after exit from sleep / power saving mode

George-Stefan, 2021-09-30:
From radio.h
```
/**
 * Get the current estimated time (64bits width) of the radio chip.
 *
 * @param[in]   aInstance    A pointer to an OpenThread instance.
 *
 * @returns The current time in microseconds. UINT64_MAX when platform does not support or radio time is not ready.
 *
 */
uint64_t otPlatRadioGetNow(otInstance *aInstance);
```

What is not explicitly stated, but is implied by the way the function is used, is that during power saving/sleep mode this radio counter/timer (returned by otPlatRadioGetNow()) should continue to run so that it'll account for the sleep period.

The problem is that during power saving/sleep mode the radio subsystem clock is turned off so the counter/timer doesn't increment.

Let's clarify the expected behavior of otPlatRadioGetNow() for the above situation.

```
void SubMac::GetCslWindowEdges(uint32_t &ahead, uint32_t &after)
{
    uint32_t semiPeriod = mCslPeriod * kUsPerTenSymbols / 2;
    uint64_t curTime    = otPlatRadioGetNow(&GetInstance());
    uint64_t elapsed;
    uint32_t semiWindow;

    if (mCslLastSync.GetValue() > curTime)
    {
        elapsed = UINT64_MAX - mCslLastSync.GetValue() + curTime;
    }
    else
    {
        elapsed = curTime - mCslLastSync.GetValue();
    }
```

George-Stefan, 2021-09-30:
^^ @doru91, @marius-preda, @mihai-ignat

abtink, 2021-09-30:
@George-Stefan yes. All the time and timer related platform functions (`otPlatAlarmMilliGetNow()`, `otPlatAlarmMicroGetNow()` and the radio one `otPlatRadioGetNow()`)  are expected to continue to run (increase) no matter the state of radio and/or the MCU itself (whether MCU/radio is put in any low-power/sleep states).

Practically the `otPlatAlarmMicroGetNow()` could be used for CSL calculation, but if I recall correctly some radio platforms preferred to have a separate get time API in `otPlatRadio` for this purpose (i.e. `otPlatRadioGetNow()`).

The platform layer can choose to use the same underlying mechanism for the two microsecond time (or use different ones). One thing to note `otPlatAlarmMicroGetNow()` return `uint32_t` but `otPlatRadioGetNow()` expects `uint64_t` (I don't recall/remember why 64-bit was used in the radio API).

jwhui, 2021-11-06:
@George-Stefan , thanks for raising this issue. I submitted #7135 to document that the timer must be free-running. Please review.

PR #7135 (2021-11-08): [docs] clarify `otPlatRadioGetNow()` as free-running
Fixes #7053
closed 2021-11-08 by jwhui