openthread/openthread #7057: Invalid processing of key sequence counter for MLE messages

konradderda, 2021-10-04:
**Describe the bug**
During the processing of secured MLE messages if the key sequence counter is higher than the recipient's current one it is updated regardless of the message type.

According to the specification, Thread Device shall check the type of incoming MLE message in order to decide whether key material should be updated by the following selection:
* Tentative messages (requests) - process the message, discard security material.
* Authoritative messages (responses) - store higher key sequence.
* Peer messages (advertisement) - if a key switch was not expected discard security material.

Currently, the counter is incremented regardless of MLE command's type:
https://github.com/openthread/openthread/blob/31320993fbf477a75c384b16529ea129a1f6d0e5/src/core/thread/mle.cpp#L2798

**To Reproduce** 
1. Git commit id
It's reproducible with the latest OT's codelab docker image's commit : 31320993fbf477a75c384b16529ea129a1f6d0e5.
2. IEEE 802.15.4 hardware platform
It's reproducible either with POSIX simulation or nRF528xx platforms.
3. Reproduction:
* Simulate a Thread Network containing two nodes according to a guide: https://openthread.io/codelabs/openthread-simulation#2
* Form the network
* Check leader's key sequence counter: `keysequence counter` (expected value: 0)
* Disable child's interface: `ifconfig down`
* Change child's counter: `keysequence counter 50`
* Enable child's interface: `ifconfig up`
* Check leader's key sequence counter: `keysequence counter` (expected value: 0, actual value: 50)

**Expected behavior**
The leader's key sequence counter should not be updated by the child. With such behavior, the network can be disrupted by a device with the same network key and higher key sequence counter, e.g. by moving from another thread partition without being properly cleared.

I would like to propose a fix for this issue. Any thoughts or comments are appreciated.

jwhui, 2021-11-06:
@konradderda , thanks for raising this issue. Please feel free to submit a proposed fix.

jwhui, 2023-03-02:
Resolved by #7672

closed 2023-03-02 by jwhui