openthread/openthread #7063: Is there a way to gracefully leave a partition?

purvisb, 2021-10-10:
When manually disconnecting a node - it appears that I have to wait for the age of the last MLE advertisement from that node to exceed some threshold before the partition determines it has been disconnected. 

Is there some way to have a node leave a partition and have that partition immediately informed of this change? Other then having nodes send out advertisements more frequently.

I Tried to use `MleRouter::RemoveNeighbor` in [this fork](https://github.com/purvisb/openthread/commit/348c88b17c59a4bd654e4ab65201f8b5b23b3f47) but that just breaks the link and immediately reforms it.

One idea i had was to manipulate an MLE advertisement from the disconnecting node to advertise its link quality as very low and have the network decide to break the connection on its own volition. I would do this as opposed to manually settings the txpower since that would mean the OT partition might not actually receive the advertisement. However I'm trying to parse through `MleRouter::SendAdvertisement` and its not clear to me how I would manipulate the link quality like that.

purvisb, 2021-10-10:
I figured out how to make a MLE advertisement that sets the router cost to infinity, and sets the neighbor Out/In Link quality to 2 for all routing table Entries. I can share that fork later. But the OT network isn't responding to this. I do see the router table updating with the new info but no response is made. 

I looked at instances of where a router is removed and I found https://github.com/openthread/openthread/blob/39fca696500e8dabbd30001cdbb868201360229c/src/core/thread/mle_router.cpp#L1884-L1889

Maybe I can take advantage of that by breaking the link, creating a link request, and then make the node not respond after that?

purvisb, 2021-10-10:
I tried to stop the OT device from sending a link accept in [this fork](https://github.com/purvisb/openthread/commit/4e518bcd947fc1c64c0ddaf5f2e2c2342488c116). But actually that appears to be sent by the commissioning node (is that the right word?) and not by the joiner node. So that wouldn't make a difference.

purvisb, 2021-10-11:
I programmed a button to send a udp multicast command to all nodes to remove its rloc16 from their neighbor table in [this fork](https://github.com/purvisb/openthread/pull/5/files). This does remove it from `neighbor tables` but the openthread partition still doesn't seem to respond right away. Am I missing something?

jwhui, 2021-10-12:
Thread 1.1.1 Section 5.9.10.3 specifies a mechanism for sending a message to the Leader and releasing the Router ID.

You can find an implementation of the Address Release mechanism in [src/core/thread/mle_router.cpp](https://github.com/openthread/openthread/blob/9e26a5deab8d8af348a2db032d78b14f5541067c/src/core/thread/mle_router.cpp#L3636).

purvisb, 2021-10-14:
I implemented that. Thank you!

closed 2021-10-14 by purvisb