openthread/openthread #7126: With the update otbr devices do not connect

kenvals, 2021-10-31:
Hello during the last year I have been working on my device with otbr installed about 1.5 years ago. Everything was great.
On standalone devices I used NRF SDK 1.0 there was a great opportunity to use joiner. I pressed the button that called the function
`otError error = otJoinerStart()`
and at the entrance to we will decide the commissioner, I saw the requests from the device.

With the upgrade, the function of adding devices directly from the GUI appeared, but nothing gets added to my network, tell me where to dig and will my work be wasted for 1.5 years?

kenvals, 2021-11-04:
I am trying to use device additions via the web interface by entering the device key (PSKd)ю As a result of connecting, I get a permanent error
`Join operation is failed. Please make sure this joiner has been added by an active commissioner.`
what am I doing wrong?

jwhui, 2021-11-10:
Using the latest OTBR and OpenThread code, I just tried joining my OTBR to an existing network and it worked fine.

On a factory reset OT CLI node:
1. dataset init new
2. dataset commit active
3. ifconfig up
4. thread start
5. commissioner start
6. commissioner joiner add * J01NME

On OTBR:
1. Go to Join page.
2. Click "Join" on the network created by the OT CLI node.
3. Select "PSKd" as the credential type.
4. Enter J01NME as the PSKd

Can you provide more detail on the steps you used?

jwhui, 2021-12-22:
Closing stale issue.

closed 2021-12-22 by jwhui