openthread/openthread #7140: [TREL] Question: shall we allow `otPlatTrelUdp6Init` failed at first

Irving-cl, 2021-11-09:
**Is your feature request related to a problem? Please describe.**
If passed with an infra interface (for example, `br-lan`), the Thread daemon (or `otbr-agent`) will not start successfully if anything goes wrong in `otPlatTrelUdp6Init`. For example, the interface cannot be found. And it will log a critical error.

We have many common cases like this: The Thread daemon is set as a service and is started during system initialization. However at this moment, the interface isn't ready yet. (It's hard to make sure the interface to be ready before starting Thread). And a few moments later, the interface becomes ready.

In these cases, we hope the Thread daemon can still work (without TREL functionality). So shall we allow `otPlatTrelUdp6Init` to fail when the interface doesn't exist instead of exiting directly?

**Describe the solution you'd like** A clear and concise description of what you want to happen.
My proposal is to add an error code to `otPlatTrelUdp6Init`, and thus add it to `ot::Trel::Interface::Init`. And we add another timer in `ot::Trel::Link`. Then in `ot::Trel::Link::AfterInit`:
```
void Link::AfterInit(void)
{
     if (mInterface.Init() != OT_ERROR_NONE)
     {
         mTrelInterfaceInitTimer.Start(maybe_five_minutes)
     }
}

void TrelInterfaceInitTimerHandler(void)
{
    if (!mInitialized)
    {
        if (otPlatTrelUdp6Init != OT_ERROR_NONE)
        {
            // Log some errors
            mTrelInterfaceInitTimer.Start(maybe_five_minutes)
        }
    }
}
```

Thoughts? @abtink @simonlingoogle

abtink, 2021-11-09:
@Irving-cl , the current TREL platform definitions and it implementation (which uses the LL multicast/broadcast) is changed (no longer the model is the Thread spec). The new Thread spec draft is based on DNS-SD to discover TREL peers.  Please check this PR:
- https://github.com/openthread/openthread/pull/7125.

Irving-cl, 2021-11-09:
Great! Sounds good to me. Then I think there won't be such problem.

closed 2021-11-09 by Irving-cl