openthread/openthread #7150: Can I send TCP or UPD packets from Cloud to a RCP/Node?

M0hammedImran, 2021-11-10:
- I am using Raspberry 3b and a `Silicon Labs` Dev board with a wireless gecko `EFR32xG21` to form a Border Router.
- I have created a network with default values and connected 2 other nodes to create a thread network.

Now I want to how and if I can send `TCP/UDP` packets to the `nodes` through border router or directly using the `ipv6` addresses from my `EC2` or any other server for that matter.

I have a node `express` server running and I want to log and send some commands to those nodes.

jwhui, 2021-11-10:
To route packets between your cloud instance and Thread network, you'll need to ensure that your IPv6 network is properly configured with routable addresses and routes. If you have that in place, then you should be able to achieve your goal.

M0hammedImran, 2021-11-11:
> To route packets between your cloud instance and Thread network, you'll need to ensure that your IPv6 network is properly configured with routable addresses and routes. If you have that in place, then you should be able to achieve your goal.

Is there a resource or doc I can use to configure my IPv6 network.

jwhui, 2021-11-11:
> Is there a resource or doc I can use to configure my IPv6 network.

Take a look at https://www.internetsociety.org/deploy360/ipv6/basics/

jwhui, 2021-11-23:
Closing stale issue.

closed 2021-11-23 by jwhui