openthread/openthread #7151: Assert when childip max set to 1

canisLupus1313, 2021-11-10:
**Describe the bug** A clear and concise description of what the bug is.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 06ca16a38f1e78a57065df0439fbe52a88e86c1e
1. IEEE 802.15.4 hardware platform Nordic
1. Network topology same as DUA-TC-18 Nordic DUT Nordic TB
Resently during running the DUA-TC-18 we have seen that [this](https://github.com/openthread/openthread/blob/06ca16a38f1e78a57065df0439fbe52a88e86c1e/src/core/thread/dua_manager.cpp#L493) assert fails for router in topology when childip max is set to 1. 

This check confuses me a little since this it looks like the function `GetDomainUnicastAddress` can return null pointer if the router device won't accept more than one child IP. Could someone explain why it was handled this way?

jwhui, 2021-11-10:
@canisLupus1313 , thanks for raising this issue.

The Thread 1.2.0 Conformance Specification requires routers to support at least 4 IPv6 addresses per MTD child. However, I understand that not reflecting the DUA is required for certification testing.

@simonlingoogle @librasungirl , thoughts?

simonlingoogle, 2021-11-11:
This is a `assert` failure observed before. 

Need further investigation.

PR #7164 (2021-11-15): [dua] assume child does not have DUA if no space to store one
Fixes #7151
closed 2021-11-15 by jwhui