openthread/openthread #7155: SSED device does not send ChildUpdateRequest to parent after receiving Data Response with updated network data

hemanth-silabs, 2021-11-11:
Test case **5_9_20: DUA-TC-20** tests thread devices ability to manually configure address. As part of this test case the PBBR registers a new DUA after the topology has been established. One of the test steps is for a MTD DUT is to receive the new BBR dataset, after the DUA is registered on the PBBR and to send `ChildUpdateRequest` to PBBR with new DUA address.

The test fails with a SSED as DUT and using Kirale PBBR. In this scenario, the Kirale BBR sends a `Data Response` packet as unicast to the SSED to report the BBR dataset which the SSED acknowledges. But the SSED fails to send `ChildUpdateRequest` to report the new dataset.

This test works when testing manually with a OT BBR, the difference being OT BBR sends a `ChildUpdateRequest` to inform the changed dataset instead of `Data Response`.

hemanth-silabs, 2021-11-11:
I am now trying to get the OT BBR to send `DataResponse` instead of `ChildUpdateRequest` using the following update to  in `SynchronizeChildNetworkData` 

```
static const uint8_t tlvs[] = {Tlv::kNetworkData};
Ip6::Address destination;

memset(&destination, 0, sizeof(destination));
child.GetMeshLocalIp6Address(destination);
SendDataResponse(destination, tlvs, sizeof(tlvs), 0);
```

This does not seem to work. Any input on this will be helpful in debugging too.

hemanth-silabs, 2021-11-11:
@jwhui @simonlingoogle @suveshpratapa

jwhui, 2021-11-11:
@hemanth-silabs , thanks for raising this issue. Can you also post the test result report, pcap files, and logs?

hemanth-silabs, 2021-11-11:
Sure @jwhui. Here we go.
[C_5_9_20_SSED.zip](https://github.com/openthread/openthread/files/7521633/C_5_9_20_SSED.zip)

jwhui, 2021-11-11:
@hemanth-silabs , do you also have the test artifacts for the passing scenario with OT BBR?

hemanth-silabs, 2021-11-11:
No. These were manually run tests not through the test harness. But easily reproducible using the following steps

On BBR create a network:
```
channel 11
panid 0xFACE
networkkey 00112233445566778899aabbccddeeff
ifconfig up
thread start
```

SSED joins the network using:
```
dataset channel 11
dataset panid 0xface
dataset networkkey 00112233445566778899aabbccddeeff
mode -
csl period 3125
csl timeout 30
dataset commit active
ifconfig up
thread start
dua iid 0004000300020001
```

Add new DUA prefix on BBR:
```
prefix add fd00:7d03:7d03:7d03::/64 pasroD med
netdata register
```

jwhui, 2021-11-11:
> I am now trying to get the OT BBR to send `DataResponse` instead of `ChildUpdateRequest` using the following update to in `SynchronizeChildNetworkData`
>
> ```
> static const uint8_t tlvs[] = [...]

The MLE message needs to be destined to the link-local address. You can do this:

```c++
static const uint8_t tlvs[] = {Tlv::kNetworkData};
Ip6::Address destination;

memset(&destination, 0, sizeof(destination));
destination.SetToLinkLocalAddress(child.GetExtAddress());
SendDataResponse(destination, tlvs, sizeof(tlvs), 0);
```

I just tested with your example scenario above and that seemed to work.

jwhui, 2021-11-12:
@hemanth-silabs , I've tried looking into this again, but I cannot see anything obvious why the SSED would not send an MLE Child Update Request after receiving updated network data.

Can you double check that you have `OPENTHREAD_CONFIG_DUA_ENABLE` set to 1 in your build?

Upon receiving updated network data with the DUA prefix, you should see this code triggered:
https://github.com/openthread/openthread/blob/5e0d1b1a32d8df616d58db73cfc551334a6bf711/src/core/backbone_router/bbr_leader.cpp#L252

You could try debugging from there.

Also, any chance you can provide some more details logs?

hemanth-silabs, 2021-11-12:
@jwhui Thank you for the suggestion about sending the network data response. Following this, I was able to manually test the scenario using OTBR, and as pointed out SSED behaved as expected. 

Also this evening, we were able to re-run the test case against Kirale node based on the newer Beta 10 harness and the SSED passes the test in the new harness. I suspect this was a harness related bug.

Thank you for all your inputs. I will close this issue now.

closed 2021-11-12 by hemanth-silabs

jwhui, 2021-11-12:
@hemanth-silabs , great to hear the issue was resolved. Thanks for reporting back.