openthread/openthread #7159: Multicast expiration time takes double the interval.

canisLupus1313, 2021-11-12:
1. Git commit id 543e32d2aef7eda8d66396c93476ce0da4e13c5b
1. Network topology BBR + Thread device + Host PC

**Reproduction steps**
1. Properly setup thread network.
1. Register the multicast address in Thread Device.
1. Ping the multicast address for m Host PC.
1. Wait for the multicast address to expire.

After the steps above I can observe that expiration time takes from 600s to 660s seconds while it should take 300s.
The reason for that is the `ExpireMulticastForwardingCache` it checks and updates the usage counters every expiration period due to 
[this](https://github.com/openthread/openthread/blob/main/src/posix/platform/multicast_routing.cpp#L394) check. Moving `&& mfc.mLastUseTime + kMulticastForwardingCacheExpireTimeout * US_PER_S < now` to [this](https://github.com/openthread/openthread/blob/main/src/posix/platform/multicast_routing.cpp#L396) if shortens the expiration period to 300s to 360s (it still fails the MATN-TC-14 in certification). The extra 60s comes from [this](https://github.com/openthread/openthread/blob/main/src/posix/platform/multicast_routing.cpp#L385) check this looks like optimization. It's easy to fix using C++ collections.

So to sum up are we allowed to use c++ collections in this part of code? I can see that adding there eg. std vector links properly and this code is meant for POSIX based platforms.

simonlingoogle, 2021-11-12:
Do we remove the multicast address between step 3 and 4?
I think the multicast address registration will be renewed if we didn't remove it. 

As for the `ExpireMulticastForwardingCache`, it is used to expire MFCs that are not actually used when the multicast address is still registered. But if the multicast address is removed, the corresponding MFCs should be removed immediately.

simonlingoogle, 2021-11-13:
> So to sum up are we allowed to use c++ collections in this part of code? I can see that adding there eg. std vector links properly and this code is meant for POSIX based platforms.

We don't have the practise of using C++ STL in posix platform code. 
But I think that should be allowed and even preferred. 
@jwhui @bukepo @wgtdkp @superwhd @abtink

canisLupus1313, 2021-11-15:
@simonlingoogle 
> Do we remove the multicast address between step 3 and 4?

Yes sorry for not stating that.

> But if the multicast address is removed, the corresponding MFCs should be removed immediately.

Section 5.24.6.1 of thread speck says how to register multicast address from a child but there is nothing about deregistration. I think that's why we have this 300 s timeout to check if the route was used. And the reason why we don't have the deregistration message is because many devices can register to one multicast address and it simplifies thing when You don't have to keep trace which ones are still actively listening those and which ones don't.

> Submitted #7167 to fix.
Have you tested this against MATN-TC-14?

simonlingoogle, 2021-11-15:
> > Submitted #7167 to fix.
>
> Have you tested this against MATN-TC-14?

No, I didn't. I don't have a handy testbed for MATN-TC-14. 
And I am not expecting to fix `MATN-TC-14` using #7167. I think #7167 fixes a potential bug that MLD Report may be closed unexpectedly when the MA is still registered.

simonlingoogle, 2021-11-15:
> Section 5.24.6.1 of thread speck says how to register multicast address from a child but there is nothing about deregistration. I think that's why we have this 300 s timeout to check if the route [...]

I am aware of that. 

Could you help upload the failed test result of MATN-TC-14 to have a look?

canisLupus1313, 2021-11-15:
[5_10_10.zip](https://github.com/openthread/openthread/files/7537185/5_10_10.zip)

> potential bug that MLD Report may be closed unexpectedly when the MA is still registered.

You are wright.

simonlingoogle, 2021-11-15:
@canisLupus1313 
Could you rerun MATN-TC-14 with https://github.com/openthread/openthread/pull/7167 fix?
I think the bug may cause device to incorrect send MLD Reports with wrong MAs, which may affect the test result.

canisLupus1313, 2021-11-15:
@simonlingoogle 
When I will get my hands on Cert setup I will try to do it (hopefully today).
BTW which packets are incorrect in your opinion?

simonlingoogle, 2021-11-15:
> When I will get my hands on Cert setup I will try to do it (hopefully today).

Thanks. 

> BTW which packets are incorrect in your opinion?

I haven't looked into the PCAP in depth. It's always painful to diagnose certification issues in a buggy environment. So, I think it's much better if we can have test results with https://github.com/openthread/openthread/pull/7167 fixed.

abtink, 2021-11-15:
> We don't have the practise of using C++ STL in posix platform code. But I think that should be allowed and even preferred. @jwhui @bukepo @wgtdkp @superwhd @abtink

Within core we definitely cannot and should not use C++ STL. I would also advise against using C++ STL in posix platform. 
As it may add limitation on the platforms that want to port/use the posix platform (we had faced such limitation in past in some internal projects where STL was not available/blocked).

canisLupus1313, 2021-11-16:
@abtink Thanks for clarification.

@simonlingoogle Your PR #7167 gives the same results for this test case. So I think it's ok to merge it.
Moreover I think I know already what was wrong with my testing locally I have changed [this](https://github.com/openthread/openthread/blob/main/src/posix/platform/multicast_routing.hpp#L70) timeout to lower value without chaneing [this](https://github.com/openthread/openthread/blob/main/src/core/thread/mle_types.hpp#L232) one which actually showed the above behavior. Sorry for the confusion and thanks for help.

closed 2021-11-16 by canisLupus1313

simonlingoogle, 2021-11-16:
> @abtink Thanks for clarification.
> 
> @simonlingoogle Your PR #7167 gives the same results for this test case. So I think it's ok to merge it. 

Thanks for verification. 

> Moreover I think I know already what was wrong with my testing locally I have changed [this](https://github.com/openthread/openthread/blob/main/src/posix/platform/multicast_routing.hpp#L70) timeout [...]

Is MATN-TC-14 passed after reverting temporary changes?

canisLupus1313, 2021-11-16:
Nope the test fails in both cases results in attachement.
[5_10_10_rerun.zip](https://github.com/openthread/openthread/files/7545491/5_10_10_rerun.zip)