openthread/openthread #7162: Question about Link-Local IPV6 Address for a Thread device

Olivier-STMicroelectronics, 2021-11-12:
I have a question regarding the IPV6 address to use for a Thread device:

I have a Thread device that I have commissioned to a Thread network thanks to OpenThread application. So basically I have the following devices:
- A Thread device freshly connected to a Thread network
- A Border Router
- An Android phone, connected to the Border Router through the Wifi.

So there is a Thread network and a Wifi network. The Border router being the gateway between those 2 networks.

For the moment, my Thread device only has a Link-Local IPV6 Address. If my understanding is correct, this Link-Local Address cannot be reached by the Android phone because the router will not forward a command to a link-local address of the Thread network. Is it correct? 
I guess that I should allocate a Unique Local address (or even a Global unicast address?) for the Thread device. Can you please confirm?

Thank you
Regards
Olivier

jwhui, 2021-11-12:
Thread devices link-local scope as interfaces that are reachable by a single radio transmission.

Wi-Fi / Ethernet defines link-local scope as interfaces that are directly attached to the Wi-Fi / Ethernet network.

Because Thread devices are not directly attached to the Wi-Fi / Ethernet network, link-local addressing cannot be used to communicate between Thread and Wi-Fi / Ethernet devices. You will need something with global scope (e.g. unique local or publicly routable addresses).

For more information:
1. See the [Thread Primer - IPv6 Addressing](https://openthread.io/guides/thread-primer/ipv6-addressing) for more details.
2. Check out our codelab on [Thread Border Router - Bidirectional IPv6 Connectivity and DNS-Based Service Discovery](https://openthread.io/codelabs/openthread-border-router#0)

OlivierGre, 2021-11-15:
Thanks for the links and information Jonathan!

Would you have some reading explaing how to create and use the OMR prefix ? I can see on https://openthread.io/codelabs/openthread-border-router#2 that a Unique Local address (fd76:a5d1:fcb0:1707:f3c7:d88c:efd1:24a9 in the given example), starting with the OMR prefix, can be reached from the Wi-Fi/Ethernet network.
In my environment (not based on nRF52840 boards), when I call "netdata show" command on the Border Router, I have a Service but I have nothing in Prefixes and Routes so I think that I lack an OMR prefix.

Update: I have read https://github.com/openthread/openthread/blob/main/src/cli/README_NETDATA.md explaining how to register an IPv6 prefix assigned to the Thread network.

BTW, I'm also wondering if I should install _**radvd RA daemon**_ on my raspberry pi. I believe that radvd RA daemon might be useful if I want the Border Router to forward the IPV6 commands to the Thread device.

Thank you

jwhui, 2021-11-15:
> Would you have some reading explaing how to create and use the OMR prefix ? I can see on https://openthread.io/codelabs/openthread-border-router#2 that a Unique Local address [...]

If you are using the latest https://github.com/openthread/ot-br-posix main branch, the OMR prefix should be generated automatically by OTBR. You should be able to follow the codelab without any additional instructions.

OlivierGre, 2021-11-16:
For the moment we are using Thread 1.1 and we cannot move yet to 1.2.
Is there a version of https://github.com/openthread/ot-br-posix based on Thread 1.1 spec and having OMR prefix generated automatically?
Thank you

simonlingoogle, 2021-11-16:
> Is there a version of https://github.com/openthread/ot-br-posix based on Thread 1.1 spec and having OMR prefix generated automatically?

I think that's possible by specifying setup options. You may try:
```
INFRA_IF_NAME=wlan0 BACKBONE_ROUTER=1 OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1" ./script/setup
```

OlivierGre, 2021-11-16:
Thank you, I'm going to try.

OlivierGre, 2021-11-18:
I have built the OTBR with:

```
cd ot-br-posix
./script/bootstrap
INFRA_IF_NAME=wlan0 OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1" ./script/setup
```
I have then connected my RCP and rebooted.
When I check the status I see an error:

```
pi@raspberrypi:~/ot-br-posix $ sudo service otbr-agent status
● otbr-agent.service - Border Router Agent
     Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Running 0.3.0-4555d899e
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Thread version: 1.2.0
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Thread interface: wpan0
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Backbone interface: wlan0
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: [INFO]-UTILS---: Radio URL: trel://wlan0
Nov 18 10:06:39 raspberrypi otbr-agent[2047]: 49d.17:07:44.261 [CRIT]-PLAT----: Init() at ../../third_party/openthread/>
Nov 18 10:06:39 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Nov 18 10:06:39 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Nov 18 10:06:39 raspberrypi systemd[1]: Stopped Border Router Agent.
```

I can see that it says "Thread version: 1.2.0" whereas I have asked to use Thread version 1.1. 
Is the option OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1"  working?

Thanks

OlivierGre, 2021-11-18:
> ```
> INFRA_IF_NAME=wlan0 BACKBONE_ROUTER=0 OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1" ./script/setup
> ```
@simonlingoogle I don't know what is "BACKBONE_ROUTER=0" and if I should use it.

OlivierGre, 2021-11-18:
When the setup command is executed, I can see a few times this line:
OPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_2

Isn't that a problem since I want to build against 1.1 version?

OlivierGre, 2021-11-18:
I have built with the line indicated by Simon Lin:
`INFRA_IF_NAME=wlan0 BACKBONE_ROUTER=0 OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1" ./script/setup`

I get the following error:
```
[13/609] Building CXX object third_party/openthread/repo/src/posix/platform/CMakeFiles/openthread-posix.dir/radio.cpp.o
FAILED: third_party/openthread/repo/src/posix/platform/CMakeFiles/openthread-posix.dir/radio.cpp.o 
/usr/bin/c++ -DOPENTHREAD_CONFIG_ASSERT_ENABLE=1 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0 -DOPENTHREAD_CONFIG_BORDER_AGENT_ENABLE=1 -DOPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE=1 -DOPENTHREAD_CONFIG_BORDER_ROUTING_ENABLE=1 -DOPENTHREAD_CONFIG_COAP_API_ENABLE=1 -DOPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE=1 -DOPENTHREAD_CONFIG_COMMISSIONER_ENABLE=1 -DOPENTHREAD_CONFIG_DNSSD_SERVER_ENABLE=1 -DOPENTHREAD_CONFIG_ECDSA_ENABLE=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-posix-config.h\" -DOPENTHREAD_CONFIG_HEAP_EXTERNAL_ENABLE=1 -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1 -DOPENTHREAD_CONFIG_JOINER_ENABLE=1 -DOPENTHREAD_CONFIG_LEGACY_ENABLE=1 -DOPENTHREAD_CONFIG_LOG_CLI=1 -DOPENTHREAD_CONFIG_LOG_LEVEL=OT_LOG_LEVEL_INFO -DOPENTHREAD_CONFIG_LOG_LEVEL_DYNAMIC_ENABLE=1 -DOPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE=0 -DOPENTHREAD_CONFIG_MAC_FILTER_ENABLE=1 -DOPENTHREAD_CONFIG_MAX_STATECHANGE_HANDLERS=3 -DOPENTHREAD_CONFIG_MLE_LINK_METRICS_SUBJECT_ENABLE=1 -DOPENTHREAD_CONFIG_MLE_STEERING_DATA_SET_OOB_ENABLE=1 -DOPENTHREAD_CONFIG_NCP_HDLC_ENABLE=1 -DOPENTHREAD_CONFIG_PING_SENDER_ENABLE=1 -DOPENTHREAD_CONFIG_PLATFORM_NETIF_ENABLE=1 -DOPENTHREAD_CONFIG_PLATFORM_UDP_ENABLE=1 -DOPENTHREAD_CONFIG_POSIX_SETTINGS_PATH=\"/var/lib/thread\" -DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1 -DOPENTHREAD_CONFIG_SRP_SERVER_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1 -DOPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE=1 -DOPENTHREAD_CONFIG_TMF_PROXY_DUA_ENABLE=0 -DOPENTHREAD_FTD=1 -DOPENTHREAD_MTD=0 -DOPENTHREAD_POSIX_CONFIG_DAEMON_ENABLE=1 -DOPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=0 -DPACKAGE_NAME=\"OPENTHREAD\" -Ithird_party/openthread/repo/etc/cmake -I../../third_party/openthread/repo/etc/cmake -I../../third_party/openthread/repo/include -I../../third_party/openthread/repo/src -I../../third_party/openthread/repo/src/core -I../../third_party/openthread/repo/third_party/mbedtls/repo/include -I../../third_party/openthread/repo/src/posix/platform/include -I../../third_party/openthread/repo/src/posix/platform -Ithird_party/openthread/repo/third_party/mbedtls -Wall -Wextra -Werror -Wfatal-errors -Wno-missing-braces -Wshadow -Wno-c++14-compat -fno-exceptions -std=c++11 -MD -MT third_party/openthread/repo/src/posix/platform/CMakeFiles/openthread-posix.dir/radio.cpp.o -MF third_party/openthread/repo/src/posix/platform/CMakeFiles/openthread-posix.dir/radio.cpp.o.d -o third_party/openthread/repo/src/posix/platform/CMakeFiles/openthread-posix.dir/radio.cpp.o -c ../../third_party/openthread/repo/src/posix/platform/radio.cpp
In file included from ../../third_party/openthread/repo/src/core/thread/topology.hpp:55,
                 from ../../third_party/openthread/repo/src/core/thread/child_table.hpp:45,
                 from ../../third_party/openthread/repo/src/core/thread/mle_router.hpp:50,
                 from ../../third_party/openthread/repo/src/core/thread/network_data.hpp:50,
                 from ../../third_party/openthread/repo/src/core/utils/slaac_address.hpp:45,
                 from ../../third_party/openthread/repo/src/core/common/settings.hpp:51,
                 from ../../third_party/openthread/repo/src/core/common/instance.hpp:66,
                 from ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:45,
                 from ../../third_party/openthread/repo/src/lib/spinel/radio_spinel.hpp:1052,
                 from ../../third_party/openthread/repo/src/posix/platform/radio.cpp:39:
../../third_party/openthread/repo/src/core/thread/link_metrics.hpp:42:2: error: #error "Thread 1.2 or higher version is required for OPENTHREAD_CONFIG_MLE_LINK_METRICS_INITIATOR_ENABLE" "and OPENTHREAD_CONFIG_MLE_LINK_METRICS_SUBJECT_ENABLE."
   42 | #error "Thread 1.2 or higher version is required for OPENTHREAD_CONFIG_MLE_LINK_METRICS_INITIATOR_ENABLE" \
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
[18/609] cd /home/pi/ot-br-posix/build/otbr/third_party/op...br-posix/third_party/openthread/repo/etc/cmake/print.cmake

```

OlivierGre, 2021-11-18:
@wgtdkp Thank you

wgtdkp, 2021-11-18:
Hi @OlivierGre 

Can you checkout https://github.com/openthread/ot-br-posix/pull/1099 and rebuild?
Remember to clear build cache with `rm -rf build`.

OlivierGre, 2021-11-18:
Sure, I'm going to do that.

OlivierGre, 2021-11-18:
What commands shall I use please?

I have tried 
```
git fetch
git cherry-pick e623f2a5e0e78e1daf64e7b4296cbf417ea1a50b
```
but this is not correct

OlivierGre, 2021-11-18:
I have installed GitHub CLI (https://lindevs.com/install-github-cli-on-raspberry-pi/) and cherry picked the commit with:
gh pr checkout 1099

OlivierGre, 2021-11-18:
> Hi @OlivierGre
> 
> Can you checkout [openthread/ot-br-posix#1099](https://github.com/openthread/ot-br-posix/pull/1099) and rebuild? Remember to clear build cache with `rm -rf build`.

@wgtdkp With your commit, it builds without error. Thanks

OlivierGre, 2021-11-18:
After the setup command, I check the Border Agent status and I see that it is dead:

```
pi@raspberrypi:~/ot-br-posix $ sudo service otbr-agent status
● otbr-agent.service - Border Router Agent
     Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Running 0.3.0-e623f2a5e
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Thread version: 1.1.1
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Thread interface: wpan0
Nov 18 15:58:27 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Backbone interface: wlan0
Nov 18 15:58:27 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Radio URL: trel://wlan0
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: 49d.17:24:25.669 [CRIT]-PLAT----: Init() at ../../third_party/openthread/>
Nov 18 15:58:29 raspberrypi systemd[1]: Stopped Border Router Agent.
```

At that stage my RCP is not connected.
What can I do to know more about this failure?

I have noticed that the Border Agent says "Thread version: 1.1.1" whereas I had set OT_THREAD_VERSION=1.1. 
Is it expected?

Thank you

wgtdkp, 2021-11-19:
> At that stage my RCP is not connected.

OTBR doesn't run without a RCP device.

> What can I do to know more about this failure?

You can collect the OTBR syslog with `cat /var/log/syslog | grep otbr-agent`.

closed 2021-11-19 by jwhui

OlivierGre, 2021-11-22:
> If you are using the latest https://github.com/openthread/ot-br-posix main branch, the OMR prefix should be generated automatically by OTBR. You should be able to follow the codelab without any additional instructions.

@jwhui Jonathan, do you know when the automatic generation of OMR prefix was added? 

My RCP only supports API version 1. SPINEL_MIN_HOST_SUPPORTED_RCP_API_VERSION was incremented in this commit (https://github.com/openthread/openthread/commit/a94e88cd5081148a8bcc99c5123337b94612f995) done in **August 2021** so I'm wondering if I should step back to the version just before this commit. 
Do you think that OMR prefix was already generated automatically in August? If you know which commits added the generation of OMR prefix, it will help me.

Thank you very much

OlivierGre, 2021-11-22:
I can see that OMR prefix implementation was started in December 2020:
https://github.com/openthread/openthread/commit/995ad66846c9d0326df549842e76f58dad71650e

And updated in this commit in August:
https://github.com/openthread/openthread/commit/5598848980ae80cfaac134e89796cb711e321464

jwhui, 2021-11-22:
Yes, the border routing feature was initially added in December 2020. However, we continue to make enhancements and bug fixes. See https://github.com/openthread/openthread/commits/main/src/core/border_router

If you want to use the border routing feature, I suggest you update your RCP to support the minimum required RCP API version. If you not planning to leverage Thread 1.2 low power enhancements, then updating the RCP should be fairly straightforward.

OlivierGre, 2021-11-23:
Hi Jonathan,
Ok thank you, I'm going to do that. I was also thinking that it was not a good idea to move back to an old version where I would have no guarantee that border routing would work and where I would not get any support.
Regards