openthread/openthread #7169: Thread 1.2 specification in openthread

bhumikaajain, 2021-11-15:
Hi,

From which commit onwards openthread has started to build upon thread 1.2 specifications?
Are all Thread 1.2 specifications incorporated in openthread?

jwhui, 2021-11-15:
We used #4327 to track the initial Thread 1.2 implementation effort.

You can use the [Thread 1.2](https://github.com/openthread/openthread/pulls?q=is%3Apr+label%3A%22Thread+1.2%22) tag to continue following Thread 1.2 related issues and PRs.

bhumikaajain, 2021-11-15:
Thank you for the reply.
My openthread application is currently based on Nordic's nrf52840 hardware. I am currently updated to "nRF5 SDK for Thread and Zigbee v4.1.0". It is based on openthread commit "https://github.com/openthread/openthread/commit/c6a258e3a5bd90aa26d8245c435c0ae5797027f4"

As openthread repo for nrf52840 is changed to ot-nrf528xx now.
I have some more follow up questions. 
Since when these 2 repos are separated ?
Also, does all latest openthread feature commit also get added in ot-nrf528xx repo?
why hardware-based openthread repo for nrf52xx has been introduced?

jwhui, 2021-11-15:
> As openthread repo for nrf52840 is changed to ot-nrf528xx now. I have some more follow up questions. Since when these 2 repos are separated ?

The first commit was in January 2021: https://github.com/openthread/ot-nrf528xx/commit/e4548c86cb664f056fe375b4bb24ce276124500b

> Also, does all latest openthread feature commit also get added in ot-nrf528xx repo? why hardware-based openthread repo for nrf52xx has been introduced?

The openthread repo is included as a submodule and is updated daily - see https://github.com/openthread/ot-nrf528xx/commits/main

jwhui, 2021-11-23:
Closing stale issue.

closed 2021-11-23 by jwhui