openthread/openthread #7175: BBR not replying to multicast ping for own registered address

edmont, 2021-11-16:
**Describe the bug** A clear and concise description of what the bug is.

OT-BR based DUT is failing Thread 1.2 Certification MATN TC 02 in the roles of TD (FED and Router). This is because it doesn't reply to a ping to a greater than realm local scope multicast address which is registered locally via `ipmaddr` CLI command.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
https://github.com/openthread/openthread/commit/19cd73d69121c33d8d65cb0c4524cc5aa9084bd0
https://github.com/openthread/ot-br-posix/commit/fe126357854877121ea24e37505b54ee6f7c70a2

2. IEEE 802.15.4 hardware platform
RPi + nRF52840 dongle

3. Build steps
Add `'is_otbr': True` to TD in the [MATN_02_MLRFirstUse](https://github.com/openthread/openthread/blob/main/tests/scripts/thread-cert/border_router/MATN_02_MLRFirstUse.py#L82-L86) test and proceed with test running instructions (https://github.com/openthread/openthread/blob/main/tests/scripts/thread-cert/border_router/README.md).

4. Network topology
As described in the test.

**Expected behavior** A clear and concise description of what you expected to happen.

DUT should reply to ping and also forward MPL packets.

**Console/log output** If applicable, add console/log output to help explain your problem.

I added some extra logging to `Ip6::HandleDatagram` just after the multicast destination if/else:
```
Nov 16 14:28:42 85343a5150ce otbr-agent[112]: 00:00:46.509 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:152, chksum:0000, from:0xc000, sec:yes, prio:normal, rss:-20.0
Nov 16 14:28:42 85343a5150ce otbr-agent[112]: 00:00:46.509 [INFO]-MAC-----:     src:[fd00:db8:0:0:9c01:ff18:d32a:fa6d]
Nov 16 14:28:42 85343a5150ce otbr-agent[112]: 00:00:46.509 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:fc]
Nov 16 14:28:42 85343a5150ce otbr-agent[112]: 00:00:46.509 [CRIT]-IP6-----: receive 1, forwardThread 0, forwardHost 0
Nov 16 14:28:42 85343a5150ce otbr-agent[112]: 00:00:46.548 [CRIT]-IP6-----: receive 0, forwardThread 1, forwardHost 0
```
What would be the best fix here?

edmont, 2021-11-16:
@simonlingoogle @abtink ?

simonlingoogle, 2021-11-17:
> This is because it doesn't reply to a ping to a greater than realm local scope multicast address which is registered locally via `ipmaddr` CLI command.

Who is the sender of the Ping Requst?

simonlingoogle, 2021-11-17:
It may suffer from the same issue described in https://github.com/openthread/openthread/pull/7145#issuecomment-964254266 :cry: 

When TD subscribes MA, no address callback is invoked so the MA is not reflected on the wpan0 interface, so kernel won't be responding to the Ping Requests. 

I tried using `add_ipmaddr_ether` to add MA1 to wpan0 TUN interface too, but it's still not working (no Ping Response from TD). 

TD tried to send the Ping Response via Thread but failed:
```
Nov 17 06:36:37 65472e64579d otbr-agent[112]: 00:00:46.898 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:152, chksum:0000, from:0x0800, sec:yes, prio:normal, rss:-20.0
Nov 17 06:36:37 65472e64579d otbr-agent[112]: 00:00:46.898 [INFO]-MAC-----:     src:[fd00:db8:0:0:c5a7:68e0:4646:b017]
Nov 17 06:36:37 65472e64579d otbr-agent[112]: 00:00:46.898 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:fc]
Nov 17 06:36:37 65472e64579d otbr-agent[112]: 00:00:46.898 [CRIT]-IP6-----: SendRaw to fd67:5ebb:3377:1f00:f218:8ff:3058:8e37 ~ fd3b:8170:db39:0:42:acff:fe1e:5: NoRoute
```

Howeer, I think if MA1 is subscrbied on wpan0 TUN, then TD should send Ping Response directly on the infrastructure link rather than going through the Thread Network since the destination address `fd3b:8170:db39:0:42:acff:fe1e:5` is on-link. 

The kernel routing table seems correct:
```
> 2021-11-17 14:36:37,349 - INFO - Otbr<3> $ ip -6 route list | grep "fd3b:8170:db39"
2021-11-17 14:36:37,433 - INFO - Otbr<3> $ 'fd3b:8170:db39::/64 dev eth0 proto kernel metric 256 expires 1783sec pref medium'
2021-11-17 14:36:37,433 - INFO - Otbr<3> $ 'fd3b:8170:db39::/64 dev wpan0 metric 512 pref medium'
```

Any thoughts? @edmont @abtink @jwhui

edmont, 2021-11-17:
@simonlingoogle thanks for looking into this, I was suspecting this is a similar case than in #7145.

Notice that in this test TD may or may not have Ethernet connection with the backbone. So the ping reply should be able to go through the Thread interface.

edmont, 2021-11-17:
By the way, most of the time I try to rerun the test I got this error, even if I start over with `git clean`:

Nov 17 08:10:50 85f93c0ab656 otbr-agent[111]: 49d.17:47:08.476 [CRIT]-PLAT----: HandleRcpTimeout() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2218: RadioSpinelNoResponse

A couple of times I had to reboot my PC to make it work. Any hint to have successful runs every time?

simonlingoogle, 2021-11-17:
> Notice that in this test TD may or may not have Ethernet connection with the backbone. So the ping reply should be able to go through the Thread interface.

TD shouldn't have backbone in test spec since otbr is only used for BR_1 and BR_2 roles. 

In practice, sending ping response via backbone should be more efficient and thus preferred.

edmont, 2021-11-17:
Actually the certification plan is currently being run with the border router device in all FTD roles (BR_1, BR_2, Router, REED, FED). The device might be physically connected to the backbone when running the test, but in any case for those other roles it's not configured as BBR in `__startOpenThread()` in the THCI.

simonlingoogle, 2021-11-17:
> Actually the certification plan is currently being run with the border router device in all FTD roles (BR_1, BR_2, Router, REED, FED). The device might be physically connected to the backbone when [...]

Should we disconnect the device from backbone link in such cases?

But before this discussion, at least we should make the ping success first.

edmont, 2021-11-17:
> Should we disconnect the device from backbone link in such cases?

I don't think there is any statement about that in the certification program.

According to the certification test, I added a default route SLAAC prefix to the BR1 so that there is a proper source address for the unicast ping reply to the host. I noticed that `Ip6::ShouldForwardToThread` (in `Ip6::HandleDatagram`) is falling in the final `else`, thus `LeaderBase::RouteLookup` returns `false`. More logging shows that the matching prefix is found, but `LeaderBase::DefaultRouteLookup` fails to assign a value to `route`, even when the flag is seen in the Network Data in the PCAP.

simonlingoogle, 2021-11-18:
This seems to be a hard one. Some suspicion and observations:
- I suspect that TD failed to route because it's also the BR for routing the prefix, so that TD needs to route to TD itself for the Ping Response, which makes little sense. 
- I mirrored MA1 on TD's wpan0 interface and it was still not sending Ping Response because **kernel always send Ping Response via the netif that have subscribed MA1**. So, kernel will choose the route to wpan0 instead of eth0. 
- When TD routes the Ping Response, not sure which behavior is more reasonable:
  1. Just forward the message back to Eth0 because itself is the BR that provides the external route
  2. Forward to other BRs

So, there seems two fixes that is needed:
- Mirror MA1 from Thread to `wpan0`
- Fix `NoRoute` issue for the `Ping Response`

simonlingoogle, 2021-11-18:
The situations are different for Border Router CI test and certification. 

Did some summary:

| Conditions and Requirements                                | Border Routing                                                                                                                        | Certification                                                                                                                                                  |
|------------------------------------------------------------|---------------------------------------------------------------------------------------------------------------------------------------|----------------------------------------------------------------------------------------------------------------------------------------------------------------|
| Ping Request Source Address                                | Host's on-link address                                                                                                                | Host's GUA address                                                                                                                                             |
| External Route for Ping Response                           | on-link prefix is added to external routes                                                                                            | GUA Prefix is NOT added to external routes                                                                                                                     |
| Ping Response source address                               | TD's OMR address                                                                                                                      | TD's DUA address (when there is no OMR address)                                                                                                                |
| R1: MA should be mirrored on wpan0                         | No, need fix                                                                                                                          | No, need fix                                                                                                                                                   |
| R2: Kernel should choose route via wpan0 for Ping Response | Yes (thanks to `OPENTHREAD_POSIX_CONFIG_INSTALL_EXTERNAL_ROUTES_ENABLE`)                                                              | No, need fix. Kernel won't be able to find a route for Ping Response because MA is subscribed on wpan0 but there is no route via wpan0 for Host's GUA address. |
| R3: TD should route the Ping Response to Host              | No, failed with NoRoute error. Need fix.   TD should forward the message back to its own backbone directly since it's the best route. | Yes because PBBR is the default route for DUA address.                                                                                                         |

Please comment if anything looks incorrect.  @edmont @jwhui @abtink

edmont, 2021-11-18:
@simonlingoogle very nice summary, I agree. Just a typo, for this specific test case there is no prefix with DUA flag enabled, so TD has GUA address, not DUA.

edmont, 2021-11-18:
>       * Unlike unicast addresses, commands like `ip -6 addr add <ip> dev wpan0` does not work for multicast addresses. Again, I would suggest to call `Address Callback` for external addresses (both Unicast and Multicast), too.

For a real application it makes little sense to me to use an address callback for multicast. If the BR device is interested in subscribing to a multicast group, it means there is an application opening such socket and the address will be added by the kernel to the TUN interface.

For certification purposes I guess https://github.com/openthread/openthread/blob/main/tests/scripts/thread-cert/mcast6.py could be used as a basic application.

simonlingoogle, 2021-11-18:
> For a real application it makes little sense to me to use an address callback for multicast. If the BR device is interested in subscribing to a multicast group, it means there is an application opening such socket and the address will be added by the kernel to the TUN interface.

Make sense to me. Responding to multicast ping is useful for certification but not that useful in real world applications.

simonlingoogle, 2021-11-18:
> @simonlingoogle very nice summary, I agree. Just a typo, for this specific test case there is no prefix with DUA flag enabled, so TD has GUA address, not DUA.

Thanks for the tip. I think it's still `fd00:7d03:7d03:7d03::/64` but with `P_dp=0`. I guess let's still call it DUA prefix since it works basically same as `P_dp=1`.

edmont, 2021-11-18:
Just added a PR with a test in order to be on the same page: https://github.com/openthread/openthread/pull/7183

edmont, 2021-11-18:
With the test the ping response message is being dropped in `ThreadNetif::RouteLookup` https://github.com/openthread/openthread/blob/main/src/core/thread/thread_netif.cpp#L240

edmont, 2021-11-18:
>    **kernel always send Ping Response via the netif that have subscribed MA1**. So, kernel will choose the route to wpan0 instead of eth0.

I believe we should live with this and assume the ping reply should go back to the Thread network. If the application wants to listen to traffic on the eth0 interface it should subscribe the multicast address to that one as well.

That said we should find the proper conditions to forward the message to another BR (next 802.15.4 destination). This basically means that under the `OPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE` condition a route matching the own device should be discarded when the message comes from the host (`aFromNcpHost`).

If another BR is providing a route, the message with be forwarded, otherwise it will be dropped.

Do you agree with this reasoning?

simonlingoogle, 2021-11-18:
> I believe we should live with this and assume the ping reply should go back to the Thread network. If the application wants to listen to traffic on the eth0 interface it should subscribe the [...]

The ping reply will go back to the TUN interface, and thus be handled by TD's Thread stack. 

> That said we should find the proper conditions to forward the message to another BR (next 802.15.4 destination). This basically means that under the `OPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE` [...]

I think that's a fair point. Currently, since the ping reply has `aFromNcpHost=true`, so it will never be forwarded back to TUN. However, the ping reply can't be forwarded by MeshForwarder either because MeshForwarder can never forward a message to itself. 

> If another BR is providing a route, the message with be forwarded, otherwise it will be dropped.

Seems logical for MeshForwarder  to do so since it can't forward to itself anyway. 

> Do you agree with this reasoning?

I wonder which is better for the ping reply in the Border Router (with OMR address) case:
1. Forward the ping reply to PBBR (by fixing MeshForwarder)
2. Forward the ping reply back to Backbone immediately (by fixing `aFromNcpHost` logic)

I tend to think 1 is the more correct behavior.

edmont, 2021-11-19:
> Maybe we can see if an implementation of 1 can fix the test #7183

Just pushed some changes that make the test pass. Let's see if it doesn't break anything else.

edmont, 2021-11-19:
`test_dua_routing_med.py` is failing, most likely because the rule breaks the case when the traffic originated in the host has one of the BR children as destination.

simonlingoogle, 2021-11-22:
@edmont 
I have sent https://github.com/openthread/openthread/pull/7187 to fix the routing issue on BR (TD). 

I think it should solve the `Border Routing` case by forwarding the ping reply back to Backbone. 

For the `certification` case, the ping reply will still be sent via Thread traffic, so it's not violating the test plan. 

To fix R2 in `certification`, there may be two approaches:
a1) Add a default route via `wpan0` in kernel.
a2) Always handle (multicast) ping requests in Thread stack if it's not PBBR.

edmont, 2021-11-22:
Thanks @simonlingoogle, I'm currently testing your changes. What do you mean by `R2`?

simonlingoogle, 2021-11-22:
> Thanks @simonlingoogle, I'm currently testing your changes. What do you mean by `R2`?

I mean R2 in https://github.com/openthread/openthread/issues/7175#issuecomment-972613573

edmont, 2021-11-22:
Thanks for clarification. I wonder if that's even needed. Test plan requirement is:
| Step | Device | Description | Pass criteria |
| --- | --- | --- | --- |
| 10 | TD | Receives the multicast ping packet and sends a ping response packet back to Host. | Receives the MPL packet containing an encapsulated ping packet to MA1, sent by Host, and unicasts a ping response packet back to Host. |

This should be met even if the ping response is sent over the Ethernet link directly.

simonlingoogle, 2021-11-22:
> This should be met even if the ping response is sent over the Ethernet link directly.

Even with https://github.com/openthread/openthread/pull/7187, I think TD won't send ping reply to Host either via Thread or via Backbone in the certification setup since R2 is not fixed.

We can run the test in certification like setup to verify.

edmont, 2021-12-15:
Finally the certification test is passing with a combination of:

- https://github.com/openthread/openthread/pull/7187
- https://github.com/openthread/openthread/pull/7193
- https://github.com/openthread/openthread/pull/7233

edmont, 2021-12-17:
The combination of https://github.com/openthread/openthread/pull/7187 and https://github.com/openthread/openthread/pull/7193 fixes the certification issue, assuming that the TD enables `OTBR_BORDER_ROUTING`.

https://github.com/openthread/openthread/pull/7187 alone fixes https://github.com/openthread/openthread/pull/7183.

simonlingoogle, 2021-12-20:
Seems #7187 is necessary for certification and fixing multicast ping issues. 

Should we merge #7187 and use Security Safeguard to help avoid message loops? 

@abtink @jwhui

abtink, 2021-12-20:
- I would prefer if we could find a maybe  more surgical solution (smaller in scope and change) to fix this:
  - For example, can we limit the behavior change (i.e. passing back the message we get from `otIp6Send` back to caller in callback `otIp6ReceiveCallback`) to only the multicast addresses or even stricter rule and limit it to to the BBR related address(es)?

- I am ok adding this PR (as a last option) but if we want to add it, I highly suggest we change the PR so that the behavior change (i.e. the fact that we may now pass back a message we get from `otIp6Send` back to caller in callbck) under an API or OT build flag (instead of fully removing such a behavior from the OT code). This way we have an option to re-enable it. 
   - I do feel that such a change will have unintended consequences (that we may not be able to see or think of now). I just don't want us to end up in a situation to need to re-add it later. On BR (or posix platform) we may be able to use firewall/filtering to limit the message, but we should consider other platforms (SEDs, ED, etc) which may be impacted.

simonlingoogle, 2021-12-21:
> * I highly suggest we change the PR so that the behavior change (i.e. the fact that we may now pass back a message we get from `otIp6Send` back to caller in callbck) under an API or OT build flag (instead of fully removing such a behavior from the OT code).

One problem with `otIp6Send` is that it does not differentiate between **initiating** a packet or **forwarding** one. 

- If the BR is initiating a packet using `otIp6Send`, it should have no difference from initiating the message from the Thread stack (e.g. using Ip6::SendDatagram). So, BR should be allowed to forward the packet back to Host if necessary. 
- If the BR is forwarding a packet (usually from infrastructure links), the message should never be forwarded back to Host. So, the `aFromNcpHost` checking make perfect sense in such case. 

That being said, I am fine with add a configuration to allow enabling it later.

abtink, 2021-12-21:
I agree.I think maybe it makes sense to later add two different public APIs for the two case. One that may pass the user msg back to it in the callback and and one that does not. We can consider it as a future change.

simonlingoogle, 2021-12-22:
> I agree.I think maybe it makes sense to later add two different public APIs for the two case. One that may pass the user msg back to it in the callback and and one that does not. We can consider it as a future change.

That make sense. 

However, in ot-br-posix, we actually don't know if it's initiating or forwarding when a message is read from the TUN interface. So, even introducing two APIs wouldn't help it in Posix.

abtink, 2021-12-22:
Good point. Maybe on posix platform we can consider always using the API that allows msg to be passed back in callback and then we rely on firewall/filtering. Other type of nodes (SEDs, EDs) or maybe other platform can choose to use the other API (if they don't have the firewall/filtering).

gjc13, 2022-01-21:
After https://github.com/openthread/openthread/pull/7187, we've observed MLDv2, RS and other ICMPv6 packets from the kernel forwarded to the Thread network. This is caused by the ICMP6 type check turned off by `aFromNcpHost`.
These packets can cause heavy network traffic and are not necessary on Thread network. Shall we always enable ICMP type checks for `forwardThread`?

simonlingoogle, 2022-01-21:
> After https://github.com/openthread/openthread/pull/7187, we've observed MLDv2, RS and other ICMPv6 packets from the kernel forwarded to the Thread network. This is caused by the ICMP6 type check turned off by `aFromNcpHost`.
> 
> These packets can cause heavy network traffic and are not necessary on Thread network. Shall we always enable ICMP type checks for `forwardThread`? @simonlingoogle @abtink 

+1 for always enabling ICMP filtering.

gjc13, 2022-01-21:
Created https://github.com/openthread/openthread/pull/7343