openthread/openthread #7186: OTBR not working with OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1"

OlivierGre, 2021-11-19:
I have built the OTBR version 1.1 thanks to the commands indicated in another ticket:

```
cd ot-br-posix
./script/bootstrap
INFRA_IF_NAME=wlan0 BACKBONE_ROUTER=0 OTBR_OPTIONS="-DOT_THREAD_VERSION=1.1" ./script/setup
```
My RCP is connected and I have rebooted the boards. 

The OTBR agent status indicates a problem:
```
pi@raspberrypi:~ $  sudo service otbr-agent status
● otbr-agent.service - Border Router Agent
     Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2021-11-19 14:05:13 CET; 890ms ago
    Process: 1182 ExecStart=/usr/sbin/otbr-agent $OTBR_AGENT_OPTS (code=exited, status=2)
   Main PID: 1182 (code=exited, status=2)
        CPU: 26ms
```

I see something strange in syslog. It contains "Thread version: 1.2.0" whereas I have built for Thread 1.1. Do you know what I could have done wrong? 

```
pi@raspberrypi:~ $ cat /var/log/syslog | grep otbr-agent
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Running 0.3.0-4555d899e
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Thread version: 1.2.0
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Thread interface: wpan0
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Backbone interface: wlan0
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Nov 18 09:03:00 raspberrypi otbr-agent[431]: [INFO]-UTILS---: Radio URL: trel://wlan0
Nov 18 09:03:00 raspberrypi otbr-agent[431]: 49d.17:02:55.059 [CRIT]-PLAT----: Init() at ../../third_party/openthread/repo/src/posix/platform/hdlc_interface.cpp:151: InvalidArgument
Nov 18 09:03:00 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Nov 18 09:03:00 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Nov 18 09:03:05 raspberrypi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 1.
```

Yesterday I have had a log which was showing the version 1.1.1:
```
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Running 0.3.0-e623f2a5e
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Thread version: 1.1.1
Nov 18 15:58:27 raspberrypi otbr-agent[2592]: [INFO]-UTILS---: Thread interface: wpan0
```
Thank you for your help

wgtdkp, 2021-11-19:
Apparently, the two OTBRs are not the same.

The OTBR in the syslog has version of `Running 0.3.0-4555d899e`, but the OTBR with Thread version 1.1.1 has version of `0.3.0-e623f2a5e` which includes fix in https://github.com/openthread/ot-br-posix/pull/1099.

If you are building and installing OTBR with `./script/setup`, please make sure it ends up with success status - not only compiles success but also installs success. Execute the two commands
```shell
sudo ./build/otbr/src/agent/otbr-agent -V
sudo otbr-agent -V
```
and make sure they give the same output. If the outputs are different, you must have failed to install the otbr-agent binary and please upload the complete building outputs here.

wgtdkp, 2021-11-19:
To remove a previously installed otbr-agent binary:
```
sudo service otbr-agent stop
sudo rm -rf `which otbr-agent`
```

OlivierGre, 2021-11-19:
Thanks.
I have the same version:
```
pi@raspberrypi:~/ot-br-posix $ sudo ./build/otbr/src/agent/otbr-agent -V
0.3.0-e623f2a
pi@raspberrypi:~/ot-br-posix $ 
pi@raspberrypi:~/ot-br-posix $ sudo otbr-agent -V
0.3.0-e623f2a
```

I'm going to uninstall the otbr-agent and to reinstall it. I will pay attention if the installation is OK.

OlivierGre, 2021-11-19:
I have noticed that syslog file contained information with yesterday's date, which is strange. I have flushed it and reinstalled the otbr agent. Please find attached the log of the setup command. 

I can see the following errors:
```
script/_otbr: line 49: install_manifests.txt: No such file or directory

+ systemctl is-enabled NetworkManager
Failed to get unit file state for NetworkManager.service: No such file or directory
```

Then I have plugged my RCP and rebooted the raspberry. I attach the syslog. Now it indicates Thread version: 1.1.1 and displays this error:
`Nov 19 15:59:48 raspberrypi otbr-agent[461]: 49d.17:02:55.030 [CRIT]-PLAT----: Init() at ../../third_party/openthread/repo/src/posix/platform/hdlc_interface.cpp:151: InvalidArgument`

For my information is there a difference concerning the communication with the RCP between Thread 1.1 and 1.1.1? I think that I have an otbr-agent running Thread 1.1.1 and a RCP running Thread 1.1.

Thank you

[setup.log](https://github.com/openthread/openthread/files/7571113/setup.log)
[syslog.log](https://github.com/openthread/openthread/files/7571115/syslog.log)

wgtdkp, 2021-11-19:
Please check your otbr-agent configuration file:
```
 ⚡  cat /etc/default/otbr-agent
# Default settings for otbr-agent. This file is sourced by systemd

# Options to pass to otbr-agent
OTBR_AGENT_OPTS="-I wpan0 -B eno1 spinel+hdlc+uart:///dev/ttyACM0 trel://eno1"
```

Make sure the RCP device (`/dev/ttyACM0` is my case) is truly your connected RCP board. You can find the RCP board in directory `/dev`. Change the configuration file to use your actual RCP board if they differ.